esphome/home-assistant-voice-pe #44: Enable out-of-the-box compatibility with third-party voice-kits like the FutureProofHomes Satellite1 PCB Dev Kit and Seeed Studio’s newReSpeaker Lite Voice Assistant Kit?

Hedda, 2024-08-08:
Maybe this is the wrong place to ask but wondering if there are plans to work in some template framework for ESPHome components like the [voice_assistant](https://github.com/esphome/voice-kit/tree/dev/esphome/components/voice_assistant) components and the [i2s_audio](https://github.com/esphome/voice-kit/tree/dev/esphome/components/i2s_audio) component for ESPHome to allow for out-of-the-box compability with third-party voice-kits that are also based on ESPHome and using similar hardware with the XMOS chips and ESP32-S3 combination but perhaps are using different pins or maybe another SKU/model of XMOS xCORE chip?

I know this is all still very early in development and Nabu Casa have not even yet formerly announced your upcoming voice-kit hardware platform but already there are third-party voice-kits being made available or announced, such as example the "**_Satellite1 PCB Dev Kit_**" by @FutureProofHomes and the new "**_ReSpeaker Lite Voice Assistant Kit_**" by @Seeed-Studio  and I am sure that we will see a few more before the end of this year. See:

- https://futureproofhomes.net/products/satellite1-pcb-dev-kit
  - https://github.com/FutureProofHomes/Satellite1-Hardware
  - https://www.youtube.com/watch?v=dRTLjQHfjSM

- https://wiki.seeedstudio.com/xiao_respeaker/
  - https://wiki.seeedstudio.com/respeaker_lite_ha/
  - https://www.seeedstudio.com/ReSpeaker-Lite-Voice-Assistant-Kit-Full-Kit-of-2-Mic-Array-pre-soldered-XIAO-ESP32S3-Mono-Enclosed-Speaker-and-Enclosure.html
  - https://www.seeedstudio.com/ReSpeaker-Lite-Voice-Assistant-Kit-p-5929.html

Then there should maybe is the future consideration of additional other projects that might not use XMOS xCORE chips but other DSP chips, such as for example this project by @alextrical where he is instead using ZL38063, though I understand that would be out of scope.

- https://community.home-assistant.io/t/far-field-satellite-with-an-elegant-3d-printed-enclosures/699893/
  - https://community.rhasspy.org/t/question-would-anyone-be-interested-in-a-open-source-dsp-mic-array/4941/

PS: Off-topic but just wanted to add that personally I am hoping to also see one ore more variants of updated ready-made clones of the "Onju Voice" open-source hardware PCB design project in the future with XMOS xCORE chips that are drop-in replacements for all various models of Google Nest speakers and displays or Amazon Echo products that can work out-of-the-box so that users can reuse the enclosure and speakers of their existing products that might soon be outdated or irrelevant once we start fully moving over to Home Assistant's Voice Assistants ecosystem.

* https://github.com/justLV/onju-voice/

tbrasser, 2024-08-08:
Quick reaction/question (on that last part), what is the current onju lacking with respect to xmos based i2s dacs? (that voice-kit relies on?)

Hedda, 2024-08-08:
> Quick reaction/question (on that last part), what is the current onju lacking with respect to xmos based i2s dacs? (that voice-kit relies on?)

Current onju-voice PCB design does not feature an XMOS chip, instead it only has ESP32-S3. See feature request discussion here:

* https://github.com/justLV/onju-voice/issues/59

kahrendt, 2024-08-11:
We are moving fast and breaking things as we figure out the best way for these components to interact, but we will add all the code to the base ESPHome project once things are stable and working well. One goal for this repo is to not be reliant on specific hardware configurations. These other boards you linked are quite exciting, and I believe they should be compatible (or relatively easily made compatible) with our changes. For example, all the I2S settings and pins are still configurable in yaml, so it should be straightforward to add support for similar boards.

Very little of the code is reliant specifically on the XMOS chip (and the few lines that are should be adaptable or won't even be there in the final version as we clean up the code), so it should be possible to add support for other DSPs in the future.

Hedda, 2024-08-13:
> it should be possible to add support for other DSPs in the future.

This is great! Yeah, I think that could be a very cool idea there could be to use the RP2040 or the new RP2350 MCU chip (from Raspberry Pi) and run open-source software audio DSP instead of the propriatory XMOS xCORE MCU chip.

That is, combine an ESP32 with RP2040 (or RP2350) on the same development board as I believe that combinatu would make for interesting development kit that could attract developers from other projects as well.

There are a few projects that use the PIO (Programmable Input Output) of the RP2040 / RPi Pico to create a audio DSP. However I could not find any projects that have voice specific software for audio DSPs, but maybe someone else know of such?

There are even projects that make sure of RP2040 (and Raspberry Pi Pico) as a GPU for graphic output to a display.

PS: I would think that having an reference hardware platform as a dev-kit for sale that would be attractive to other projects as well would be benefitial to achievibg better economy-of-scale that should keep the price down, which in turn should attract even more developers, especially those who like the concept of keeping everything 100% open-source.

Hedda, 2024-08-13:
> , I think that could be a very cool idea there could be to use the RP2040 or the new RP2350 MCU chip (Raspberry Pi Pico 2) and run open-source software audio DSP instead of the propriatory XMOS xCORE MCU chip.
> 
> That is, combine an ESP32 with a RP2xxx chip on the same development board as I believe that combinatu would make for interesting development kit that could attract developers from other projects as well.
> 
> UPDATE: FYI, just read now that the new RP2350 ARM Cortex-M33 cores have integrated DSP and FPU hardware acceleration pipelines, so the new RP2350 would at least deffinitivly be a better choice over the older RP2040.

@kahrendt Awesome timing! FYI, Raspberry Pi Foundation themselves has just a few days ago posted this bog article about Real-time ML audio noise suppression on Raspberry Pi Pico 2:

https://www.raspberrypi.com/news/real-time-ml-audio-noise-suppression-on-raspberry-pi-pico-2/

That blog delve into how an existing ML-based audio noise suppression algorithm can be deployed to Raspberry Pi’s RP2350 microcontroller used in the new Pico 2 board.

nielsnl68, 2024-08-16:
> I spotted @nielsnl68 and @nanosonde writing they are using XMOS’s XK-VOICE-L71 (xCORE Voice Reference Design Evaluation Kit)and XFV3610-IN 
You spotted that wrong 😆 i'm using the M5Atom Echo for my Combadge project and th Raspaudio protokits for my inhouse audio services :D

closed 2024-08-19 by jesserockz