esphome/home-assistant-voice-pe #203: Project overview, purpose and scope

richardsj, 2024-11-19:
The description of the purpose, proposed functionality, skills needed (hardware or just software) and state of development are missing. Instead the Github page for this project just shows the Code of Conduct.

It appears to be a self-sufficient satellite with local micro model for the wakeword, and something to do with ESPHome hardware.

The vision might be exciting, encourage others to get involved!

Hedda, 2024-11-21:
Nabu Casa's devs are still [in-company stealth mode](https://en.wikipedia.org/wiki/Stealth_mode) about this and trying to keep details on the "voice satellite" hardware as secret as possible until it is time to be released, however much can be gleaned and my inofficial take on what is going on is here:

* https://github.com/esphome/home-assistant-voice-pe/issues/52

FYI, Home Assistant's core developers have during each of the last few Release Parties videos been giving several not-so-subtle verbal hints that some kind of official Home Assistant "voice satellite" hardware based on ESPHome is not only in the works by Nabu Casa developers but that hardware is likely to be released relatively soon, though there are no details yet if that first "voice satellite" hardware to be released will be a more of a all-in-one development kit or more of a finished retail product with "Home Acceptance Factor", so keep watching -> https://www.youtube.com/@home_assistant and https://www.home-assistant.io/blog/

Anyway, if you are a developer or an advanced tinkerer who can not wait and wants to try this code/project today on similar development hardware to what is used to develop this then suggest check out the “ReSpeaker Lite” discussion in Home Assistant's community forum here -> https://community.home-assistant.io/t/respeaker-lite-new-seeed-studio-voice-assistant-development-kit-hardware-combine-esp32-with-xmos-xu316-dsp-chip-for-advanced-audio-processing-as-a-esphome-based-home-assistant-assist-satellite-voice-devkit/756944

balloob, 2024-11-23:
Keep an eye out on the Home Assistant blog to learn more soon!

closed 2024-11-23 by balloob