esphome/home-assistant-voice-pe #230: Upstream custom components (nabu, media_player, ...) to esphome

flubshi, 2024-12-21:
Hi there,

I am following this repo for a few months and I am very happy that you released the hardware this week :tada: Thanks for all the awesome work!

I noticed that you have created custom components and modified the original esphome components, e.g. *nabu*, *media_player*:

```
external_components:
  - source:
      type: git
      url: https://github.com/esphome/home-assistant-voice-pe
      ref: dev
    components:
      - aic3204
      - audio_dac
      - media_player
      - micro_wake_word
      - microphone
      - nabu
      - nabu_microphone
      - voice_assistant
      - voice_kit
    refresh: 0s
```

The nabu media player is awesome because it supports ducking and local files. I have embed these external components in my configuration files for a while.

Do you plan to upstream those new components and the other improvements to esphome?

kahrendt, 2024-12-22:
Yes, we are working on merging all the changes upstream into ESPHome. For example, the media player code has a preliminary PR here <https://github.com/esphome/esphome/pull/7672>. I'm actively working on further compartmentalizing that PR to make it work well in more use cases; e.g., non ESP32-S3 devices.

AndreKR, 2025-01-05:
@kahrendt You probably already know that, but just in case: Currently the custom `voice_assistant` component from this repo has a hardcoded dependency on `micro_wake_word`.

```
src/esphome/components/voice_assistant/voice_assistant.cpp: In member function 'void esphome::voice_assistant::VoiceAssistant::on_set_configuration(const std::vector<std::__cxx11::basic_string<char> >&)':
src/esphome/components/voice_assistant/voice_assistant.cpp:899:13: error: 'class esphome::voice_assistant::VoiceAssistant' has no member named 'micro_wake_word_'; did you mean 'use_wake_word_'?
  899 |   if (this->micro_wake_word_) {
      |             ^~~~~~~~~~~~~~~~
      |             use_wake_word_
```

kahrendt, 2025-01-21:
> [@kahrendt](https://github.com/kahrendt) You probably already know that, but just in case: Currently the custom `voice_assistant` component from this repo has a hardcoded dependency on `micro_wake_word`.

Thanks for pointing this out! #291 fixed this.

flubshi, 2025-05-01:
It is absolutely awesome to see the progress in the voice projects. It's fantastic that the new features aren't limited to VPE, but that @kahrendt you're making them available to everyone upstream via esphome. Thank you for your great work!
I think we can close this issue.