esphome/home-assistant-voice-pe #231: Feature Request: Select HA Media Player Entity as default Speaker

bluenazgul, 2024-12-22:
Like it is possible on Amazon Echo Devices and Google Home / Google Nest Devices it would be great to be able to select an Media Player via HA the is the default Speaker for the Voice PE

balloob, 2024-12-22:
Thanks for the feedback. This would actually be a feature that Home Assistant would need to add, not something the VPE could add. 

Feature requests can be added in the HA forums: https://community.home-assistant.io/c/feature-requests/13

Plawasan, 2025-01-05:
> Thanks for the feedback. This would actually be a feature that Home Assistant would need to add, not something the VPE could add.
> 
> Feature requests can be added in the HA forums: https://community.home-assistant.io/c/feature-requests/13

Would it not be possible directly from ESPhome? I've seen this method used by a different project:

[image]