esphome/home-assistant-voice-pe #236: No longer starting up after power disconnect

Xitee1, 2024-12-23:
I've just got my Home Assistant Voice Preview Edition and after initial setup it worked really good.
Then I wanted to move it to another location, so I unplugged it and plugged it back in. It now no longer starts up.
The LEDs are blinking weirdly and when checking the logs, I'm getting these wifi error messages:

```
12:04:53][D][power_supply:033]: Enabling power supply.
[12:04:54][D][wifi:497]: Found networks:
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][D][wifi:546]: - [redacted] [redacted]▂▄▆█
[12:04:54][W][wifi:551]: No matching network found!
[12:04:54][D][power_supply:033]: Enabling power supply.
[12:04:55][D][power_supply:033]: Enabling power supply.
[12:04:56][D][power_supply:033]: Enabling power supply.
[12:04:57][D][power_supply:033]: Enabling power supply.
[12:04:59][D][power_supply:033]: Enabling power supply.
[12:04:59][D][wifi:482]: Starting scan...
[12:04:59][D][esp-idf:000][wifi]: W (86935) wifi:
[12:04:59][D][esp-idf:000][wifi]: Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!
[12:04:59]
[12:04:59][D][esp-idf:000][wifi]: 
[12:04:59]
[12:05:00][D][power_supply:033]: Enabling power supply.
[12:05:01][D][power_supply:033]: Enabling power supply.
[12:05:02][D][power_supply:033]: Enabling power supply.
```

I don't know if that could be the reason but my network is hidden and normally I have to set the option `fast_connect` to `True`.

Xitee1, 2024-12-23:
By temporarily making my wifi network visible I was able to connect it again.

alvinchen1, 2024-12-23:
Yup, this is a common thing with ESPHome and WiFi.  If the Network SSID you are connecting to is HIDDEN, you need an additional configuration values which is NOT on the current firmware.  You need to add your SSID, Password and the fast_connect:true

```
wifi:
  id: wifi_id
###  Add 
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
###  End
  on_connect:
    - lambda: id(improv_ble_in_progress) = false;
    - script.execute: control_leds
  on_disconnect:
    - script.execute: control_leds

```

If you are familiar with ESPHome Addon (or now called ESPHome Device Builder), you can grab this YAML file
https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml
And at 
`
wifi: 
`
add the values and compile and the device will reconnect after reboots, instead of starting at ImprovBLE or out of the box connection configuration.

Xitee1, 2024-12-23:
Thanks, that works for now.

lpdescamps, 2025-01-12:
> Yup, this is a common thing with ESPHome and WiFi. If the Network SSID you are connecting to is HIDDEN, you need an additional configuration values which is NOT on the current firmware. You need to add your SSID, Password and the fast_connect:true
> 
> ```
> wifi:
>   id: wifi_id
> ###  Add 
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   fast_connect: true
> ###  End
>   on_connect:
>     - lambda: id(improv_ble_in_progress) = false;
>     - script.execute: control_leds
>   on_disconnect:
>     - script.execute: control_leds
> ```
> 
> If you are familiar with ESPHome Addon (or now called ESPHome Device Builder), you can grab this YAML file https://github.com/esphome/home-assistant-voice-pe/blob/dev/home-assistant-voice.yaml And at `wifi: ` add the values and compile and the device will reconnect after reboots, instead of starting at ImprovBLE or out of the box connection configuration or having to leave your network visible during boot up.

i copied this yaml to my esphome server, modified the wifi but got an error when validating,

Validate home-assistant-voice.yaml ❌
INFO ESPHome 2024.10.1
INFO Reading configuration /root/config/home-assistant-voice.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
INFO Updating https://github.com/esphome/esphome.git@pull/8018/head
INFO Unable to import component audio: No module named 'esphome.components.audio'
Failed config

audio: None
  {}
Component not found: audio

boutch, 2025-03-13:
Same problem here. I had to adopt manually the device with ESPHome which means I won't receive automatically the new updates. It would be great to add fast_connect=true to the master code.

hardwareadictos, 2025-03-30:
> Same problem here. I had to adopt manually the device with ESPHome which means I won't receive automatically the new updates. It would be great to add fast_connect=true to the master code.

Affected with this too, that would be also nice to implement as this will be (supposedly) on the future an "end user" device. End users don't want to mess with yaml files and/or flash/parametrize devices manually.

sk7, 2025-04-08:
The [home-assistant-voice.yaml](https://github.com/esphome/home-assistant-voice-pe/blob/3d992dd52bf7dad3f307510ed768aee7d4cc4672/home-assistant-voice.yaml#L18) has now this line in the `substitutions` section at the beginning:
```
hidden_ssid: "false"
```
but you still need to compile the firmware, adopt the device with ESPHome manually and also update the firmware manually, like already mentioned.

Xitee1, 2025-04-10:
> but since the Voice PE asks for SSID and password during the initial setup, I don't see how this can be an issue.

This has nothing to do with SSID/password but rather to which AP it connects to.
According to the docs:
> even if that network is very far away and better ones are available.

So if it first sees an access point that is very far away, it connects to it without looking for closer ones.

However, I would still vote for enabling it by default because I could imagine that more people have a hidden wifi network than multiple APs. And people with multiple APs probably know their setup better and will most likely lock the HA voice to a specific AP anyway (especially if they notice the device is connecting to the "wrong" AP). For the most users, this would probably be the better default value. But that's just what I am thinking and I could be wrong.

Best case would be to just ask the user on setup if the network is hidden.

hardwareadictos, 2025-06-07:
Hey!! Do we have any news about this? Yesterday i had a power cutout and today i discovered that my Voice PE was on a loop having to reconnect it to the wifi and having to reconfigure it. Any progress? :)

Kieros, 2025-07-08:
> Oh ok thanks for the clarification. And you've got a good idea there! It would be cool to have a checkbox when adding the SSID and the password of the wifi to indicate if it's a hidden network. The checkbox would be by default unchecked and when checking it, it would enable fast_connect. Something like this:
> 
> [image]

This is also a very nice way to do it I think.

hardwareadictos, 2025-11-10:
Is there any progress?

Xitee1, 2026-10-03:
Seems to be fixed..

I've recently added a new ESPHome device which is working completely fine.
Now I've just noticed that I forgot to set fast_connect to true, which really confused me because it never worked without that.
So I went ahead and plugged my Voice PE into my PC and flashed the official firmware.
And it connected. Only thing I needed to do was confirm something in the HA integrations page that the device has new encryption or something like that and it detected it as the same device, no new setup needed.

Thanks... for ignoring this issue and not posting any comment for years... but at least it works now.
Leaving this open for now, until other people confirm it working too.

joshuadavidson, 2026-10-07:
Can confirm. I've been compiling my own ESPHome firmware ever since release to get around this hidden wifi network limitation. I just flashed the official 26.9.0 version from https://esphome.github.io/home-assistant-voice-pe/?version=26.9.0 and it connects to my hidden network without issue.