esphome/home-assistant-voice-pe #238: Home assistant Voice PE SSL certificate verification failed

limitless-code, 2024-12-24:
Hello. 

I have purchased HA voice Preview edition and added the device to HASSIO core 2024.12.5. some features work but TTS/ responses do not.

When I select "set up voice assistant" in integrations > ESPHome

After awhile I see the following:

"The voice assistant is unable to connect to Home Assistant
To play audio, the voice assistant device has to connect to Home Assistant to fetch the files. Our test shows that the device is unable to reach the Home Assistant server." 

After some troubleshooting and not getting very far I decided to "Take control" using "ESP Home builder".

After compiling and installing, I see the following in the logs.

```
[23:45:15][D][esp-idf:000][ann_read]: E (2819428) esp-x509-crt-bundle: Failed to verify certificate

[23:45:15][D][esp-idf:000][ann_read]: E (2819431) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000

[23:45:15][D][esp-idf:000][ann_read]: E (2819432) esp-tls: Failed to open new connection

[23:45:15][D][esp-idf:000][ann_read]: E (2819433) transport_base: Failed to open a new connection

[23:45:15][D][esp-idf:000][ann_read]: E (2819434) HTTP_CLIENT: Connection failed, sock < 0

[23:45:15][E][nabu_media_player.pipeline:171]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
[23:45:15][E][nabu_media_player:305]: The announcement pipeline's file reader encountered an error.
```
It seems perhaps the "unable to connect to home assistant to play voice" error may be due to a certificate error? 

Is there a way to get round this problem? Perhaps an ignore SSL error configuration entry or ideally add the appropriate root CA to the bundle? If that is the cause? I believe it is a self-signed certificate.

Commands such as "turn the light on" or off do work, just responses to things like "what time is it?" Or the audio response to successfully turning the light off are not audible. There's no audio from TTS but I can see the text response in the logs using ESP home builder and connecting via WiFi to see the log output.

Your input is very much appreciated.

kahrendt, 2024-12-24:
The media player does support validating certificates, but not all certificate types are supported. For example, it doesn't support TLS1.3 certificates due to a limitation from the esp-tls library.

limitless-code, 2024-12-24:
> The media player does support validating certificates, but not all certificate types are supported. For example, it doesn't support TLS1.3 certificates due to a limitation from the esp-tls library.

Thank you for your input.

[image]

It is indeed using TLS1.3. Will downgrading the certificate to TLS1.2 work for a self signed certificate on HA on HA Voice PE? Would a configuration to ignore (not ideal I know) warnings etc work as a work around and is there such an available configuration option in ESPHome builder?

kahrendt, 2024-12-24:
Downgrading should work. The ESP32 either universally ignores or universally validates certificates. For audio, it is probably not an issue to ignore the certificate, but it is also used when we check for new firmware updates where it is very important. If you take control of the device in the ESPHome builder, then you won't need to check for firmware updates, so I believe you can set it to ignore validating

limitless-code, 2024-12-24:
You are correct

Adding...

```
http_request:
   verify_ssl: false
```

To the ESPHome builder configuration of the device,
stops the device verifying the TLS certificate. As I have self-signed the certificate this is the quickest way to a work around. Pity there isn't a way to specify the host URLs, there seems to be a way for action > http_request.get and post but not http_request from what I can see.

The other more preferred method would be to recreate the certificate with TLS1.2 and then add the root CA to the bundle that is on the esp32 device not sure how to add the bundle to the device.

At least it is working now and something for everyone at home to play with over Xmas.

Many thanks for your help!

closed 2024-12-26 by balloob

abliss, 2025-02-01:
I also have TLS1.3.  i believe I followed the instructions at https://community.home-assistant.io/t/how-to-configure-lets-encrypt-ssl-certificates-for-home-assistant-completely-100-free-updated-for-2022-2023/508329 to use the letsencrypt addon .  Perhaps that should be changed to use tls1.2 by default

davidjmeier, 2025-03-11:
Thank you for posting the workaround here.  Can we please get this in official documentation and potentially as part of the onboarding sequence?  For those of us who have an appropriately deployed local implementation (only local certs to access HA) this breaks in very unfortunate ways - meaning that the device will work sporadically, but is not reliable.  This gives the impression to the end user that something else may be wrong and the certificate may not actually be the issue.  Getting to a point where seeing the actual error is quite a bit of work for a less seasoned end user which will cause frustration.

I definitely did not expect the certificate version (TLS1.3) to be the issue at all given how the device was reacting (and not).  It may be due to the fact that I also subscribe to HA Cloud and maybe the device was somehow flipping between trying to access locally and via cloud?  Not exactly sure, but hopefully this gets resolved and we can get TLS1.3 certs validated with no workarounds in the future.

Appreciate all of the insight in the thread here!

PR #244 (2024-12-26): Enable TLS1.3 support
Adds an SDKCONFIG option to support TLS1.3. It is newly supported in esp-idf 5, but it not enabled by default on the current version (5.1.5) used in ESPHome. This should fix #238