esphome/home-assistant-voice-pe #239: Create sensor (or adjust existing) to recognize current audio output

malhalla, 2024-12-24:
When Voice PE has a status other than idle i would mute my Media Player (eG Sonos) for the time Voice PE is talking. 

But the existing sensors go back to "idle" before audio output from Voice PE is finished.

jackwilsdon, 2024-12-24:
The `assist_satellite` entity should be in state "responding" until it has finished.

I use this automation to pause my TV when interacting with the Voice PE:

```yaml
alias: "Assist: pause TV on action"
description: ""
triggers:
  - trigger: state
    entity_id:
      - assist_satellite.living_room_home_assistant_voice_assist_satellite
    from: idle
conditions:
  - condition: device
    device_id: 903ad7d4030876d2394d9e92ab5600df
    domain: media_player
    entity_id: 4134fcd48e39ad6c18e12c0729a8ca7d
    type: is_playing
actions:
  - action: media_player.media_pause
    metadata: {}
    data: {}
    target:
      device_id: 903ad7d4030876d2394d9e92ab5600df
  - wait_for_trigger:
      - trigger: state
        entity_id:
          - assist_satellite.living_room_home_assistant_voice_assist_satellite
        to: idle
  - action: media_player.media_play
    metadata: {}
    data: {}
    target:
      device_id: 903ad7d4030876d2394d9e92ab5600df
mode: single
```

jlpouffier, 2025-01-17:
That is correct, use the `assist_satellite` is the way to go if you want to detect when the Voice Preview Edition is streaking a response.

closed 2025-01-17 by jlpouffier