esphome/home-assistant-voice-pe #240: WakeWord "Okay Nabu" also reacts to "Okay Google"

bluenazgul, 2024-12-24:
As shown in short Video, when i set my Voice PE to "Okay Nabu" it also reacts when i try to activate Google Home by using "Okay Google"

https://github.com/user-attachments/assets/3434de74-aea0-4fc6-9ad7-6bdc131af412

kahrendt, 2024-12-26:
Wake words are definitely not a precise thing! The model used in the latest beta 2024.12.8 doesn't seem to trigger when I play your video clip right next to the Voice PE, so I think it was trained out.

I do generate TTS samples of "Okay Google" to use as negative samples while training to avoid this as much as possible. Different intonations may still cause it to trigger, but it should hopefully be minimal. Now if my Google speakers didn't falsely accept "okay nabu" the majority of the time, I would be happy!

closed 2024-12-26 by kahrendt

bluenazgul, 2024-12-27:
you are right, with the 2024.12.9 Beta Firmware it seems to be fixed