esphome/home-assistant-voice-pe #241: Voice PE bootloops after power reconnect and wifi is connected (watchdog)

BobMcGlobus, 2024-12-24:
I received two voice pe today. After the first setup the bootloop problems started. When I reflash the boards everything works until I remove the power, then they just bootloop.
Does anyone have the same problem, or has an idea?
```
[18:01:54][I][wifi:617]: WiFi Connected!
[18:01:54][C][wifi:428]:   Local MAC: 20:F8:3B:09:30:F4
[18:01:54][C][wifi:433]:   SSID: 'MyWifi'[redacted]
[18:01:54][C][wifi:436]:   IP Address: 192.168.2.78
[18:01:54][C][wifi:439]:   BSSID: 5C:E9:31:B1:71:F2[redacted]
[18:01:54][C][wifi:441]:   Hostname: 'home-assistant-voice-0930f4'
[18:01:54][C][wifi:443]:   Signal strength: -44 dB ▂▄▆█
[18:01:54][C][wifi:447]:   Channel: 6
[18:01:54][C][wifi:448]:   Subnet: 255.255.255.0
[18:01:54][C][wifi:449]:   Gateway: 192.168.2.1
[18:01:54][C][wifi:450]:   DNS1: 192.168.2.160
[18:01:54][C][wifi:451]:   DNS2: 1.1.1.1
[18:01:54][C][api:026]: Setting up Home Assistant API server...
[18:01:54][C][micro_wake_word:083]: Setting up microWakeWord...
[18:01:54][C][micro_wake_word:109]: Micro Wake Word initialized
[18:01:54][I][app:062]: setup() finished successfully!
[18:01:58][D][esp-idf:000][wifi]: I (20245) wifi:
[18:01:58][D][esp-idf:000][wifi]: <ba-add>idx:1 (ifx:0, 5c:e9:31:b1:71:f2), tid:0, ssn:0, winSize:64
[18:01:58][D][esp-idf:000][wifi]: 
[18:01:58]
[18:01:59]E (21342) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[18:01:59]E (21342) task_wdt:  - loopTask (CPU 0)
[18:01:59]E (21342) task_wdt: Tasks currently running:
[18:01:59]E (21342) task_wdt: CPU 0: IDLE0
[18:01:59]E (21342) task_wdt: CPU 1: IDLE1
[18:01:59]E (21342) task_wdt: Aborting.
[18:01:59]E (21342) task_wdt: Print CPU 0 (current core) backtrace
[18:01:59]
[18:01:59]
[18:01:59]
[18:01:59]
[18:01:59]Backtrace: 0x40383577:0x3fcaffc0 0x420b2d31:0x3fcaffe0 0x4038717b:0x3fcb0000
[18:01:59]
[18:01:59]
[18:01:59]
[18:01:59]
[18:01:59]ELF file SHA256: c77aced757c7321d
[18:01:59]
[18:01:59]Rebooting...
```

aaron-baff-ad-net, 2024-12-26:
I'm getting the exactly same thing. I'm using a discrete IOT VLAN/WiFi, which has no Internet access. Successfully got it added (after doing `esptool erase_flash` and then re-flashing with `ha-voice-8mb-esp32s3.factory.bin`. Tried with both 24.12.8, and now 24.12.9 just today. I was logging from the USB Serial port, from the first boot, adding to WiFi, then clicking the Setup Voice Assistant on the ESPHome Device and subsequent start of boot loop.

[voice-pe.log](https://github.com/user-attachments/files/18256057/voice-pe.log)

Drizzt321, 2024-12-27:
I'm getting the exactly same thing. I'm using a discrete IOT VLAN/WiFi, which has no Internet access. Successfully got it added (after doing esptool erase_flash and then re-flashing with ha-voice-8mb-esp32s3.factory.bin. Tried with both 24.12.8, and now 24.12.9 just today. I was logging from the USB Serial port, from the first boot, adding to WiFi, then clicking the Setup Voice Assistant on the ESPHome Device and subsequent start of boot loop.

Deleted and re-added to be my personal account, not my work account, oops.

[voice-pe.log](https://github.com/user-attachments/files/18256467/voice-pe.log)

Bre77, 2024-12-27:
I am getting the same thing, seperate IOT VLAN except it has full internet access, and plently of other ESPHome devices on it. Moving my Voice PE to the same subnet as my Home Assistant instance has fixed the issue.

Drizzt321, 2024-12-27:
My HA instance is dual-homed, a VM on FreeBSD, with 2 virtual NICs, one on my Public, and one on my IOT VLANs, so there's no particular reason why there would be an issue. Both have the same IPs, so not routed.

eastman-tech, 2024-12-27:
I am experiencing the same issue (specifically task watchdog trigged/forced repeating reboot). I do have vlans in my network, but they are unrelated and HA and the Voice PE device are on the same vlan.

Drizzt321, 2024-12-27:
@eastman-tech is the VLAN that HA & Voice PE are on able to connect to the internet or not?

Drizzt321, 2024-12-27:
I actually just had a thought, when starting the Voice Setup, I see the UI in HA saying something about checking for updates, and I wonder if something in that, with the Voice PE firmware, since in my case it can't connect to the internet, even though HA can, but on a different NIC than the Voice PE, unable to connect to the internet is why Voice PE is bootlooping?

eastman-tech, 2024-12-27:
> @eastman-tech is the VLAN that HA & Voice PE are on able to connect to the internet or not?

Yes, but I do have a pihole and Unifi gear with some custom rules.

Drizzt321, 2024-12-27:
Hmph. I wonder if, despite the 24.12.8 supposedly fixing it (according to the email I got back from support), it still expects to be able to talk to the Internet, rather than HA talking to the internet, and the Voice PE not, officially, needing to since it's supposed to be able to do everything locally via HA.

jackwilsdon, 2024-12-27:
Update checks (and updating from the official releases) are performed by the Voice PE itself, so it needs internet for that.

Drizzt321, 2024-12-27:
@jackwilsdon I thought part of the point of Voice PE was being able to be completely offline, including HA. I'll definitely agree that any updates would necessarily need access to the internet somehow, although I would imagine that manual updates would be possible if someone was so drive to download elsewhere and copy the files over, or that it could be updated from HA, rather than itself.

I would still contend that lack of internet shouldn't cause a crash/bootloop however.

eastman-tech, 2024-12-28:
It might have been an issue with my piholes. Rebooted and patched them, and I'm not experiencing the issue anymore currently. Will report back if the issue returns.

I concur though that I'd love for those of us trying to run as much as possibly locally, including not bootlooping if it thinks the internet is down for some reason.

Drizzt321, 2024-12-28:
Because, well, the internet can be down sometimes anyways!

jf-uu, 2024-12-28:
I think it's pretty clear that it isn't intended for it to crash without internet - there's a fix open on the esphome repo already.

The Voice PE was never marketed as an offline device from what I can tell - it's marketed as a privacy focused device. It is meant to work without internet, you just won't get updates.

I agree it would be nice if it didn't require internet for updates. You could take control of the device in ESPHome Device Builder, which would allow pushing updates (but it seems like it's not recommended if you aren't making changes).

Drizzt321, 2024-12-28:
> The Voice PE was never marketed as an offline device from what I can tell - it's marketed as a privacy focused device. It is meant to work without internet, you just won't get updates.
>
> I agree it [...]

The way I read https://voice-pe.home-assistant.io/guides/update/ is that HomeAssistant was the one that checked for updates, not the device itself.

And on it's main page https://www.home-assistant.io/voice-pe/

```
Let’s make open, local, and private voice assistants a reality
```

Local, to me, means offline, with some which way to update it as needed. 

Under `Can I run this voice assistant fully local?`
```
Yes, provided your language is supported and you have hardware powerful enough to run local text-to-speech and speech-to-text models at a speed that is acceptable to you. Speech-to-text is the main limiting factor for many languages to run locally, as it has mixed results and often requires powerful hardware.
```

Which is talking about the host (aka HA) hardware requirements, but "Yes", to me, is not needing to access the Internet. Other than to install any necessary HA Add-Ons or the like of course. 

So I suppose it's a communications/expectations failure/misalignment. I expected updates to come through the HA instance that it was connected to, and otherwise never need the internet. Unless I wanted to use the HA Cloud Voice part, in which case that would be done through the HA instance that the device was connected.

Drizzt321, 2024-12-28:
Hopefully the potential bugfix in upstream ESPHome gets pulled in quickly, and a new Voice PE firmware version gets built soon.

latargere, 2024-12-30:
Same issue on my side, any update for the fix?

aaron-baff-ad-net, 2024-12-30:
> Same issue on my side, any update for the fix?

The linked ticket https://github.com/esphome/esphome/pull/8018 in ESPHome hasn't been merged yet. Seems ready to go, but with the holidays and all it's not surprising that it hasn't been finished off and merged. Wish it would be, but it is what it is.

Drizzt321, 2024-12-30:
> Same issue on my side, any update for the fix?

The linked ticket https://github.com/esphome/esphome/pull/8018 in ESPHome hasn't been merged yet. Seems ready to go, but with the holidays and all it's not surprising that it hasn't been finished off and merged. Wish it would be, but it is what it is.

closed 2025-01-02 by balloob

Drizzt321, 2025-01-09:
Confirming this works fine after the new firmware.