esphome/home-assistant-voice-pe #242: Feature Request: Support for "Alexa" wake word

AlexanderGlogger, 2024-12-24:
I noticed that micro-wake-word has support for "Alexa" as a wake word, but it's not available in home-assistant-voice-pe. Would it be possible to add this as an option? Thanks for considering!

synesthesiam, 2024-12-25:
For potential legal reasons, I don't think it would be a good idea to include this in the firmware.

However, it is possible for users to modify the ESPHome YAML and add external wake words via URL. See this for an example: https://github.com/esphome/micro-wake-word-models/blob/main/models%2Fv2%2Fexperiments%2FREADME.md

AlexanderGlogger, 2024-12-25:
Thank you for the clarification and the hint!

Quick follow-up question: If I modify the YAML to add custom wake words, would I still be able to receive the automatic firmware updates somehow?

pove, 2024-12-30:
Reviewing docs, I found this:

https://voice-pe.home-assistant.io/guides/grove_port/#connecting-a-sensor-to-the-grove-port

https://esphome.io/guides/made_for_esphome.html
`
All components/platforms used must have an id specified so users can easily refer to, Extend and/or Remove configuration variables should they choose to “take control”
`
https://esphome.io/components/packages#config-packages-extend

We should be able to extend the configuration using the id "mww". I don't know exactly how as I don't have any "made for esphome" device yet and my Voice PE is expected on late Jan, but I think this is the way to keep firmware with last updates but with some customisations.

Drizzt321, 2024-12-31:
Is there a reason the wake word is baked into the yaml? Don't know enough about ESPHome internals to know if there's a configuration setting capability, but it'd be pretty great to be able to set a custom wake word or other similar settings via HA UI, pushing the configuration to it, rather than having to rebuild the firmware.

pove, 2024-12-31:
> Is there a reason the wake word is baked into the yaml? Don't know enough about ESPHome internals to know if there's a configuration setting capability, but it'd be pretty great to be able to set a [...]

The firmware is built using ESPHome, that's why is using yaml, and that's also the beauty of this thing, you can take control of the firmware with a few clicks and customise it just touching a yaml file. No need to learn C++.

There are 3 micro wake words models configured in the yaml (4 if we count "stop" word), and you can choose between them in HA UI. And these models reference to the ones trained in the microWakeWord project. Adding new custom micro wake word is not that trivial, because you have to train the model, perform some fine tuning and other operations I don't even know. Maybe in the future they create an easy way for us to create new micro wake words, the same way they did it for regular wake words (micro wake words are different from regular wake words, because they have to run in tiny hardware locally).

Yo can learn more about the project here:
https://github.com/kahrendt/microWakeWord

lweberru, 2025-01-09:
At least it should be synched with the HA setting of the WakeWord. Right now, there is a list within HA and another within the device itself. Both lists have different options (the device has less).

jf-uu, 2025-01-09:
@lweberru what list are you referring to in HA? The only wake word option I have is the one exposed on the Voice PE which has Hey Jarvis, Hey Mycroft and Okay Nabu.

lweberru, 2025-01-09:
In the pipeline config:
[image]

jf-uu, 2025-01-09:
That's for devices using audio streaming for wake word detection, which the Voice PE doesn't use (it uses microWakeWord locally on the device). There should probably be an indication somewhere that the Voice PE doesn't use that setting, as the models are different.

AlexGustafsson, 2025-01-10:
@pove It's probably the notebook referenced here: https://www.home-assistant.io/voice_control/create_wake_word/#creating-your-own-wake-word.

You basically enter the wake word you want, press enter three times, wait 30 minutes and then you get the tflite and json files out. There's of course some trial and error and it could be hit and miss, but it's approaching being trivial.

Are all wake words stored in firmware immediately? I've seen a manifest file referencing the wake words to use. Is that fetched during build? It would be really nice to be able to configure the wake word OTA without a firmware rebuild.

pove, 2025-01-10:
> @pove It's probably the notebook referenced here: https://www.home-assistant.io/voice_control/create_wake_word/#creating-your-own-wake-word.
> 

Nope, that's "openWakeWord", not for "microWakeWord". In the microWakeWord GitHub link I posted, you can see the notebooks folder, with something similar to the one you linked. But they are totally different, and you will see in the microWakeWord notebook comments that it's not trivial.

AlexGustafsson, 2025-01-11:
> Nope, that's "openWakeWord", not for "microWakeWord"

Thanks for clarifying that. Sorry for the confusion.

jlpouffier, 2025-01-17:
All seems clarified, the initial issue will be not fixed, as Mike pointed, due to legal reason. (Alexa being Amazon's property)

closed 2025-01-17 by jlpouffier