esphome/home-assistant-voice-pe #246: I2S APIs has been deprecated

VACIndustries, 2024-12-26:
Received two PEs; setup in earlier today (maybe 8 hours ago?) and the firmware update went fine.  Just tried to setup the second one and now I'm receiving this:

```
INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/home-assistant-voice-0916d9.yaml...
INFO Updating https://github.com/esphome/home-assistant-voice-pe@dev
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO45 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Updating https://github.com/espressif/esp-tflite-micro@v1.3.1
INFO Compiling app...
Processing home-assistant-voice-0916d9 (board: esp32-s3-devkitc-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - framework-espidf @ 3.50105.0 (5.1.5) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- esp-audio-libs @ 1.0.0
|-- noise-c @ 0.1.6
|-- ESPMicroSpeechFeatures @ 1.1.0
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/i2c/i2c.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/i2s_audio/i2s_audio.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/light/addressable_light.cpp.o
In file included from src/esphome/components/i2s_audio/i2s_audio.h:5,
                 from src/esphome/components/i2s_audio/i2s_audio.cpp:1:
/data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/light/automation.cpp.o
Compiling .pioenvs/home-assistant-voice-0916d9/src/esphome/components/light/esp_color_correction.cpp.o
In file included from src/esphome/components/i2s_audio/speaker/../i2s_audio.h:5,
                 from src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.h:5,
                 from src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.cpp:1:
/data/cache/platformio/packages/framework-espidf@src-6e6cd5244a43bdfe453daac50cfe73b6/components/driver/deprecated/driver/i2s.h:27:2: warning: #warning "This set of I2S APIs has been deprecated, please include 'driver/i2s_std.h', 'driver/i2s_pdm.h' or 'driver/i2s_tdm.h' instead. if you want to keep using the old APIs and ignore this warning, you can enable 'Suppress leagcy driver deprecated warning' option under 'I2S Configuration' menu in Kconfig" [-Wcpp]
   27 | #warning "This set of I2S APIs has been deprecated, \
      |  ^~~~~~~
In file included from src/esphome/components/micro_wake_word/micro_wake_word.h:6,
                 from src/esphome/components/voice_assistant/voice_assistant.h:16,
                 from src/esphome/components/api/api_connection.cpp:22:
src/esphome/components/micro_wake_word/streaming_model.h:9:10: fatal error: tensorflow/lite/core/c/common.h: No such file or directory
    9 | #include <tensorflow/lite/core/c/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/home-assistant-voice-0916d9/src/esphome/components/api/api_connection.cpp.o] Error 1
========================== [FAILED] Took 5.03 seconds ==========================

```

kahrendt, 2024-12-26:
The I2S deprecation is just a warning, the code itself runs fine.

I'm not sure why you are getting this error, as it seems to be missing the TensorFlow libraries, but they should automatically be added by ESPHome. How are you running the ESPHome device builder?

In general, you do not need to take control of the Voice PE firmware unless you are modifying it. You should be able to add, use, and update the device in HA without needing the ESPHome device builder.

VACIndustries, 2024-12-26:
Correct! That's why I was taking control/updating before replicating the repo and making some adjustments -- just running the ESPHome Device Builder add on off of my HAOS x86 laptop

VACIndustries, 2024-12-30:
Problem went away "magically" (I assume a backend dependency issue?).  Closing as a result

closed 2024-12-30 by VACIndustries