esphome/home-assistant-voice-pe #248: Question about the Bluetooth function

Smarthome-Creator, 2024-12-27:
Can the Bluetooth be used for more than just setup? I have been trying to use a room presence in HA for a very long time now, but unfortunately I always fail due to undeveloped integrations such as “room-assistant” or “ESP32-mqtt-room”.

And now I was wondering if it is possible to use the Assist satellite with “ESPresence” to build a room presence?

Drizzt321, 2024-12-28:
Well, there's a Grove connector, so if you add a Grove connected presence sensor, you probably can end up with a room presence sensor connected. See 

https://voice-pe.home-assistant.io/guides/grove_port/

However, then ESPHomeBuilder does any/all updates, rather than coming separately. As per https://voice-pe.home-assistant.io/guides/update/  it's recommended to not use ESPHomeBuilder, although I suppose if want to add a Grove sensor, you must use ESPHomeBuilder, and so that's how you'd get updates from then on.

tbrasser, 2024-12-28:
Not sure hardware is capable enough to function normally when also adding bluetooth proxy. But you can try and use bermuda custom integration.

"taking control" of the firmware via esphome device builder is a prerequisite for any firmware customizations as mentioned above

jlpouffier, 2025-01-17:
We will not add Bluetooth proxy capabilities in the official firmware because we want to keep the headroom available on the chip for future voice features.
However I just want to point out that it is indeed working today if you take control.

But you will have to do the change on your side.

Please note that Bluetooth proxy and voice assistant are two of the heaviest tasks we can run on a ESP32 chip, and mixing them is probably a good recipe for instabilities.

closed 2025-01-17 by jlpouffier