esphome/home-assistant-voice-pe #250: Can't join WiFI with WPA3 enabled

lordexorr, 2024-12-28:
It seems the Voice PE devices are unable to connect to WiFi networks with WPA3 enabled. I was unable to get either of my 2 devices to connect to my network. As a test I created a guest network with WPA2 only and was able to connect the devices to WiFi.

My network also has Management Frame Protection enabled; but I'm not sure this is related. I suspect it's the WPA3 item, but wanted to mention it just in case.

I was running the Firmware Shipped with the devices at the time; I have since upgraded to the latest (24.12.8).

The thread below shows another user had the same issue.

https://community.home-assistant.io/t/voice-pe-unable-to-connect-to-wifi/817454/8

Drizzt321, 2024-12-28:
Is your WiFi WPA3 _only_, or does it allow mixed WPA3 & WPA2 devices?

I think they're at the mercy of ESP WiFi hardware, which probably only supports WPA2.

lordexorr, 2024-12-28:
It allows mixed; WPA2/3 as well as using MLO to combine 2.4/5/6GHz bands in one. If it doesn’t support some of these types of networks or frequencies that should be documented and noted prior to purchasing.

Drizzt321, 2024-12-28:
Hm, what channel width on 2.4GHz are you running with?

So the [specs](https://www.home-assistant.io/voice-pe/) specifically say 2.4GHz WiFi, and says ESP32-S3. So easy to find the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf). Which for WiFi specs/MAC says, most important bits:

> IEEE 802.11b/g/n-compliant
> Supports 20 MHz and 40 MHz bandwidth in 2.4 GHz band
> 1T1R mode with data rate up to 150 Mbps
> ...
> GCMP, CCMP, TKIP, WAPI, WEP, BIP, WPA2-PSK/WPA2-Enterprise, and WPA3-PSK/WPA3-Enterprise

So make sure your WiFi settings would on 2.4GHz would support it, allow older N clients, and only 20 or 40 MHz channels.

lordexorr, 2024-12-29:
The specs say 2.4GHz WiFi and my network supports 2.4GHz WiFi. I have many other smart home devices that support only 2.4GHz devices and have no issues joining my network (the first one below). Below are screenshots of the network configurations.

Network that does NOT work;
[image]

Network that does work;
[image]

What aspect of the first network is preventing it from working? It still seems like a bug to me which is why I'm posting it here.

Drizzt321, 2024-12-29:
Do you know if any of those other devices are ESP chips? If so then it's something with this particular firmware.

I'm unfamiliar with whatever AP/router software that those are screenshots of, but what's the channel width on whichever WiFi that it's unable to connect to? Specifically on the 2.4GHz network. It can only handle 20 or 40 MHz channel width. Which for 2.4GHz, unless you live way out away from other folks, the 20MHz channels are better, and choose one of 1, 6, or 11.

lordexorr, 2024-12-29:
It's not something I can specifically set when using WiFi 7/MLO. However, it supports below. Any device that requires any of these should be supported and should be able to join the network without me needing to configure anything else.

2.4GHz BE: 4x4 (Tx/Rx) 4096 QAM 20/40MHz, up to 1376Mbps
5GHz BE: 4x4 (Tx/Rx) 4096 QAM 20/40/80/160MHz, up to 5764Mbp
6G-1Hz BE: 4x4 (Tx/Rx) 4096 QAM 20/40/80/160/320MHz, up to 11529Mbp
6G-2Hz BE: 4x4 (Tx/Rx) 4096 QAM 20/40/80/160/320MHz, up to 11529Mbp

I do not know if anything else uses ESP chips; that's above my knowledge :)

Drizzt321, 2024-12-30:
Hm, so I see there's a WiFi7 checkbox. Does that mean it's WiFi7 only? Or just WiFi7 with WiFi7 clients? Really not sure what's going on here. Maybe someone more familiar with WiFi7, or WiFi in general, could help troubleshoot what might be the issue? I still suspect it's something to do with the modes that it's in that isn't compatible with the underlying ESP32 hardware.

lordexorr, 2024-12-30:
Ok; its not the ESP32 hardware. I ordered 2 ratgdo32's for my garage doors and set them up today. They had zero issues connecting to my WiFi.

aaron-baff-ad-net, 2024-12-30:
That's a great sample, really not sure what's going on then.

Drizzt321, 2024-12-30:
That's a great sample, not sure what's going on then.

kbx81, 2025-01-09:
FWIW, all of my ESPHome devices are connecting to an SSID shared among 2.4, 5 and 6 GHz bands and with only WPA3 enabled. I have no issues with it at all and there's nothing special in my ESPHome configs.

sokyrko, 2025-02-24:
@lordexorr looking on you screenshots, I have the same or similar router. I tried wpa2 network too, but it did not work. However, disabling `Roaming assistant` feature on 2.4Ghz band helped, not a single disconnect since then.

lordexorr, 2025-04-05:
I decided to reset these to factory defaults today and add again. This time it joined my normal network with no issues. A firmware update must've resolved this issue.

closed 2025-04-05 by lordexorr