esphome/home-assistant-voice-pe #252: Voice Assistant still responses with Timers which have been just canceled

HowlingCoder, 2024-12-29:
Hey there :)!
I'm currently tinkering with the Timers of Voice PE and my goal would be to control them directly from HomeAssistant. Therefore I created the following service:

```yaml
 services:
    - service: stop_timer
      variables:
        timer_id: string
      then:
        - lambda:  |-
            api::VoiceAssistantTimerEventResponse msg;
            msg.timer_id = timer_id;
            msg.event_type = api::enums::VOICE_ASSISTANT_TIMER_CANCELLED;
            msg.total_seconds = 0;
            msg.seconds_left = 0;
            msg.is_active = false;
            id(va).on_timer_event(msg);

            ESP_LOGW("main", "Timer Active %d",id(is_timer_active)); // Always shows 0

            if (id(is_timer_active)){
              id(fetch_first_active_timer).execute();
            } else {
              id(first_active_timer) = {
                  .id = "invalid",
                  .name = "",
                  .total_seconds = 0,
                  .seconds_left = 0,
                  .is_active = false,
              };
              
            }
```

Everything seems to work, but the **Assist seems to be not in sync with the data**.

**Things I checked:**
1) Everything gets called, timers get removed from the timers_ map
2) LEDs stop and no alarm occur, the Assistant will insist on the timer existing until the timer normally would've ended
3) I forked the repository with additional logging messages to make sure I'm not insane

```bash
[18:40:16][D][voice_assistant:860]: Before cancellation - Number of timers: 1
[18:40:16][D][voice_assistant:861]: Attempting to cancel timer with ID: '01JG9R22J15SQQQ2A8GFRFC00M'
[18:40:16][D][voice_assistant:865]: Existing timer ID: '01JG9R22J15SQQQ2A8GFRFC00M'
[18:40:16][D][light:036]: 'voice_assistant_leds' Setting:
[18:40:16][D][light:047]:   State: OFF
[18:40:16][D][light:109]:   Effect: 'None'
[18:40:16][D][voice_assistant:871]: After cancellation - Number of timers: 0
[18:40:16][D][voice_assistant:882]: Are the timer ticks running?: 0
[18:40:16][W][main:069]: Timer Active 0
...
[18:40:28][D][voice_assistant:669]: Speech recognised as: "Läuft ein Timer?" # (Is a timer running?)
[18:40:34][D][voice_assistant:697]: Response: "Ja, ein Timer läuft noch. Er hat noch 2 Minuten und 39 Sekunden übrig.  # (Yes, 2:39 remaining)"
```

I'm not sure if this is a bug or if I'm just missing something. I really appreciate your time and effort. Thank you!
Max

HowlingCoder, 2025-01-21:
I've taken the whole thing a bit further. For me it seems that there is dessence between timers created by script and by voice in general. Could this be? 

| Action A  | Action B | State Timer Array | Result |
| ------------- | ------------- |------------- |------------- |
| Create Timer via Script  |  | Timer created, counting down  | No ringing, LED updates initially, Circle doesn't count down |
| Create Timer via Voice  |  |Timer created, counting down | LED works as intended Ringing | 
| Create Timer via Script | Stop/ Pause Timer via Script | Timer paused/ removed from Array | (Since Creation didn't work in the first place, stopping "works")
| Create Timer via Voice | Stop/ Pause Timer via Script | Timer paused/ removed from Array | After timer would've initially run out, Alarm sounds (although visible paused, deleted) |
| Create Timer via Voice | Stop/ Pause Timer via Voice | Timer paused/ removed from Array | Works|
| Create Timer via Voice | Asking Assist for Timer Count | 1 Timer in Array| Response: 1|
| Create Timer via Script| Asking Assist for Timer Count | 1 Timer in Array| Response: 0|

This is basically what I'm using:
https://gist.github.com/HowlingCoder/05484c247e52be96a9462d6b3f714bb4

For me it seems that there are two versions running which are not in sync but I don't understand why.

ronmichel, 2025-04-11:
Dear @HowlingCoder,
any update on this?
Thanks

HowlingCoder, 2025-04-21:
@ronmichel Nope, I gave up on the issue. I also asked around in the discord and the last thing I heard was that the internal HA team works on the timers - this was in January though. :/