esphome/home-assistant-voice-pe #253: stops responding to wake word

bbrendon, 2024-12-29:
The button on top works to activate and recognize commands but wake word stops work. Reboot fixes it for a few hours. So far I've been rebooting 3-5 times per day. 

I have two units, and it happens to both of them.

RichardMawdsley, 2025-01-03:
+1 for me too. I have two of these but it only happens to one of them.

If I change a setting on the device such as the wake word or anything, then it usually kicks it into gear and it starts listening again. It's as if it goes to sleep or something.

RichardMawdsley, 2025-01-09:
> "Hey Jarvis" does not work well. Only one or two tries, then it needs a reboot. If you switch to "Okay Nabu" it works quite stable for me.

I've got two and it's absolutley fine on one, and has this issue on the other. Not convinced the wake word is the issue.

bbrendon, 2025-01-10:
I'm starting to try ok nabu and it seems to be better. Jarvis is atrocious.

jlpouffier, 2025-01-17:
Do you have logs of pressing the button and not being able to use the wake words afterwards please?

bbrendon, 2025-01-19:
@jlpouffier I don't think anyone here has that issue.

RichardMawdsley, 2025-01-20:
I've pretty much narrowed this down to 'Hey Jarvis', but only on one of my units.  If I leave it set as Hey Nabu or Hey Mycroft, it's absolutely fine and always responds.  Leave it on Hey Jarvis, it becomes intermittent after a 'while' (anywhere from hours to days).

Again, this only happens on one of my two units.  The other, set to Hey Jarvis, is absolutely perfect.

Unsure how to troubleshoot this further, if there's any particular information you need please let me know.

jlpouffier, 2025-01-22:
> [@jlpouffier](https://github.com/jlpouffier) I don't think anyone here has that issue.

Then what is the issue exactly.
> The button on top works to activate and recognize commands but it stops recognizing the wake word.

For what I understand, if you use the center button, then the wake word does not work afterwards?

RocketRammer, 2025-01-22:
> Then what is the issue exactly.
>
>
> For what I understand, if you use the center button, then the wake word does not work afterwards?

The issue is the 'Hey Jarvis' wake word stops responding.  Pressing the center button works as expected and then afterwards, Hey Jarvis responds absolutely fine.. until it doesn't again.

It's as if its going to sleep on it...  though I say that when actually some of the time it seems like it's not listening at its maximum ability.  If I get completely, lip to lip close to it and 'Hey Jarvis' it, it works... and then continues to work from normal distances.

I completely understand I'm giving very odd circumstances, just trying to report what I'm finding with this one unit.  If there's anything specific I can provide please let me know.

jlpouffier, 2025-01-22:
If I read between the lines, it's just a problem of difficulties triggering a wake word.
The center button has nothing to do with this, right?

RocketRammer, 2025-01-22:
Correct.  There is no fault with the center button.  That works every time.

jlpouffier, 2025-01-23:
OK, then this is, unfortunately, expected.
We're working hard on improving the wake words, but you need to understand that there is no shortcut on this process, unfortunately.

Wake words are initially trained on synthetic samples of the wake word generated from Piper.
What is skyrocketing the quality of our wake word is to fine-tune it with real data samples.

This is why we opened the Wake Word Collective, you can read more about it on [our blog](https://www.home-assistant.io/blog/2024/10/24/wake-word-collective/)

As of today, we are working on improving the quality of `OK Nabu`.
We may open the wake word collective to other wake words once we are happy with the quality of `OK Nabu`.

I suggest participating in the wake word collective, and, again, unfortunately, being patient.

Again, there are no shortcuts when creating good quality wake words, it's a huge undertaking.

Thx for your understanding!

Closing this issue for now.

JLo

closed 2025-01-23 by jlpouffier

RocketRammer, 2025-01-23:
If only people wanted 'OK Nabu'...

bbrendon, 2025-01-24:
I would suggest removing Jarvis until it’s at least usable.