esphome/home-assistant-voice-pe #255: Poor WiFi, drops outs, time outs, turning on LED in HA takes 15-60 seconds.

cooljimy84, 2024-12-30:
Not sure what you would need for this one. I've got a Unfi AP, with a mix of ESP32 and ESP8266 devices that ping perfectly, tho they are all setup from ESPhome yaml files i have created (i think i turned off powersave on the wifi)

ping 192.168.1.165 -c 10
PING 192.168.1.165 (192.168.1.165) 56(84) bytes of data.
64 bytes from 192.168.1.165: icmp_seq=1 ttl=64 time=16984 ms
64 bytes from 192.168.1.165: icmp_seq=2 ttl=64 time=15927 ms
64 bytes from 192.168.1.165: icmp_seq=3 ttl=64 time=14903 ms
64 bytes from 192.168.1.165: icmp_seq=4 ttl=64 time=13879 ms
64 bytes from 192.168.1.165: icmp_seq=5 ttl=64 time=12855 ms
64 bytes from 192.168.1.165: icmp_seq=6 ttl=64 time=11831 ms
64 bytes from 192.168.1.165: icmp_seq=7 ttl=64 time=10807 ms
64 bytes from 192.168.1.165: icmp_seq=8 ttl=64 time=9783 ms
64 bytes from 192.168.1.165: icmp_seq=9 ttl=64 time=8759 ms

--- 192.168.1.165 ping statistics ---
10 packets transmitted, 9 received, 10% packet loss, time 9249ms
rtt min/avg/max/mdev = 8758.945/12858.740/16984.322/2649.692 ms, pipe 10

If i play with the volume up/down on the device it pings all the time i'm doing that.

cooljimy84, 2024-12-30:
copying the yaml from here and then adding the below the device pings all the time and is near instant when doing things.

```
  power_save_mode: none
  fast_connect: true
```

Drizzt321, 2024-12-31:
Does your WiFi access point report the connected rates & connection sensitivity for the device? Or can you visit the built-in HTTP on the device, and see what it reports as it's connected WiFi reception? Or is that visible in HA at all?

jdohnstad, 2024-12-31:
I had a similar experience with mine - it would process voice commands quite quickly (regardless of HA Cloud or local agents). At first I thought it was the text-to-speech (since it would then sit a while before speaking a response) - but then I noticed that simply toggling the LED light ring would take a while and sometimes not even work. I also have a Unifi AP and also have tons of ESPHome devices that work just fine. I added the snippet from above and things seemed to be much better.

cooljimy84, 2024-12-31:
> Does your WiFi access point report the connected rates & connection sensitivity for the device? Or can you visit the built-in HTTP on the device, and see what it reports as it's connected WiFi reception? Or is that visible in HA at all?

I can post info from the original firmware and with changes. I'll see if I can get it done today (as I'm sure after the holiday break there are gonna be more people with these devices)

cooljimy84, 2024-12-31:
With wifi tweaks in ESP yaml
Ch. 6 (2.4 GHz, 20 MHz) 1x1  WiFi 4 -51 dBm
TX Retries0.0%
AP/Client Signal BalanceGood
StandardWiFi 4
Tx Rate58.0 Mbps
Rx Rate65.0 Mbps

OG from chrome flasher (issue was still happening at this point)
Ch. 6 (2.4 GHz, 20 MHz) 1x1 WiFi 4 -51 dBm
TX Retries0.0%
AP/Client Signal BalanceGood
StandardWiFi 4
Tx Rate65.0 Mbps
Rx Rate52.0 Mbps

whats strange is that at first it was having the same issue on the OG firmware from the chrome web browser flasher, after clicking "reset device" to try and get some cleaner logs it pings/behaves fine.

OG from chrome flasher (issue no longer happening)
Ch. 6 (2.4 GHz, 20 MHz) 1x1 WiFi 4 -54 dBm
TX Retries0.0%
AP/Client Signal BalanceGood
StandardWiFi 4
Tx Rate65.0 Mbps
Rx Rate52.0 Mbps

haginator, 2025-01-03:
I'm experiencing a similar issue with my Voice PE. I have tested both firmware version 24.10.15 and the latest manually installed version (Firmware: 25.1.0 (ESPHome 2024.12.2)) from the [official site ](https://esphome.github.io/home-assistant-voice-pe/).

The problem arises during audio playback. The playback stops after about 30-40 seconds with a brief hiccup, displaying the error message "The media pipeline's file reader encountered an error." in the ESP console. I have tested multiple sources, including Music Assistant, external MP3 files, and TTS, and all function correctly with other media players on my network. However, with TTS, long responses do not play at all, while shorter ones work perfectly. The Voice PE responds quickly on my end.

Here are two examples of playback failures:

Music Assistant & External MP3 File : Stops after a few seconds.
```

[17:58:41][D][media_player:080]: 'Media Player' - Setting
[17:58:41][D][media_player:087]:   Media URL: http://192.168.10.8:8097/flow/media_player.home_assistant_voice_092295_media_player/6fac555f15844e36bf8579683e70c09a.flac?ts=1735923521
[17:58:41][D][nabu_media_player.pipeline:173]: Reading FLAC file type
[17:58:43][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[17:59:27][E][nabu_media_player:297]: The media pipeline's file reader encountered an error.
[18:01:40][D][media_player:080]: 'Media Player' - Setting
[18:01:40][D][media_player:087]:   Media URL: http://192.168.10.8:8123/api/esphome/ffmpeg_proxy/949340ec89ebf409682a7bb60bb6c7f8/ysFKDDARw1a9hJPglhqzYg.flac
[18:01:40][D][nabu_media_player.pipeline:173]: Reading FLAC file type
[18:01:40][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[18:02:30][E][nabu_media_player:297]: The media pipeline's file reader encountered an error.
```
TTS (Text-to-Speech) : Long TTS responses do not play fully or at all.

```
[18:16:15][D][voice_assistant:641]: Event Type: 7
[18:16:15][D][voice_assistant:697]: Response: "Es war einmal in der pulsierenden Stadt Hamburg, wo der junge Olaf Scholz aufwuchs. Schon in seiner Kindheit zeigte er ein großes Interesse an Politik und sozialer Gerechtigkeit. In der Schule engagierte er sich in verschiedenen Projekten und trat der Schülervertretung bei, wo er lernte, die Stimmen seiner Mitschüler zu hören und ihre Anliegen zu vertreten.
[18:16:15]
[18:16:15]Nach dem Abitur entschied sich Olaf, Rechtswissenschaften zu studieren. Seine Leidenschaft für die Po
[18:16:15][D][light:036]: 'voice_assistant_leds' Setting:
[18:16:15][D][light:051]:   Brightness: 66%
[18:16:15][D][light:109]:   Effect: 'Replying'
[18:16:15][D][voice_assistant:641]: Event Type: 8
[18:16:15][D][voice_assistant:719]: Response URL: "http://192.168.10.8:8123/api/tts_proxy/hnWkYqVz89ZlOj8YWJHAyg.flac"
[18:16:15][D][voice_assistant:515]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[18:16:15][D][voice_assistant:522]: Desired state set to STREAMING_RESPONSE
[18:16:15][D][media_player:080]: 'Media Player' - Setting
[18:16:15][D][media_player:087]:   Media URL: http://192.168.10.8:8123/api/tts_proxy/hnWkYqVz89ZlOj8YWJHAyg.flac
[18:16:15][D][media_player:093]:  Announcement: yes
[18:16:15][D][power_supply:033]: Enabling power supply.
[18:16:15][D][voice_assistant:641]: Event Type: 2
[18:16:15][D][voice_assistant:733]: Assist Pipeline ended
[18:16:20][D][esp-idf:000][ann_read]: W (1839751) HTTP_CLIENT: Connection timed out before data was ready!
[18:16:20]
[18:16:20][D][nabu_media_player.pipeline:173]: Reading FLAC file type
[18:16:20][D][voice_assistant:515]: State changed from STREAMING_RESPONSE to IDLE
[18:16:20][D][voice_assistant:522]: Desired state set to IDLE
```
For reference regarding connection quality, I have observed good Wi-Fi performance (-49 dB) and minimal ping issues:

```
PING 192.168.20.59 (192.168.20.59): 56 data bytes
64 bytes from 192.168.20.59: icmp_seq=0 ttl=63 time=102.758 ms
64 bytes from 192.168.20.59: icmp_seq=1 ttl=63 time=17.109 ms
64 bytes from 192.168.20.59: icmp_seq=2 ttl=63 time=38.218 ms
64 bytes from 192.168.20.59: icmp_seq=3 ttl=63 time=59.010 ms
64 bytes from 192.168.20.59: icmp_seq=4 ttl=63 time=75.650 ms
64 bytes from 192.168.20.59: icmp_seq=5 ttl=63 time=108.168 ms
64 bytes from 192.168.20.59: icmp_seq=6 ttl=63 time=16.855 ms
64 bytes from 192.168.20.59: icmp_seq=7 ttl=63 time=38.400 ms
64 bytes from 192.168.20.59: icmp_seq=8 ttl=63 time=59.649 ms
64 bytes from 192.168.20.59: icmp_seq=9 ttl=63 time=79.076 ms
64 bytes from 192.168.20.59: icmp_seq=10 ttl=63 time=99.795 ms
64 bytes from 192.168.20.59: icmp_seq=11 ttl=63 time=17.057 ms
64 bytes from 192.168.20.59: icmp_seq=12 ttl=63 time=39.325 ms
64 bytes from 192.168.20.59: icmp_seq=13 ttl=63 time=64.371 ms
64 bytes from 192.168.20.59: icmp_seq=14 ttl=63 time=82.828 ms
64 bytes from 192.168.20.59: icmp_seq=15 ttl=63 time=106.872 ms
64 bytes from 192.168.20.59: icmp_seq=16 ttl=63 time=24.074 ms
64 bytes from 192.168.20.59: icmp_seq=17 ttl=63 time=41.324 ms
--- 192.168.20.59 ping statistics ---
18 packets transmitted, 18 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 16.855/59.474/108.168/31.294 ms
```
Thank you for your assistance!

cooljimy84, 2025-10-01:
After messing around with compling ESPHome 25.9.0 from my own yaml with power saving disabled, then flashing the managed firmware and testing i've found a new setting in my Unifi controller,

Yes Unifi AP strikes again it seems, i ticked "Enhanced IoT Connectivity" in the radio settings for that SSID and bing, stable.
Tho readying what it does i had already done to the settings, so must be a secret settings not exposed on the GUI it sets.