esphome/home-assistant-voice-pe #256: Feature Request: Button press device triggers missing!

picotrain77, 2024-12-31:
Love the new PE, however although I normally avoid using device triggers, I note there are many device triggers listed for the PE, however no device triggers for the button press (double, triple & long).

It would be really useful to have these available for those that prefer the UI way of working

mackeapple22, 2025-01-02:
You can use the button by adding an Entity trigger and searching for the name of the button (something like event.home_assistant_voice ... button_press). Don't add the anything in the optional "Attribute", "From" or "To" fields, when I did this the automation did not trigger at all. Instead under "Then do" add a "Choose" building block, with a condition that checks the entity (event.home_assistant_voice ... button_press) for the state. Here you add the Event type attribute and the State you want to check (double_press, long_press, triple_press), then the action to perform. For example I have set a long_press to run a script that plays/pauses the music that's playing.

I too would like to have the button under Device Triggers, but I assume it being an entity makes more sense
[image]
[image]

flo269, 2025-01-08:
> however no device triggers for the button press (double, triple & long).

I had to actually double press etc. once each and after that I was able to choose the states in the UI.

badnetmask, 2025-01-08:
> > however no device triggers for the button press (double, triple & long).
> 
> I had to actually double press etc. once each and after that I was able to choose the states in the UI.

For future reference: I can here trying to find why my button wasn't available, and it indeed became available after I did a physical double press.

Nevertheless, I do think it's strange to have to create the automation based on the button as an entity in itself, when looking at other devices (like Aqara) the button is a trigger under a device.

Not sure if this is the kind of thing worth re-working though. I'll leave that to the developers.

jlpouffier, 2025-01-17:
Agreed. Buttons are complex in Home Assistant right now.
This is unfortunately a commun solution, I think we will get there on time but there is nothing we can do additionally on the Voice Preview Edition itself.

closed 2025-01-17 by jlpouffier