esphome/home-assistant-voice-pe #258: RFE: Allow custom CA and TLS certificate without rebuild

badnetmask, 2024-12-31:
I run my own CA (Certificate Authority) at home using a [Step CA Server](https://smallstep.com/docs/step-ca/). Home Assistant is exposed to my internal network using a certificate issued by that CA via a [Caddy proxy](https://caddyserver.com/docs/quick-starts/https). The Voice Assistant PE is unable to respond to my voice commands because it does not recognize the Home Assistant's certificate.

I would like to ask if it would be possible to implement some way to either disable TLS verification (although not good), allow me to accept or not a given "invalid" certificate (less bad), or to allow loading the custom CA bundle into the device (best solution). 

I know this can be done by rebuilding the the image with `verify_ssl: false`, but the intention here would be to avoid rebuilding the image, this way it would be possible to maintain the automatic OTA updates.

Here are the relevant logs, for reference:
```
[13:48:08][D][media_player:080]: 'Media Player' - Setting
[13:48:08][D][media_player:087]:   Media URL: https://<my-ha-IP-address>/api/tts_proxy/exb_9-ouOqpNL7hY-722QQ.flac
[13:48:08][D][media_player:093]:  Announcement: yes
[13:48:08][D][voice_assistant:641]: Event Type: 2
[13:48:08][D][voice_assistant:733]: Assist Pipeline ended
[13:48:08][D][esp-idf:000][ann_read]: E (2065965) esp-x509-crt-bundle: Failed to verify certificate
[13:48:08]
[13:48:08][D][esp-idf:000][ann_read]: E (2065969) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
[13:48:08]
[13:48:08][D][esp-idf:000][ann_read]: E (2065971) esp-tls: Failed to open new connection
[13:48:08]
[13:48:08][D][esp-idf:000][ann_read]: E (2065972) transport_base: Failed to open a new connection
[13:48:08]
[13:48:08][D][esp-idf:000][ann_read]: E (2065974) HTTP_CLIENT: Connection failed, sock < 0
[13:48:08]
[13:48:08][E][nabu_media_player.pipeline:171]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
[13:48:08][E][nabu_media_player:305]: The announcement pipeline's file reader encountered an error.
[13:48:08][D][voice_assistant:515]: State changed from STREAMING_RESPONSE to IDLE
```

balloob, 2024-12-31:
This is not something that we want to support in the built-in firmware. Building your own firmware can be used to work around this issue. If you run a custom CA authority, a lot of smart home devices won't work! It's not recommended.

closed 2024-12-31 by balloob

badnetmask, 2024-12-31:
While I understand there is some complexity in the implementation, having a custom CA is one piece of having a private cloud-independent infrastructure. As much as Let's Encrypt is a great tool, it still requires some level of exposure to external cloud services.

Right now my entire home, full of smart devices, is running just fine with my custom CA, except this one device.

I would kindly ask you to reconsider.

AlexGustafsson, 2025-01-09:
As a workaround these are the path prefixes you need to allow HTTP access to for Voice PE to work:

- `/api/tts_proxy/`
- `/api/assist_satellite/connection_test/`
- `/api/esphome/ffmpeg_proxy/`

Not ideal, but it works in the interim.

badnetmask, 2025-01-09:
> As a workaround these are the path prefixes you need to allow HTTP access to for Voice PE to work:
> 
> * `/api/tts_proxy/`
> * `/api/assist_satellite/connection_test/`
> * `/api/esphome/ffmpeg_proxy/`
> 
> Not ideal, but it works in the interim.

Those exceptions did work, but I had to change the URL on the network settings to "http", instead of "https", so I don't know what else could break.

AlexGustafsson, 2025-01-10:
Good point, sorry for forgetting to mention that.

Also, my reverse proxy is set up to redirect all HTTP requests not matched by the above prefixes to HTTPS. The companion apps can be configured to only ever use the HTTPS endpoints as well despite what the config says. With HSTS enabled you can also ensure that browsers won't try to connect to HTTP.

It's all just workarounds. Disabling HTTP would of course have been nice.

aliher1911, 2025-04-27:
Curious if you managed to get this resolved? I run HA on NAS which is terminating TLS and all local URL's go through https endpoint which custom certs. I don't think I can tell HA to only let voice go through special http only URL.

I tried to build it from source since esp-idf has `CONFIG_ESP_TLS_INSECURE` and `CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY` options to do exactly that, but for some reason, generated build files always have those options disabled regardless of config. I must be missing something here.

depahk, 2025-05-21:
> smart home devices won't work

Over 150 devices and the only one that doesn't work with custom CA is HA Voice Preview :/.

A lot of people use https with custom CA, it seems strange don't enable a buotton/config entry to enable it and also not to write it clearly in the requirements and configuration phase.. disappointed with this purchase