esphome/home-assistant-voice-pe #260: Radio Stream fails on Music Assistant

ramapge123, 2025-01-01:
Attempting to play a Radio stream from Music Assistant on the device however fails after a few seconds up to 30 seconds on different stations.

Radio stations work fine on other devices such as a satellite or google homes.

Is there further debugging I can do to get better logs?

[14:31:44][D][media_player:080]: 'Media Player' - Setting
[14:31:44][D][media_player:087]:   Media URL: http://192.168.1.4:8098/flow/media_player.home_assistant_voice_0908c7_media_player/3b53ed54ca384bb2a6bc20e49a379b5d.mp3?ts=1735702304
[14:31:44][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[14:31:46][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[14:31:48][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.

ramapge123, 2025-01-02:
Issue was fixed with Music Assistant Update.

kahrendt, 2025-01-03:
This should be fixed with Music Assistant 2.3.5. Previous versions were sending an mp3 stream instead of FLAC, and the mp3 decoder on the device is a bit fragile. If it still is problematic after updating, please reopen!

closed 2025-01-03 by kahrendt

marcelveldt, 2025-01-22:
One additional comment on this is that you may have to delete the player from MA and re-add it to clear any old config where the codec was enforced to MP3. Also, if you are not running the default firmware for Voice PE provided by Nabu Casa but a custom one, ensure that you do not change the model name. We currently use a [hardcoded list of model names ](https://github.com/music-assistant/server/blob/dev/music_assistant/providers/hass_players/__init__.py#L101)to detect ESPHome based media players that run the new media player component.