esphome/home-assistant-voice-pe #261: IPv6 support

mweinelt, 2025-01-02:
My home assistant instance runs on a hostname that only exposes an AAAA record. The HA Voice PE is on a VLAN that allows for stateless address autoconfiguration.

During onboarding, I get an error that no audio can be streamed from home-assistant.

Running tcpdump reveals that after DNS resolution no further traffic is sent by the device.

I wonder if [this preprocessor condition](https://github.com/esphome/home-assistant-voice-pe/blob/336063d9648bfd0c3a45c4d55606be68676d94ce/esphome/components/voice_assistant/voice_assistant.cpp#L560-L564) is what prevents that from working.

And if not, Is it possible to make this project IPv6 capable?

frebib, 2025-04-29:
I discovered today that my ESP Home Nabu device thingy doesn't support IPv6 at all (seemingly). tcpdump confirms that not even a single router solicitation or even neighbour advertisement is emitted. I'm not sure what kind of kernel runs on this device (I've done very little research) but it feels to me like the IPv6 functionality has been disabled (maybe deliberately?)
I'd be interested in trying to get this working if at all possible. Any pointers on where to start looking to enable v6 support for this device's firmware?

jesserockz, 2025-04-30:
IPv6 is disabled by default in ESPHome and requires the yaml config below to be added to enable it.

```yaml
network:
  enable_ipv6: true
```

frebib, 2025-04-30:
After quite a lot of digging through code and issues I came to that conclusion late last night. I'm shocked that it's disabled by default - it really shouldn't be (maybe this can be fixed too, afterwards).
I can send a PR for this soon and hopefully test it out too (if I can get the flashing working).

closed 2025-05-22 by jesserockz