esphome/home-assistant-voice-pe #263: Built-in speaker and 3.5mm at the same time

mackeapple22, 2025-01-02:
I would like to use the Voice PE mainly as a network media player. I have connected it using 3.5mm to an AV receiver, but once I connect it with 3.5mm it shuts off the internal speaker of the Voice PE.
I would like a choice of having Assist responses through the Voice PE's internal speaker, and music playing through my AV system (I can't get an audio response from the Voice PE if my receiver is turned off).

jackwilsdon, 2025-01-03:
It looks like the speaker muting is implemented in hardware from the [schematic](https://voice-pe.home-assistant.io/resources/home_assistant_voice_pe_schematic_v1.0_241009.pdf):

[image]

MICDET is normally connected to GND but is disconnected when a jack is inserted, causing Q2 to turn on (pulled up by R26). This pulls AMP_CTL down and turns off the amplifier for the built-in speaker. The ESP can control the amplifier with AUDIO_PA_EN, but it can't overpower Q2.

closed 2025-01-03 by mackeapple22

ItsHarper, 2025-02-25:
The ReSpeaker's headphone jack detection circuit seems to work similarly to this one (at a bare minimum in that `MICDET` is connected to ground when no plug is inserted into the jack), and their documentation explains that you can short `MICDET` to ground in order to disable detection, enabling software control. It even documents how to switch the output on the TLV320AIC3204 DAC/codec (the same one used in the Voice PE).

https://github.com/respeaker/ReSpeaker_Lite?tab=readme-ov-file#disable-the-headphone-jack-detection-circuit

Now, the ReSpeaker is physically designed so that it's very easy to short `MICDET` to ground. I don't know if the Voice PE is set up that way or not (and of course you'll have to disassemble the device regardless).

BTW, I'm sure it's not a coincidence that the circuitry is similar and that they use the same codec, as it appears that both devices were designed by Seeed Studio.

WhiteSockedDancer, 2025-03-01:
I got exactly the same usecase.

But this would just disable the detection of the jack? How would we control output with software then?

bram96, 2025-11-01:
@WhiteSockedDancer I managed to get it working with the following code in epshome

``` 
i2c_device:
  id: internal_i2c_device
  address: 0x18

button:
  - platform: template
    name: "Switch to Speaker Output"
    icon: mdi:speaker
    on_press:
      - lambda: |-
          ESP_LOGI("aic3204", "Switching to AUX (headphone) output...");

          id(internal_i2c_device).write_byte(0x00, 0x01);
          id(internal_i2c_device).write_byte(0x12, 0x00);
          id(internal_i2c_device).write_byte(0x13, 0x00);
          id(internal_i2c_device).write_byte(0x10, 0x40);
          id(internal_i2c_device).write_byte(0x11, 0x40);
          ESP_LOGI("aic3204", "AUX output enabled.");
  - platform: template
    name: "Switch to AUX Output"
    icon: mdi:audio-input-stereo-minijack
    on_press:
       - lambda: |-
          ESP_LOGI("aic3204", "Switching to Speaker output...");
          
          id(internal_i2c_device).write_byte(0x00, 0x01);

          id(internal_i2c_device).write_byte(0x10, 0x00);
          id(internal_i2c_device).write_byte(0x11, 0x00);
          id(internal_i2c_device).write_byte(0x12, 0x40);
          id(internal_i2c_device).write_byte(0x13, 0x40);
          ESP_LOGI("aic3204", "Speaker output enabled.");    
```

felixschndr, 2025-11-02:
@bram96 do I understand this right that your changes are only done in software or did you also short that pin to ground as @ItsHarper described?

bram96, 2025-11-03:
@felixschndr yes I also shorted that pin. If you don't the 3.5 jack will physically block the other speaker

lgnap, 2026-01-06:
I took the excellent work of @bram96 (thanks).
Made a select with it.

On the esp reboot the chipset fallback on speaker mode.
For me this is not an issue (as I want it as default mode) and I pay attention to put back the right status into HA.

```
i2c_device:
  id: internal_i2c_device
  address: 0x18

select:
  - platform: template
    name: "Audio Output"
    id: audio_output
    icon: mdi:audio-video
    options:
      - "Speaker"
      - "AUX"
    initial_option: "Speaker"
    optimistic: true
    restore_value: false  # DONT restore old state, speaker mode on reboot
    on_value:
      then:
        - lambda: |-
            if (x == "Speaker") {
              ESP_LOGI("aic3204", "Switching to Speaker output...");
              id(internal_i2c_device).write_byte(0x00, 0x01);
              id(internal_i2c_device).write_byte(0x12, 0x00);
              id(internal_i2c_device).write_byte(0x13, 0x00);
              id(internal_i2c_device).write_byte(0x10, 0x40);
              id(internal_i2c_device).write_byte(0x11, 0x40);
              ESP_LOGI("aic3204", "Speaker output enabled.");
            } else if (x == "AUX") {
              ESP_LOGI("aic3204", "Switching to AUX output...");
              id(internal_i2c_device).write_byte(0x00, 0x01);
              id(internal_i2c_device).write_byte(0x10, 0x00);
              id(internal_i2c_device).write_byte(0x11, 0x00);
              id(internal_i2c_device).write_byte(0x12, 0x40);
              id(internal_i2c_device).write_byte(0x13, 0x40);
              ESP_LOGI("aic3204", "AUX output enabled.");
            }
```

Enjoy ;-)

gohamstergo, 2026-10-17:
Apologies for pinging an old thread, but @bram96 or @lgnap did you short the pins on the Voice PE or ReSpeaker before using your code? I just want to be sure this works for VPE before I proceed.

bram96, 2026-10-17:
@gohamstergo see https://github.com/esphome/home-assistant-voice-pe/issues/263#issuecomment-3479124427

gohamstergo, 2026-10-17:
@bram96 hey, yes. but you replied to someone mentioning ItsHarper's post and their post was about the ReSpeaker. hence my double-check. a simple "yes, on vpe" would suffice.

lgnap, 2026-10-18:
@gohamstergo I had to remove a side of a transistor. (Unsoldering and remove a connector) But I don't remember if it's shortening something, I won't remember the circuit.