esphome/home-assistant-voice-pe #265: Getting started process fails with ESPHome problem

tarbard, 2025-01-03:
I'm following the instructions at https://voice-pe.home-assistant.io/getting-started/ to setup a new unit. 

I installed Home Assistant Companion app on my phone, added the BLE integration, then set the wifi password and the device connected to wifi successfully.  

Next according to the getting started guide it should show Home Assistant Voice under Discovered, however what I actually got was this dialog asking me to set a host IP for "ESPHome" 

[image]

I tried setting this to my HA IP address but it errors as shown in the screenshot.  I installed ESPHome as an Addon in case that was a requirement  that was omitted from the getting started page but it still gives this "Can't connect to ESP. Please make sure your YAML file contains an api line.."  error.  

Am I supposed to set this IP address to the voice-pe's ip address, it's not obvious what (if any) IP address it has.  

It has flashing blue lights at this stage.   Note I've factory resetted and retried a few times.   If I just close the "please enter connection settings of your ESPHome node" dialog then the voice-pe doesn't show up at all in Discovered, Devices or anything in entities and I have to factory reset to try again.  

I'm not sure why my experience is different to what is shown on the getting started page.   I did a search but didn't find any mention of this.

jackwilsdon, 2025-01-03:
Home Assistant is meant to discover the Voice PE. If it's not able to discover it, you should be able to put its IP in the "Host" field and it'll do the Voice PE onboarding. You can get the IP from your router or by using a network scanning app.

tarbard, 2025-01-03:
Thanks for the quick reply,  that helped - I did try some IP addresses before creating the issue but there's so many on my network I wasn't sure if I was doing the right thing or something else was wrong.  

Eventually found the right IP and it's connected now.  Cheers.

closed 2025-01-03 by tarbard

swyxio, 2025-01-20:
WHAT WAS THE RIGHT IP?!?!?!?! what did you see tarbard?!?!?

tarbard, 2025-01-20:
> WHAT WAS THE RIGHT IP?!?!?!?! what did you see tarbard?!?!? how do i find the right IP?
> 
> this is such an xkcd situation. i'm in the exact same boat and tarbard just didnt say the one bit of information i needed lol

:-)  Well it depends on what IP your router assigned it.  You need to go to your routers admin page and find a page that lists assigned IP addresses and devices and look for one that says voice assistant or something and that will tell you what it is.     Not ideal but it worked in the end.

ottocoster, 2025-02-11:
I ran into the same issue. To find the IP address of the Home Assistant, you can run `arp -a`:

[image]

quinndiggity, 2025-06-19:
Just a note, in case anyone else runs into this, `home-assistant` itself (not the `speaker`) will naively attempt to use the configured local address during an initial connectivity test (prior to the speaker itself making any attempts), by making its own request to that address; in cases where this obviously won't work (ie. via reverse proxy, where upstream proxied services aren't allowed to talk to each other via the proxy, such as when configured to disallow server-side request forgery), a special exemption will need to be added to allow home-assistant to speak with itself via the proxy 🙄