esphome/home-assistant-voice-pe #266: Voice PE bricked?

Grekkor, 2025-01-03:
Hey all.

I got my Voice PE today and I was trying to set it up using my iPhone and the companion app.

It was discovered using BLE I entered the Wifi credentials and now the ring on the Voice PE is solid white and it seems to reboot every 10ish seconds making a slight plop sound (presumably the speakers).

And I cannot connect to the ESPHome Node. I made sure that the IP is correct (its the IP of the Voice PE not HAOS), I even gave it the static IP just to be sure.

[image]

The worst part is that I cannot reset it according to the instructions in the documentation. I keep holding and holding the button but the circle never turns red it stays solid white.

Any help?

rlg6767, 2025-01-04:
I have this exact same issue.

rlg6767, 2025-01-04:
OK,  managed to get it working.  Follow these steps (from memory):

1. Navigate to ESPHome addon in Home Assistant
2. Click discover button towards the top right of the page
3. Once device has been discovered you can click install -->  wirelessly
4. ESPHome will then recompile the firmware and attempt to install (I had to do it twice as I had that boot loop issue you mentioned)
5. If successful you should then be able to import it into Home Assistant

Grekkor, 2025-01-04:
Hey thanks.

 I fixed it in almost the same way but using the esphome CLI.

Your solution would have been faster.

However this issue needs to be ironed out for the full release. The average person will not want to use a CLI or esphome.

vilhelmericson, 2025-01-05:
Hi,
I had a similar situation, the installation failed, it ended with a solid white LED ring on the device. In the router I noticed that the device was successfully connected to the local network and in HA had the ESPHome integration been added but the device was not added in ESPHome. When I tried to add it to ESPHome I got the message "Can't connect to ESP. Please make sure your YAML file contains an 'api:' line." As the hardware reset routine described in the documentation did not work I looked for another solution and found it:

1. Navigate to [Home Assistant Voice Preview Edition Documentation page](https://voice-pe.home-assistant.io/resources/)
2. Click on the [Firmware source code](https://github.com/esphome/home-assistant-voice-pe) link near the bottom of the page
3. Click on the [If you need to re-install the firmware, use this installer link](https://esphome.github.io/home-assistant-voice-pe/) at the bottom of the page.
4. Connect your device to your computer and press the Connect button to start a wizard that installs the latest firmware on the device.
This worked just fine!