esphome/home-assistant-voice-pe #267: Feature Request: Option to Disable LED Indicators for Mute and Microphone at Night

highTowerSU, 2025-01-03:
Currently, the LED indicators for mute and microphone status remain illuminated during the night. While this is useful during the day, it can be disruptive in dark environments, such as bedrooms. An option to automatically disable or dim these LEDs during specified hours (e.g., night mode) would significantly improve the user experience.

Proposed Solution:

Introduce a "Night Mode" or "LED Control" feature in the settings that allows users to:

1. Schedule a time range when the LEDs should be turned off or dimmed.

2. Manually toggle the LED indicators for mute and microphone.

3. Optionally adjust the brightness of the LEDs for less intrusive lighting.

Benefits:

Improved user comfort, especially in dark environments.

Enhanced flexibility and customization for users.

Potentially reduced energy consumption.

Use Case:

A user places the device in their bedroom and prefers a completely dark environment while sleeping. With this feature, they could schedule the LEDs to turn off between 10 PM and 7 AM, ensuring an uninterrupted night's sleep.

jackwilsdon, 2025-01-03:
It would be nice to have a separate brightness slider for the light. I'm currently using this script as a workaround:

```yaml
sequence:
  - variables:
      mute_entity_id: |
        {% for entity_id in device_entities(device) %}
          {% if entity_id.startswith("switch.") and entity_id.endswith("_mute") %}
            {{ entity_id }}
          {% endif %}
        {% endfor %}
      light_entity_id: |
        {% for entity_id in device_entities(device) %}
          {% if entity_id.startswith("light.") %}
            {{ entity_id }}
          {% endif %}
        {% endfor %}
          
    alias: Get entities
  - variables:
      was_muted: "{{ states(mute_entity_id) == \"on\" }}"
      was_lit: "{{ states(light_entity_id) == \"on\" }}"
    alias: Get state
  - alias: Unmute if muted
    if:
      - condition: template
        value_template: "{{ was_muted }}"
    then:
      - action: switch.turn_off
        metadata: {}
        data:
          entity_id: "{{ mute_entity_id }}"
      - delay:
          hours: 0
          minutes: 0
          seconds: 0.5
  - action: light.turn_on
    data:
      brightness: "{{ brightness }}"
      entity_id: "{{ light_entity_id }}"
    alias: Turn on light
  - if:
      - condition: template
        value_template: "{{ not was_lit }}"
    then:
      - action: light.turn_off
        data:
          entity_id: "{{ light_entity_id }}"
    alias: Turn off if previously unlit
  - alias: Mute if previously muted
    if:
      - condition: template
        value_template: "{{ was_muted }}"
    then:
      - action: switch.turn_on
        metadata: {}
        data:
          entity_id: "{{ mute_entity_id }}"
alias: Set assist brightness
description: ""
fields:
  device:
    selector:
      device: {}
    name: Device
    required: true
  brightness:
    selector:
      number:
        min: 1
        max: 255
    name: Brightness
    required: true

```

It's a bit of a pain as changing brightness whilst muted doesn't actually work (seems like a bug?), so the script unmutes (if previously muted), changes the brightness and then remutes (if previously muted). Thankfully turning the light on and off again quickly like this doesn't cause it to flash unless it was already on.

SpencerDub, 2025-01-29:
I just posted a question about this very thing on the HA forums.

I would greatly appreciate this feature, for exactly the reasons you outline. I'd like a Voice PE in my bedroom; I'd also like it to be fully muted _and_ dark when I'm sleeping.

unreturnable, 2025-02-09:
👍🏻 for this feature. We have one of the preview edition devices we want to just use as a smart speaker in a bedroom in our house. So we want to be able to have the mute continuously turned on for that device but not have the red led on while sleeping

thewan056, 2025-03-28:
Call it DND or Do Not Disturb toggle, and allow the user to toggle one switch where, in addition to the above suggestions of scheduling lighting and muting the mic, also control the volume or mute the sound output too. 
Just in case. We want a good nights sleep with a peace of mind that it doesn't light up, doesn't listen and doesn't make noise too.

KostaMadorsky, 2025-07-01:
I might just put a smart plug for the voice device in the bedroom and schedule it to turn off during the night...

itpropro, 2026-02-07:
Is this still not fixed? The home assistant voice feels like an echo dot from 5 years ago with how little you can configure and how bad it works currently…