esphome/home-assistant-voice-pe #268: Voice, ChatGPT and Android TV Remote not functioning

Swerfer, 2025-01-04:
Maybe this is an HA issue, but the Android TV Remote integration doesn't work with Voice PE and ChatGPT. If I ask to pause my 'Televisie' it reacts with (translated from Dutch): 'It looks like there is no television which can be paused'. See screenshot. If I press the buttons, then the television will pause or play.

[image]

jackwilsdon, 2025-01-04:
What's the current state of the TV (I assume it's a `media_player`)? https://github.com/home-assistant/intents/issues/2782 notes that you can't control idle media players.

Swerfer, 2025-01-04:
It's a TV box from my fiber provider with Android TV as OS. It is always active.