esphome/home-assistant-voice-pe #280: Timer logic for LED ring should use 1-11 not 0-11 for the percentage remaining

dotorg, 2025-01-10:
It's a little weird the way the LED ring works for timers right now. It is lighting them purely based on the percentage complete of the timer. That puts you in a weird state with timers where the last 4% of the time there are no LEDs lit because it rounds down to zero, so there's no visual indicator a timer is running.

This line:
            auto timer_ratio = 12.0f * id(first_active_timer).seconds_left / max(id(first_active_timer).total_seconds , static_cast<uint32_t>(1));

really should be

            auto timer_ratio = 1 + (11.0f * id(first_active_timer).seconds_left / max(id(first_active_timer).total_seconds , static_cast<uint32_t>(1)));

jlpouffier, 2025-01-17:
Interesting.
Let me give it a try locally and push this as a fix if it's better. 
Thx a lot

jlpouffier, 2025-01-17:
Well your timer ration can be bigger than 11 so it's going to be out of bound when the timer is at 100%. 
But I can adapt