esphome/home-assistant-voice-pe #283: Conversation ability

Bengarman, 2025-01-11:
It would be great if we could have the option to use the conversations from the LLM. If I currently say something and Chat GPT asks me a question back like do you mean turn the heating on downstairs I would like to be able to say yes please.

This would massively improve the adoptability in my home.

pierrewessman, 2025-01-12:
This [PR](https://github.com/esphome/home-assistant-voice-pe/pull/272) might help ypu achieve that when its merged.

martijnrusschen, 2025-01-13:
Agree, currently I need to respond with "ok nabu, yes" which other members in my household find hard to understand :)

kahrendt, 2025-04-02:
This is supported in Home Assistant 2025.4 and VPE firmware 25.3.0 when using an LLM, so long as the LLM response ends with a question.

closed 2025-04-02 by kahrendt