esphome/home-assistant-voice-pe #284: The components are not compatible with the i2s microphone

KristopherMackowiak, 2025-01-11:
If we want to use the code in a DIY project, after changing the microphone from nuba to i2s the device gets a bootloop

KristopherMackowiak, 2025-01-11:
To test you can flash esphome code to clean ESP32 S3 board with replace microphone section to 

microphone:
  - platform: i2s_audio
    id: asr_mic
    i2s_din_pin: GPIO15
    adc_type: external
    bits_per_sample: 32bit
    i2s_audio_id: i2s_input

and set mic id in wake word and assistant section

synesthesiam, 2025-01-14:
The various nabu components are being upstreamed into ESPHome, but this will take time.

ItsHarper, 2025-02-25:
ESPHome 2025.2 moves a lot of the code here upstream, I would look into this again

whc2001, 2025-03-18:
Same here, the microphone component on 2025.3 still does not support `TickType_t ticks_to_wait`, and using the microphone component in this repo I got something wrong in the underlying I2S reading process:
```
[20:13:21][D][micro_wake_word:379]: Starting wake word detection
[20:13:21]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[20:13:21]
[20:13:21]Core  0 register dump:
[20:13:21]PC      : 0x4205f2c8  PS      : 0x00060630  A0      : 0x8200e0d3  A1      : 0x3fcc8e50
WARNING Decoded 0x4205f2c8: i2s_read at C:\Users\WHC\.platformio\packages\framework-espidf\components\driver\deprecated/i2s_legacy.c:1739 (discriminator 2)
[20:13:21]A2      : 0x00000000  A3      : 0x3c1f4670  A4      : 0x00000140  A5      : 0x3fcc8e90
[20:13:21]A6      : 0x00000000  A7      : 0x3fca2e28  A8      : 0x3fca2e28  A9      : 0x00000001
[20:13:21]A10     : 0x03cc8200  A11     : 0x3fcc8200  A12     : 0xb33fffff  A13     : 0xfffffffd
[20:13:21]A14     : 0x00000404  A15     : 0x0000000a  SAR     : 0x00000001  EXCCAUSE: 0x0000001c
[20:13:21]EXCVADDR: 0x00000014  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
[20:13:21]
[20:13:21]
[20:13:21]Backtrace: 0x4205f2c5:0x3fcc8e50 0x4200e0d0:0x3fcc8e90 0x420ed479:0x3fcc8ec0 0x4201331a:0x3fcc8ee0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4205f2c5: i2s_read at C:\Users\WHC\.platformio\packages\framework-espidf\components\driver\deprecated/i2s_legacy.c:1739 (discriminator 2)
WARNING Decoded 0x4200e0d0: esphome::i2s_audio::I2SAudioMicrophone::read(short*, unsigned int) at E:\ESPHome_Projects\esp32-voice-assistant\.esphome\build\esp32-voice-assistant/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.cpp:161
WARNING Decoded 0x420ed479: esphome::microphone::Microphone::read(short*, unsigned int, unsigned long) at E:\ESPHome_Projects\esp32-voice-assistant\.esphome\build\esp32-voice-assistant/src/esphome/components/microphone/microphone.h:31
WARNING Decoded 0x4201331a: esphome::micro_wake_word::MicroWakeWord::preprocessor_task_(void*) at E:\ESPHome_Projects\esp32-voice-assistant\.esphome\build\esp32-voice-assistant/src/esphome/components/micro_wake_word/micro_wake_word.cpp:165
```

Hedda, 2025-05-09:
Will be solved by https://github.com/esphome/home-assistant-voice-pe/pull/392 once or ESPHome 2025.5.0 has been released?

* https://github.com/esphome/home-assistant-voice-pe/pull/392

  * https://github.com/esphome/esphome/pull/8645

    * https://esphome.io/changelog/

PR #392 (2025-05-21): Update to ESPHome 2025.5 - use upstream audio components
This will currently fail CI tests, since ESPHome 2025.5.0 hasn't been released yet.

- Updates the configuration to use the upstream audio code in ESPHome 2025.5. 
  - Microphones are handled differently - uses callbacks instead of restricting each microphone to exactly 1 downstream reader
  - Uses built-in actions to enable and disable the "stop" wake word instead of using lambdas
- Removes all external components' code (except for ``voice_kit``, which is not planned to be merged upstream).

This will finish implementing a couple outsta [... cut]
closed 2025-05-21 by jesserockz