esphome/home-assistant-voice-pe #285: Unable to install

cayalag, 2025-01-12:
Hello !

I have been trying to install my new device for 3 or 4 days now but there is no way... I manage to configure the WiFi network , but when I try to Set Up the Voice Assistant in ESPHome I get the following message:

"The voice assistant is unable to connect to Home Assistant

To play audio, the voice assistant device has to connect to Home Assistant to fetch the files. Our test shows that the device is unable to reach the Home Assistant server."

[image]

I have checked System->Network and IPs are properly setup. HAVPE is running the latest firmware, same as HA.

Any idea what to try next ?

tommyjlong, 2025-01-12:
Although i don't have my VPE yet, I am seeing this problem show up on the HA Forum, Discord.  There seems to be a TTS problem, where the VPE tries to get a file from the HA Core/server to play and retrieving this file fails.  

In your HA UI->Settings->Network there is a section: "Home Assistant URL".  and under that is "Local network".  Is your Local network setting using https?

cayalag, 2025-01-12:
That was the first thing I checked… it is just http and the IP address is correct. I’ve tried setting it to Automatic as well but same thing.

pierrewessman, 2025-01-13:
You could try looking at the HAVPE logs while installing by using [aioesphomeapi-logs](https://github.com/esphome/aioesphomeapi).

oddlama, 2025-02-14:
I have the same issue. I traced what the device is doing and found that it successfully connects via TLSv1.3 to my home assistant instance and accesses a `.flac` file. But then the home assistant frontend tries to call `assist_satellite/test_connection` on the entity over a websocket connection which times out:

https://github.com/home-assistant/frontend/blob/5dc5879773293a9418238940c9658aa6c5142fc2/src/data/assist_satellite.ts#L38-L47

Unfortunately I have not yet figured out how to trace what home assistant is doing in the background so I can understand the reason for the timeout.

oddlama, 2025-02-18:
I was able to fix my problem by allowing my instance to connect to its own URL, which I had blocked in nginx (your setup may not use this). You may get more information by enabling debug logging on the esphome integration in home assistant prior to adding the voice pe device. See https://github.com/home-assistant/core/issues/138813.