esphome/home-assistant-voice-pe #287: home_assistant_voice esphome binary sensor are deprecated

hugo1906, 2025-01-13:
Deixará de funcionar na versão 2025.4. Corrija o problema antes de atualizar.

O sensor binário de ESPHome em curso binary_sensor.home_assistant_voice_090818_assist_em_curso está desatualizado.

Migre sua configuração para usar a entidade correspondente assist_satellite e, em seguida, clique em ENVIAR para desativar o sensor binário em curso e corrigir este problema.

jf-uu, 2025-01-14:
Did you enable the "In progress" entity for the Voice PE? That repair only shows up if it's enabled, and it's meant to be disabled by default (it was for me). Also same issue as #237.

hugo1906, 2025-01-14:
Where is the in progress located

jf-uu, 2025-01-14:
It's on the page for your device. It can be found by going to Settings -> Devices & services -> ESPHome -> "1 device" under your voice PE, it shows up under Sensors:
[image]

apbarratt, 2025-01-14:
I do indeed have it on, I'm rather fond of having access to all the datas so tend to keep enable any and all sensors 😅

jf-uu, 2025-01-14:
It's being removed in 2025.4 - the `assist_satellite` entity offers the same functionality (and more).

hugo1906, 2025-01-14:
thanks this solved the problem.

closed 2025-01-17 by jlpouffier