esphome/home-assistant-voice-pe #289: Using HAVPE without Home Assistant

skorokithakis, 2025-01-13:
I just got a HAVPE and I'd like to hack on it and use it for my own projects. It's great that it's open source and ESPhome-based, however I can't find any information on how to talk to it, or add it to my network, without Home Assistant. Is there any info on that? I'm just looking for a way to go from "unboxing" to "I have a connection that can talk Wyoming", and I can figure out the rest on my own.

Thanks for any help!

skorokithakis, 2025-01-14:
Hmm, I'd rather not reflash it. How does it talk to Home Assistant?

apbarratt, 2025-01-14:
...you'd rather not reflash it.  Mate, I think this maybe isn't the right project for you, it's been specifically designed to talk to home assistant but all ESPHome projects are designed to be tinkered with and reflashed to your desire.  If you want it to do new things, you've got to open up that ESPHome yaml code and start changing things and then flashing what you make onto it.

skorokithakis, 2025-01-14:
@apbarratt I want to write a server that pretends to be Home Assistant. It would defeat the purpose of doing that if I reflashed the device. Given that, is there any documentation that I can use to do what I want to do? I'm not looking to challenge the purpose of what I'm building, I'm looking for documentation on how to build it.

apbarratt, 2025-01-14:
I'm not really sure how I'd begin doing that.  But given Home Assistant is free open source software, I'd kind of wonder why you wouldn't just have a server pretend to be home assistant so hard that it actually is just an installation of home assistant ;)

skorokithakis, 2025-01-14:
We can discuss that outside of this ticket, which is specifically about documentation on the HAVPE protocol.

synesthesiam, 2025-01-14:
ESPHome uses Protobuf messages over TCP. This is the library used on the Home Assistant side: https://github.com/esphome/aioesphomeapi

skorokithakis, 2025-01-14:
Fantastic, thank you! Do you know how discovery is done? I assume there's an mDNS (or something) query to find the device, and then HA connects to ESPHome?

Rudd-O, 2025-01-23:
Yes.

closed 2025-04-02 by kahrendt