esphome/home-assistant-voice-pe #297: Can't update past 24.12.3

fhk1407, 2025-01-18:
Hi all. I just tok delivery of a Nabu Home Assistant Voice PE. I have been struggling quite a lot with it, and now seeking help.

It was delivered to me with a firmware in the 24.10.X series (don’t remember exactly which version). Initial setup went fine (connecting to wifi via companion app), but as soon as I reached the firmware update part, the problems started. Basically it goes into an update loop (looking for updates → updating → looking for updates → etc).

I have flashed quite a few firmware versions to the device, using the web-based update tool referenced here, and here are my results:

-All versions below and including 24.12.3 allow the device to boot and be discovered in HA (but update will still fail)
-All versions above and including the latest 25.1.0 seem to flash well, but render the device unbootable (soft brick). I have [attached ](https://pastebin.com/MiYVWNV7)the boot logs from the device attempting to boot 24.12.4 unsuccessfully.

Anyone else seen this? How do it get this Voice PE updated to latest version?

Razseal, 2025-01-18:
I have a notification in HA to update from 2024.12.2 to 2024.12.4 but it fails every time the ESP builder add-on doesn't even see it needing and update. So I'm seeing this as well.

lecrams, 2025-01-20:
I have exactly the same issue when updating to the latest version 24.12.4

Razseal, 2025-01-27:
I was finally able to get mine updated yesterday but it took a long time of ignoring it while I was processing. Now it is on 24.12.4