esphome/home-assistant-voice-pe #299: Poor wake word pickup

vhamburger, 2025-01-21:
Hey, got my device last week and I notice that it really is poorly performing when using the wake word.
I have set it to "Okay Nabu" and it sits in my living room next to my Alexa (which I want to get rid of...).
However, when I calmly say "Okay Nabu" it never picks up my voice - from like 2m away. I have to raise my voice clearly to let it pick up. 
On the funny side - I have a self built voice assistant based on a raspberry pi and mic hat - which picks up my command every time - but it is sitting some 12 meters away in my home office. That cant be correct.

Is there a way to adopt the microphone gain / or set the xmos different to make it work better? Because right now, wife approval factor is below 0 since it never picks up any command. 
I am happy to adopt it in ESP home and change the params manually if necessary - just want to make it work reliably.

RocketRammer, 2025-01-23:
+1 for wife approval factor.  I understand that these are Preview kits, but that's not how consumer law works when you've marketed it as having X feature.  Having to sit with my face pressed against it is not how it's marketed.

Warren-0, 2025-01-30:
I have this issue too. Okay Nabu will only wait it 1 in 43 times or so. I tried Hey Jarvis and I have to be few inches away from it to respond. 

Australian accent here though. I do have the pi zero* and mic hat and that picks up every time.  I can use the PE this in the office for now, but not in a room where it must sit 1 to 8 meters away at any time. Just feedback nothing more from me.

RassK, 2025-02-02:
Not sure if this is mic / hw issue. Faster whisper with large-v3 model seems very good even with a very rare accent (estonian). In the other hand, wake word works very poorly here too.

grigi, 2025-02-05:
I found that if I deliberately say the wake words in a higher pitch it's significantly more likely to get it, but it's really not good enough for me.
I tried to setup openwakeword (the add-on) but i can't get it added to wyoming, nor can I find where to configure it in the assist config. If it was deprecated the documentation isn't making it clear. A bit frustrating.

jf-uu, 2025-02-05:
@grigi the Voice PE uses microWakeWord to perform on-device wake word detection, so you won't be able to use openWakeWord with it.

vhamburger, 2025-02-20:
Is anyone working on this one or is this only my device. Since ESPHome 2025.2.0 it has gotten worse. I practically have to yell at the device standing 1m in front of it. Is my device broken? Do I have to return it - or is this just "how it is"?
Can anyone let me know if you see a similar pattern? 
For comparison - I have built an assist satellite on an RPI with a 2mic hat and it works flawlessly any time - even from across the room. However, it streams the wake word to HA which probably has a more powerful detection engine and noise cancellation capabilities. 

The current state makes the Voice PE unusable in my household, no one wants to use it as it is never picking up the wake word. Sad.

RocketRammer, 2025-02-20:
I've entirely stopped using mine for Voice @vhamburger and purely use them as speakers now.  Raised a few times across issues here that they are terrible at this, but much like the PE, it seems to fall on deaf ears.

For voice now, I've just setup the action button on our Samsung Galaxy Watches to be Assist and it works incredibly well.  I use local Whisper/Piper with fallback to OpenAI and it's absolutely Rapid.  I wish PE was capable to provide this too.

RassK, 2025-02-22:
Same here, unusable without a good wake word support.

vhamburger, 2025-02-24:
Hey all - so since no one from Home Assistant or the Voice PE team even got back to this (not sure if just ignoring or too much to do otherwise) I started to experiment around on my own. For the following "fix" you need to adopt your device and _slightly_ edit the firmware yaml - if that is not for you, may be the home assistant folks pick this up and fix it in a later release.
I must say I am a bit disappointed that I did not even get a reaction to this - they claim they get so much good response to the device and that it is awesome - which it is don't get me wrong - but also if you look in the issues part in GitHub you can clearly see that I am not the only one with this issue. 

However - "Never give up, never surrender" (quote: Captain Peter Quincy Taggart) - so I started to play with. options and I think a found a combination wich works great for my setting (I am a native German speaker - so it might vary for you ;)

I made the following changes to the original yaml:
```
micro_wake_word:
  id: mww
  models:
    - model: https://github.com/kahrendt/microWakeWord/releases/download/okay_nabu_20241205.4/okay_nabu.json # original: https://github.com/kahrendt/microWakeWord/releases/download/okay_nabu_20241226.3/okay_nabu.json
      id: okay_nabu
      probability_cutoff: 0.78 # Added this to lower detection barrier - may lead to more false positives though
```

So I switched to an older "okay nabu" model by @kahrendt (thanks for all your hard work on that buddy!) Interestingly, the older model seems to be picking up my voice much easier, better and faster. I no more have to "yell" at the device. The lower probability cutoff helps as well. 
Note: I have played with lowering the cutoff only, but that didn't create the wanted results. It helped a bit to detect more wake words, but it also lead to a lot of false positives (mostly during movies where someone said something with "Ok" and a German word that is phonetically similar to "nabu" like "nanu, Kanu, Manu, etc..) 

Long story short: Since I switched to an older wake word model it improved a lot for me. Until @kahrendt might improve the existing model, I plan to stick to that one. It also proves that my devices microphones are fine (I could verify that easily since all voice commands always work - it is just the wake word pickups that is flawed). 

I hope this might help some people. The only "barrier" to this is that you need to adopt the device in ESP Home. If you have never done it, it is quite straightforward. Just remember you probably need to setup your wifi and api encryption key to make it work - but there are plenty of guides on how to do that. 

Enjoy your voice PE. I will keep this updated if I find out more... (or another improvement - still need to test it more with the whole family...)

StrandmonYellow, 2025-03-03:
I thought it was just me, but it feels like it got worse indeed with later versions when compared to an erlier one. Is this being picked up? @synesthesiam  🙃

StuartIanNaylor, 2025-03-17:
Its sort of the same here with no reply but the wake words are never going to work well with the dataset and training routines they have that are incredably poor.

https://github.com/kahrendt/microWakeWord/issues/28#issuecomment-2564783183

mirml, 2025-03-21:
Both of my google mini and nest picks wake word far better than voice assistant pe 😔👎

vhamburger, 2025-05-18:
Hey - this topic seems to be highly ignored by the Voice PE team - I wonder why that is? 
So maybe I do have a faulty device - it is hard to tell. The other day I set up an M5 Stack Mini and it was picking up "OK Nabu" from across the room when Voice PE did not pick it up? How is that even possible with its tiny microphone... 

Is there a way to debug the wake word sound? Can you set it to debug mode or something to examine the audio it picks up. I am really upset now since basically my whole family is rejecting the device as it almost never picks up the wake word - never. 
Everybody is revering back to Alexa now and is making fun of it... Kind of sad, as I wanted to get rid of the Amazon spy...

StrandmonYellow, 2025-05-18:
I don't want to judge because the team is doing many things, but indeed this topic has been open for a while. I have to say the wakeword twice or three times allmost every time for it to trigger correctly. I don't know why that is the case and currently do not have the time to dig deeper into this issue unfortunately.

kahrendt, 2025-05-20:
Let me start by apologizing for our delay in responding to this issue! I do understand that the wake word performance can be a frustrating experience with the device, and hopefully we can improve it.

This will be a longer response, so let me first start by sharing some summarized plans to help improve this, and then I will go into more details on the reasoning/difficulties each one has.

First, the next Voice PE firmware will include a [selector for wake word sensitivity](https://github.com/esphome/home-assistant-voice-pe/pull/403). Keep in mind, it being more sensitive means it's more likely to have false accepts where no one said the wake word.

Second, the microphone gain for the audio fed into microWakeWord is too low for some specific devices, causing false rejections. We will share some experimental firmwares to help find the best settings. One firmware will have an auto-tuned gain level and the other will have expose all the settings to experiment with the many different settings on the device. We want to collect feedback on what works and what doesn't. Once we are confident with either the auto-tuning algorithm or better default settings are an improvement, then we will include it in the stable firmware.

Third, the training process for microWakeWord needs to be improved. This is an open-ended problem with no clear approaches that will guarantee better performance, but there are many research papers available on this topic that could potentially help.

-----

Technical details for those curious and those who want to help:

When I tuned the probability cutoffs, I set it so that there were no false accepts on the [Dinner Party corpus](https://www.amazon.science/publications/dipco-dinner-party-corpus), which contains several hours of recordings (from many different microphones) where groups of people converse. This is probably too strict, and having a selector will let you have a choice in the false rejection vs false acceptance rate that is best for you. One thing other note is that the "Hey Mycroft" models and "Hey Jarvis" models don't fully meet the no false-accepts standard I set above. When I trained those two models, the probability cutoffs were only tuned against the first channel on the first microphone unit. The "Okay Nabu" model is tuned against every microphone channel and unit. All future models I train will train will be tuned against all channels and units.

The XMOS chip has several [algorithm stages](https://www.xmos.com/documentation/XM-014785-PC/html/modules/voice/doc/user_guide/audio_processing/index.html) for cleaning the audio. First, it applies Acoustic Echo Cancellation (AEC) to remove any audio played on the device from the microphone signals. It then is processed by an Interference Cancellation (IC) filter that combines the two on-board microphones into one signal while removing non-speech noises. Then it is fed into a Noise Suppression (NS) filter that further removes stationary noise sources (think things like a fan). Finally, it is fed into an Automatic Gain Control module that adjusts the volume so that speech is at a consistent loudness level.

microWakeWord takes audio in and converts it into a spectrogram. As part of this process, it applies its own [noise suppression and automatic gain control](https://github.com/tensorflow/tflite-micro/tree/main/tensorflow/lite/experimental/microfrontend/lib). On the VPE, microWakeWord uses all the XMOS processing stages except for the AGC; we found that the XMOS AGC interfered with mWW's AGC causing more false accepts. Note that the audio we send for STT processing goes through all the XMOS stages, including XMOS's AGC.

We currently hardcode the microphone gain for the audio fed into mWW to be ~6dB, but the next ESPHome release has [changes](https://github.com/esphome/esphome/pull/8695) to make tweaking this easier. This is probably too low in general, and on some specific devices this is far too low (there are probably subtle manufacturing differences from device to device). The first experimental firmware will have an auto-tuning gain level for the audio fed into mWW's spectrogram generator. The current algorithm monitors the audio's [RMS level and the peak levels](https://github.com/esphome/esphome/pull/8737). When the two levels are close, we assume the RMS level measures the noise floor for the device and room. We then slowly adjust the gain so that the RMS level hits a target level. My own tests show this is beneficial, but there are several parameters that are currently just guesses (especially the target noise floor). Hopefully with some more testing (beyond what I can do!), we can find optimal parameters that improve the experience for every device. The second experimental firmware will expose all of the settings I discussed above (which pipeline stage, which gain factor, and which probability cutoff). This second firmware is meant for people to do some in-depth experiments to see if there is anything with these settings that we are missing and that work better in more situations.

Improving [microWakeWord](https://github.com/kahrendt/microWakeWord/) is a fun challenge, but I could definitely use some help! Some relatively low-hanging fruit is to update it to support newer versions of Python so people can easily train new models in Google Colab. Once it is easier for people to train new models, I need feedback on what are good hyper-parameters for the training process. Please keep in mind that the training noteboo [... cut ...] l, and I primarily used real samples for the "Hey Mycroft" model.

If you are experienced with machine learning (or are just eager to learn and try things), I'd love your help with improving microWakeWord. I think fine-tuning models with individualized real samples has a lot of potential. It may even be possible to train very specific models that only work for a few people if they collect enough personal samples. This all requires experimentation, so the more people trying, the better it can be!

StrandmonYellow, 2025-05-20:
@kahrendt Thankyou for the detailed explanation and the amount of work that is going into this! Very much appreciated!

vhamburger, 2025-05-20:
Thanks for this very thorough and detailed reply! Highly appreciated!
Also - if there is any data you need for improvement - let us know and I am sure most of us would be happy to share!

StrandmonYellow, 2025-05-23:
I updated my VPE, I will do some tests this evening!

kahrendt, 2025-05-27:
I have one of the experimental firmwares ready to go for testing. [This PR comment](https://github.com/esphome/home-assistant-voice-pe/pull/410#issuecomment-2912639610) has instructions on how to download and upload the firmware to your VPE. 

It attempts to set the gain level so that the RMS sound level is near a target value. It initially targets -75 dB, but there is a number entity to change what it targets. Keep in mind this is meant to change things on long time scales, so it won't necessarily immediately be noticeable. The device needs to be in a quiet room for it to start adapting (quiet room meaning the RMS and peak dB levels are within 20 dB over the span of 30 seconds). 

I'd appreciate anyone taking the time to run it and reporting there results! I'm curious if it affects how many false activations and false rejections you experience. If you experiment with the target dB level, please report if any particular setting is better or worse!

Ramblurr, 2025-12-27:
For my family the Mycroft WW worked best, but once I took control of the device and used the dev branch's `github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml@dev` the wake word selector entity is unavailable in HA and the PE only responds to "Okay Nabu".. searching the git repo shows that the WW selector was removed EOY 2024, before I received my PEs in Jan 2025.

https://github.com/esphome/home-assistant-voice-pe/commit/dea3a2c10ad0a2f6c89bfbb773bf4412cc36434d

Is it not supported to have multiple simultaneous WWs on the PE dev/latest firmware?