esphome/home-assistant-voice-pe #300: [REQUEST] Add an ESP32-C6 as "coprocessor" and "IEEE 802.15.4 radio" module to future voice boards (making a single PCB that have both a ESP32-S3 and a ESP32-C6) so that second SoC can be used as Touchscreen Display Controller and/or Thread Border Router

Hedda, 2025-01-21:
For the next-generation voice satellite board design/schematics please consider also adding another ESP32 SoC to the same PCB board as well as code support for that secondary ESP32 SoC to in the future be used as a Touchscreen Display Controller and/or Thread Border Router. Specifically suggest adding an [ESP32-C6](https://www.espressif.com/en/products/socs/esp32-c6) as a second radio SoC to the same so PCB and make that secondary act as a "co-processor" and "IEEE 802.15.4 radio" module. 

The two main real-world use cases I see justify adding an ESP32-C6 SoC as coprocessor and radio module are primarily two-fold:

1. That ESP32-C6 could be used as a display/controller (or offload some other related things to free up resources and make it more future-proof):

2. Perhaps more interestingly, the ESP32-C6 has a built-in IEEE 802.15.4 radio which means it can be used as a "Thread Border Router" (with OpenThread Border Router RCP firmware), to enable a Thread network and another entry point that can be used by Matter standard fabric. So if add this and could get it working as a TBR/OTBR then it would be super easy for users to add multiple Thread Border Routers to their network.  

* https://www.home-assistant.io/integrations/thread/
    * https://www.home-assistant.io/integrations/matter/

Regardless if having an integrated "Thread Border Router" would be the most practical new feature or not in the long-run. For the short-term I personally believe that the coolness of having a alternative model variant of Home Assistant voice satellite is the coolest fearute that woulc attract the most new users and developers, as having at least one early model with a a relativly large touchscreen should attract interest for even those users and developers who are not interested in voice.

As such having the suggested secondary ESP32 on the same board should in theory allow support for a capacitive touchscreen display with a local UI without sacrificing resources on the primary ESP32. It would at least be awesome to have a variant of a Home Assistant Voice Satellite with a large touchscreen display (.i.e. something similar in design to the existing Google Nest Hub / Google Home Hub and the Amazon Echo Show products).

  * Check out for example this 7-inch series of capacitive touchscreen displays from Elecrow with display driver for embedded ESP32 (which actually do have a microphone so could potentially be used as-is for development) -> https://www.cnx-software.com/2025/01/17/crowpanel-advance-esp32-s3-displays-with-replaceable-wifi-6-thread-zigbee-lora-and-2-4ghz-wireless-modules/    [image]

From a marketing point of view it would also allow you to add Thread's Thread Border Router icons on the product box

* https://csa-iot.org/wp-content/uploads/2023/08/Matter-Retail-Education-Presenation.pdf

[image]

One downside of adding an extra ESP32 (ESP32-C6 or ESP32-H2) it obviously that it would take up some additional space on the PCB and complitate FCC certification a little because of it now having multiple radios, however I do not believe it would add that much larger BOM cost, but the additional possibilities such an extra ESP32 SoC could add should hopefully more than make-up for that slightly higher cost(?).

While I think that it having "Thread Border Router" will be of most interest to many users in the future, and alternativly use case for ESP32-C6 could be to instead use it as a remote Zigbee Coordinator (also known as a Serial-over-IP Zigbee controller adapter) for [Home Assistant's built-in ZHA integration (native Zigbee Gateway)](https://www.home-assistant.io/integrations/zha), see/follow this work-in-progress but note that the ESP Zigbee radio for zigpy is is still very experimental and not yet fully working with ZHA:

* https://github.com/lhespress/zigpy-espzb
  * https://github.com/zigpy/zigpy-cli/pull/45
    * https://github.com/zigpy/zigpy/issues/1052

And yet another option could of course be to also use the ESP32-C6 as a "coprocessor" module a a dedicated Bluetooth Proxy chip with ESPHome:

* https://esphome.io/components/bluetooth_proxy.html

Note that the reason main you want to combine ESP32-S3 and ESP32-C6 is not only to off-load those processes but more importantly so you separate Wi-Fi signals from Thread/Zigbee signals, and that is due to WiFi tendacy to overwhelm and interfer with Thread/Zigbee communication because how they both work. Same goes by the way for WiFi and Bluetooth Low Entergy but it usually is not a great impact on user experience as for Thread/Zigbee:

* https://www.home-assistant.io/integrations/zha#zigbee-interference-avoidance-and-network-rangecoverage-optimization
* https://www.home-assistant.io/integrations/bluetooth#bluetooth-interference-with-other-devices

Anyway, what I think what would make a products like Home Assistant voice satellites perfect as a "Thread Border Router" is that a user can add multiple Thread Border Routers (TBR/OTBR) to their Thread/Matter network for redundacy (using fail-over) and having multiple voice assistant hardware apöliances around thier home is something that many Home Assistant users are likely to have many of and placed all over their home, so having many Thread Border Routers is one of the keys to getting a more robust Thread network.

* https://www.home-assistant.io/integrations/thread/#about-thread-border-routers

[image]

_Image taken from [the Thread Smart Home Fact Sheet](https://www.threadgroup.org/support#Resources) by the Thread Group. It illustrates the landscape of Matter, Thread, and Border routers. Instead of Matter, you could also see another protocol here, such as HomeKit._

_Unlike other IoT protocols, Thread can use multiple border routers in a single network. This increases wireless coverage and reduces the risk of a single point of failure. Ideal for home automation, with a potentially large number of devices spread over a large area._

Slightly different image here on the threadgroup.org when they describe a Thread based Matter fabric using the same text but also mentioning that _Thread Border Router can be built int many devices such as access points, **smart speakers**, etc._:

* https://www.threadgroup.org/BUILT-FOR-IOT/Smart-Home

[image]

Also check out this blog which better describe what a Thread Border Router actually is and why it is very good to have many:

* https://www.threadgroup.org/news-events/blog/ID/356/WHAT-IS-A-THREAD-BORDER-ROUTER-AND-HOW-IS-IT-DIFFERENT-FROM-A-HUB-OR-BRIDGE

[image]

As proof-of-concept for the idea of using a ESP32-S3 + ESP32-H2 combination board made for this purpose check out the "**_ESP Thread Border Router_**" board which is an official Espressif official SDK reference hardware for this purpose (can that use used as either Thread Border Router or as a remote Zigbee Coordinator):

* https://openthread.io/guides/border-router/espressif-esp32

  * https://www.sparkfun.com/products/25497

Not tested myself but suposedly you can already get those working with Home Assistant's Thread integration:

* https://www.youtube.com/watch?v=nHH7j--lfDo

  * https://github.com/home-assistant-libs/python-otbr-api/issues/61

Development is primarily being done with Espressif's ESP Thread Border Router SDK hardware (based on ESP32-H2-MINI-1 module):

- https://docs.espressif.com/projects/esp-thread-br/en/latest/
  - https://github.com/espressif/esp-zigbee-sdk/tree/main/examples/esp_zigbee_gateway
    - https://github.com/espressif/esp-thread-br
    - https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br
    - https://openthread.io/guides/border-router/espressif-esp32

[image]

[image]

OpenThread Border Router Example as in a similar two-SoC setup combining both a ESP32-S3 and ESP32-H2 in the same solution:

* https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

[image]

Hedda, 2025-03-11:
For reference, FYI, looks like [clydebarrow](https://github.com/clydebarrow) is working on adding a serial packet transport platform to ESPHome which could maybe be used to facilitate sending data between two ESPHome devices?

- https://github.com/esphome/esphome/pull/8214

GrumpyMeow, 2025-11-15:
I've just started getting my WT99P4C5-S1 up and running. Anybody have something to share?

Hedda, 2026-10-09:
For reference, @ale1800 who is another Home Assistant community member, is currently working on a drop-in PCB replacement for the Amazon Echo Dot 3 / Amazon Echo Dot 3rd generation in a new project he calls "***LocalDot***", and somewhat relevant to this idea he is using an ESP32-P4 as a co-prosssor instead an XMOS MCU, (while still uisng the ESP32-S3 as the main host processor) however he is using that ESP32-P4 as a DSP for audio processing:

* https://community.home-assistant.io/t/re-using-an-echo-dot-3rd-gen-with-voice-assistant/1020256/

  * https://github.com/ale1800/LocalDot

Notable it sounds like (pun intended) like he has is using an ESP32-P4 as a co-prosssor instead an XMOS MCU, (while keeping an ESP32-S3 as the main host MCU instead of also upgrading that the ESP32-S3 to the newer ESP32-S31 to gain the benefits of its built-in DSP, plus Bluetooth Sink possibility, as well as having the option using its EEE 802.15.4 radio for Thread or Zigbee in the future).
  
   * https://www.reddit.com/r/homeassistant/comments/1v4626r/echo_dot_3rd_with_voice_assistant/

      * https://www.reddit.com/r/homeassistant/comments/1wvmpxs/echo_dot_3rd_dropin_replacement_progress/

PS: FYI; he is also  planning to launch a crowdfunding campaign on Crowd Supply for the LocalDot.

  * https://www.crowdsupply.com/aaelectronics/localdot

Hedda, 2026-10-16:
FYI and for reference, ESPHome has now merged a PR for initial experimental support to use ESP32 as an OpenThread Border Router:

* https://github.com/esphome/esphome/pull/18104

    "_Lets an ESP32 act as an OpenThread Border Router over its Wi-Fi connection, with the native 802.15.4 radio on the ESP32-C5 and ESP32-C6 or an external radio co-processor on a UART for the other Wi-Fi ESP32 variants._"

  * https://github.com/esphome/esphome.io/pull/7289

    "_This adds an experimental way for ESP32 to use an ESP32 as a Thread border router, connecting a Thread network through Wi-Fi. ESP32-C5 and C6 boards use their built-in radio; other Wi-Fi ESP32 variants can use a separate radio connected by serial wiring._
    
    _Pawel Puterla tested the built-in-radio path on an ESP32-C6 for two weeks without a restart or router hang. An ESP32-S3 with a separate radio hung two or three times over about three weeks, so that path needs particular care. The work includes setup guidance and is his first contribution to ESPHome._

      _OpenThread configurations now require ESP-IDF 5.5.5 or newer, including ordinary Thread devices. That version has already been the default for several releases; makers who pinned an older version need to remove or raise the pin._"

Hedda, 2026-11-07:
Heads-up, pre-order campaign on Crowd Supply for "LocalDot" ESPHome-based drop-in replacement PCB for Amazon Echo Dot 3 (3rd-generation)** is finally live: 

* https://www.crowdsupply.com/aaelectronics/localdot

  * https://community.home-assistant.io/t/drop-in-replacement-for-echo-dot-3rd-with-voice-assistant/1020256

Its PCB schematics have not been released yet but the creator has said that he will fully open-source the hardware design for it after the campaign if anyone want to modify it and make one themselves, and intrestingly for this discussion it uses the combination of an ESP32-S3 and an ESP32-P4, where the ESP32-P4 replaces the XMOS chip from the Home Assistant Voice Preview Edition design as the audio processor:

* https://github.com/ale1800/LocalDot

  * https://github.com/ale1800/LocalDot/discussions