esphome/home-assistant-voice-pe #301: The Wizard will not appear.  Is it version dependent?

bwims, 2025-01-21:
Hello

As there were no versions specified in the prerequisites for the HAVPE, other than just "Home Assistant already configured on another
device."  I attempted to install this under HA 2024.2

However, after authorizing the device, the wizard did not appear.  Can you confirm whether this is due to the HA version, or if there is any other way of kicking off the wizard?

This is as far as I have got with configuration.  The device responds to the wake word but piper and whisper aren't there.

[image]

Thanks

B.

jf-uu, 2025-01-21:
It looks like the voice setup wizard was only added in 2024.10.b0:

Frontend commit: https://github.com/home-assistant/frontend/commit/305cecb21368265c348578d8ce5427f9b1c3d109
Core commit: https://github.com/home-assistant/core/commit/e7e86d7a323d29599bfd8a6850645e6d21c04a67

Is there a reason you're running such an old version of HA?

bwims, 2025-01-21:
Only that I was not able to upgrade every month due to house moving.  Is there a reason why Nabu Casa did not specify a version number in their prerequisites?  It's quite a presumption that "Home Assistant is configured" implies "not more than 2 months old" don't you think?  How hard would it have been to have mentioned the version number needed?

Thanks