esphome/home-assistant-voice-pe #303: [HW] Mic&sound got toast

dgolovach, 2025-01-22:
Hi, this seems to be a hardware problem.
Got Voice PE working just fine, but after moving it to a different room after the config with power disconnect (static IP), it stopped responding to wake words, and hangs while trying to play something on it. 

Looking into the logs getting random errors like this:

> 2025-01-22 16:53:46.572 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140570804088704] Unexpected exception
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
>     response = await hass.services.async_call(
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     ...<7 lines>...
>     )
>     ^
>   File "/usr/src/homeassistant/homeassistant/core.py", line 2795, in async_call
>     response_data = await coro
>                     ^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/core.py", line 2838, in _execute_service
>     return await target(service_call)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1006, in entity_service_call
>     single_response = await _handle_entity_call(
>                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
>         hass, entity, func, data, call.context
>         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     )
>     ^
>   File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1078, in _handle_entity_call
>     result = await task
>              ^^^^^^^^^^
>   File "/usr/src/homeassistant/homeassistant/components/assist_satellite/entity.py", line 233, in async_internal_announce
>     raise SatelliteBusyError
> homeassistant.components.assist_satellite.errors.SatelliteBusyError

or "broken pipe errors"

Tried to take it over with the ESP Builder and flashed yaml based on github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml firmware that monitors the mic and tries to play something, getting no sound and no samples from the mic.

Tried flashing\reflashing from scratch with no aval.

Posting here since it seems to be a hardware problem, waiting for another sample to arrive to test out.

dgolovach, 2025-01-23:
Digging deeper, apparently it's DFU XMOS error here
https://github.com/esphome/home-assistant-voice-pe/blob/5288dde504325130d50f9465b92a6d712203f49a/esphome/components/voice_kit/voice_kit.cpp#L310

so i guess HW is dead

cogneato, 2025-01-24:
@dgolovach I would contact the vendor you ordered it from and see if you can arrange for a replacement.