esphome/home-assistant-voice-pe #304: new Voice PE - unable to listen (deaf)

frostworx, 2025-01-22:
I received my Voice PE today.
It was immediately detected after enabling BLE.
Adding it to the network worked straightforward.

Initially it was not allowed to access the internet, but did not complain. 

Problem:

I am not allowed to test the "Okay Nabu" in the setup wizard, as the device was set to "deaf":

```
Sag „Okay Nabu“, um das Gerät zu aktivieren

Die Einrichtung wird fortgesetzt, sobald das Gerät aktiviert ist.

Dein Gerät ist taubgeschaltet

Bitte hebe die Taubschaltung deines Geräts auf, um fortzufahren.
```

Button presses are registered, but don't make the device listen.

toggling `home_assistant_voice_XXXXX_mute` doesn't help as well.

after enabling internet access in the firewall, a firmware update was found which flashed without any problems _(Firmware: 25.1.0 (ESPHome 2024.12.2)_ 

the problem persists though and is reproducible when re-entering the "Voice Assistant Set Up"

no idea if voice testing in the debug mode is implemented (especially as only text mode is documented here https://www.home-assistant.io/voice_control/troubleshooting/) but it is greyed out here.

12 entities were created, HA switches do work, button presses on the device as well.
`binary.sensor.home_assistant_voice_XXXXX_assist_in_arbeit `is disabled, in case that is important.

no idea how to proceed from here without deeper troubleshooting.
would be nice if someone has a clue.

dgolovach, 2025-01-22:
Same here, debugging via serial shows that nic doesn't seem to work
https://github.com/esphome/home-assistant-voice-pe/issues/303

frostworx, 2025-01-23:
thank you for confirming, @dgolovach! (sorry for the late reply, I did not get any notification email...).

Just wanted to update the issue and mention that removing the device from HA, factory resetting it and adding it again into HA didn't help as well.

So no need to try the steps from [this manual](https://voice-pe.home-assistant.io/guides/reset/)

cogneato, 2025-01-23:
Is the physical mute switch toggled? I saw someone else who thought it was a power switch.

frostworx, 2025-01-23:
waaaah, that's the first thing I tried yesterday, but now it just worked indeed! a bit embarrassing tbh...
Thank you so much, I guess I would not have tried it again so fast :)

frostworx, 2025-01-23:
@dgolovach 
As you have a separate issue open anyway, I'll close here. I wish you good luck with your (likely different) problem!

closed 2025-01-23 by frostworx