esphome/home-assistant-voice-pe #306: iniatal update after first Home Assistant integration stops.

thogitnet, 2025-01-22:
I have put a new Home Assistant Voice into operation. As in the instructions via Phone.
_PS: In the instructions, it would be good to note that Bluetooth must be allowed for the app ..._

The update from 24.10.15 to version 25.1.0 was then started immediately. The update stopped at 16%. After more than an hour, I restarted the Home Assistant Voice by power reconnect. However, the next attempt did not get past 3% and I canceled my last attempt at 6%.

Is this behavior known?
Should I do an update with esphome ota ? If so what is the correct image ?

timkolloch, 2025-01-23:
I have a similar problem - the first 85% worked without a problem but now 1,5 hours later I am at 93% :(

fhk1407, 2025-01-26:
> I have put a new Home Assistant Voice into operation. As in the instructions via Phone. _PS: In the instructions, it would be good to note that Bluetooth must be allowed for the app ..._
>
> The update [...]

Yes, see #297 below. Same thing.

thogitnet, 2025-01-26:
I wanted to take another look at the problem and connect using the esphome logs command. When I started the update, it went through in about 60 seconds and I was able to start the setup.  Funny thing.

rlust, 2025-02-06:
Same issue I think, started new device, then it wanted to do an update. Has been 15 minutes and only 1%!