esphome/home-assistant-voice-pe #309: After play sounds nothing else happens

enx1998, 2025-01-25:
I have a HAVPE with firmware version 25.1.0 (ESPHome 2024.12.2), and too often the device recognises the wake word, plays the sound, but nothing else happens: the ring stays off, and the voice recognition doesn't start.
This is what happens

[image]

Is there anything can I do?

Thanks, ENx

BelenCaSa, 2025-01-26:
Hello
Same here, it usually happens more often after a period of inactivity. That is, once the device responds, you call it again and it responds again, however, in the mornings it gets "stuck", you have to call it at least three or four times for it to activate.

StrandmonYellow, 2025-01-28:
I saw some of this behavior also since updating a few days ago. It seems to also react less often to the wake word.

enx1998, 2025-01-28:
> I saw some of this behavior also since updating a few days ago. It seems to also react less often to the wake word.

@StrandmonYellow what firmware version do you have installed on the device?

BelenCaSa, 2025-01-29:
[image]

enx1998, 2025-02-03:
OK, I have taken control of HAVPE and I see this error:

[E][voice_assistant:776]: Error: duplicate_wake_up_detected - Duplicate wake-up detected for Okay Nabu

Same issue here: https://github.com/esphome/issues/issues/5887

Is it possible to fix this error?

Thank you

StrandmonYellow, 2025-02-04:
@enx1998 I am running the version as shown below:

[image]

However, I don't think this is correct because I have the restart button showing with my device, and this was added in a newer release than what is showing.

eikel, 2025-03-08:
I observe the same with a Home Assistant Voice PE since I have it (firmware currently at 25.3.0 (ESPHome 2025.2.0)). Raw pipeline output of a failure:
```yaml
stage: done
run:
  pipeline: 01jhwqkwt8s7dc64eqzfrfwcyb
  language: de
  conversation_id: 01JNVT6BFWWP757MVMWQ7D2GX8
events:
  - type: run-start
    data:
      pipeline: 01jhwqkwt8s7dc64eqzfrfwcyb
      language: de
      conversation_id: 01JNVT6BFWWP757MVMWQ7D2GX8
    timestamp: "2025-03-08T21:25:37.870296+00:00"
  - type: error
    data:
      code: duplicate_wake_up_detected
      message: Duplicate wake-up detected for Okay Nabu
    timestamp: "2025-03-08T21:25:37.870645+00:00"
  - type: run-end
    data: null
    timestamp: "2025-03-08T21:25:37.871023+00:00"
error:
  code: duplicate_wake_up_detected
  message: Duplicate wake-up detected for Okay Nabu
```
I also have a M5Stack Atom Echo (firmware currently at 25.2.1 (ESPHome 2025.2.0)).

enx1998, 2025-04-04:
I updated my HAVPE with the latest firmware 25.3.4 (ESPHome 2025.3.3), but the problem persists. Right now it is pretty unusable, because 90% of the time the assist pipeline doesn't start, and only a sound is heard.