esphome/home-assistant-voice-pe #310: Music (file) starts then stops

BillBlight, 2025-01-28:
Have tested with multiple formats, same result, occasionally a file will play completely but about 1 in 25

Have tried with just Media and Music Assistant, same results.

relevant log section:

`[16:01:02][D][debug:035]: ESPHome version 2024.12.4
[16:01:02][D][debug:039]: Free Heap Size: 122960 bytes
[16:01:02][D][debug:217]: Chip: Model=ESP32-S3, Features=WIFI_BGN,BLE, Cores=2, Revision=2
[16:01:02][D][debug:226]: ESP-IDF Version: 5.1.5
[16:01:02][D][debug:231]: EFuse MAC: 20:F8:3B:09:29:38
[16:01:02][D][debug:137]: Reset Reason: Software Reset Digital Core
[16:01:02][D][debug:279]: Wakeup Reason: Unknown
[16:01:29][D][media_player:080]: 'Media Player' - Setting
[16:01:29][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/c5d7ca9f337d4d6488419cfeaa49f0f6.mp3?ts=1738098089
[16:01:29][D][ring_buffer:034]: Created ring buffer with size 48000
[16:01:29][D][ring_buffer:034]: Created ring buffer with size 48000
[16:01:29][D][ring_buffer:034]: Created ring buffer with size 65536
[16:01:29][D][ring_buffer:034]: Created ring buffer with size 65536
[16:01:29][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:01:31][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:01:31][D][ring_buffer:034][speaker_task]: Created ring buffer with size 19200
[16:01:31][D][i2s_audio.speaker:111]: Starting Speaker
[16:01:31][D][i2s_audio.speaker:116]: Started Speaker
[16:01:32][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.
[16:03:36][D][media_player:080]: 'Media Player' - Setting
[16:03:36][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/c5d7ca9f337d4d6488419cfeaa49f0f6.mp3?ts=1738098216
[16:03:36][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:03:40][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:03:41][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.
[16:03:52][D][media_player:080]: 'Media Player' - Setting
[16:03:52][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/37af76f39f5e4a15b02828fb4ee05ddc.mp3?ts=1738098232
[16:03:52][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:03:54][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:03:54][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.
[16:03:58][D][media_player:080]: 'Media Player' - Setting
[16:03:58][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/be097d8151334db2bef61b3f979d11d1.mp3?ts=1738098237
[16:03:58][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:04:00][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:04:01][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.
[16:09:45][D][media_player:080]: 'Media Player' - Setting
[16:09:45][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/1bc3c244b0604927bdb5c70837aa2003.mp3?ts=1738098584
[16:09:45][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:09:46][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:10:54][D][media_player:080]: 'Media Player' - Setting
[16:10:54][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/3e56bc6baf344438bfa0c0e3abf76d6b.mp3?ts=1738098654
[16:10:54][W][component:237]: Component nabu.media_player took a long time for an operation (75 ms).
[16:10:54][W][component:238]: Components should block for at most 30 ms.
[16:10:54][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:10:56][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:11:24][D][media_player:080]: 'Media Player' - Setting
[16:11:24][D][media_player:087]:   Media URL: http://172.16.16.10:8097/flow/media_player.jarvis_1_media_player/1b07dfce94564203b42382fa4be2f6f7.mp3?ts=1738098684
[16:11:25][D][nabu_media_player.pipeline:173]: Reading MP3 file type
[16:11:27][D][nabu_media_player.pipeline:184]: Decoded audio has 2 channels, 48000 Hz sample rate, and 16 bits per sample
[16:11:29][E][nabu_media_player:299]: The media pipeline's audio decoder encountered an error.
`

efschu, 2025-04-16:
same for me

kahrendt, 2025-05-22:
Thanks for reporting! A potential fix is included in the latest beta firmware (25.5). If it continues to be a problem with that VPE firmware, then please open a new issue.

closed 2025-05-22 by kahrendt