esphome/home-assistant-voice-pe #312: Custom wake word

Felitendo, 2025-01-30:
Is it possible to train a custom wakeword with microWakeWord and then use it as a wakeword for Home Assistant Voice PE?

devinhedge, 2025-03-09:
I noticed there is several approaches to this. Have you tried [this one](https://smarthomecircle.com/custom-wake-word-for-voice-assistant-with-home-assistant)?

devinhedge, 2025-03-09:
There is also a separate request for this [here](https://github.com/esphome/home-assistant-voice-pe/issues/334) with lots of ideas.

Recommend this one be closed and difference be given to the other request.

jf-uu, 2025-03-09:
> I noticed there are several approaches to this. Have you tried [this one](https://smarthomecircle.com/custom-wake-word-for-voice-assistant-with-home-assistant)?

This approach seems to be for devices using off-device wake words (i.e. all audio is constantly streamed from the device and a separate computer performs wake word detection). The Voice PE uses microWakeWord which is on-device, so these instructions won't work.

Marcel0024, 2025-05-25:
In ESPHome builder you can take over your Voice PE and specify a new MicroWakeWord

Omega-sko, 2026-02-06:
> In ESPHome builder you can take over your Voice PE and specify a new MicroWakeWord

And how exactly do I have to do that?