esphome/home-assistant-voice-pe #313: Feature request: Allow user to add effects to the LED ring

SpencerDub, 2025-01-30:
The LED ring is such a nice feature on the device, and I'd love to add extra light effects to it so I could use them in other automations.

However, I don't want to take control of the full ESPHome configuration and lose out on future upstream automatic updates.

I don't know how this would quite work, given the structure of the ESPHome configuration, but I'd sure like it if I could just extend the existing YAML by adding extra light effects to the [user-facing LED ring](https://github.com/esphome/home-assistant-voice-pe/blob/5288dde504325130d50f9465b92a6d712203f49a/home-assistant-voice.yaml#L920) light entity, without having to adopt all 1672 lines of the current ESPHome config.

StrandmonYellow, 2025-01-31:
I adopted my HAVPE in the ESPHome builder addon. The image below is what I got when I did this. The Wifi part (power save and fast connect) I added myself to improve the connection and stability of the device. I already installed an update once and it just updated the underlying part of the configuration with my custom addons. Updating is just a case of pressing a button and waiting for it to complete. That should not be any different if you keep the automatic updates. I think this should be the best thing to do, because you can add custom parts this way without getting a huge configuration file! 

[image]

Jieper001, 2025-01-31:
A killer use case would be to be able to use the led ring as a "physical gauge", just like the gauge card in home assistant.  I would love to be able to use the led ring to visually represent my energy consumption.  I could even dream further to have it alternate between "green" colour to show when solar exceeds the house consumption etc.  Please, pretty please ;-)

SpencerDub, 2025-02-06:
> I adopted my HAVPE in the ESPHome builder addon. The image below is what I got when I did this. The Wifi part (power save and fast connect) I added myself to improve the connection and stability of [...]

Thanks for this tip!

As I linked above, the user-facing light ring entity is defined starting on [line 920 of the default YAML file](https://github.com/esphome/home-assistant-voice-pe/blob/5288dde504325130d50f9465b92a6d712203f49a/home-assistant-voice.yaml#L920). If I wanted to extend it as you describe to add effects, do you know what specifically I'd put in my ESPHome Builder YAML? I'm most curious if I need to copy this entire block:

```

  # User facing LED ring. Remapping of the internal LEDs.
  # Exposed to be used by the user.
  - platform: partition
    id: led_ring
    name: LED Ring
    entity_category: config
    icon: "mdi:circle-outline"
    default_transition_length: 0ms
    restore_mode: RESTORE_DEFAULT_OFF
    initial_state:
      color_mode: rgb
      brightness: 66%
      red: 9.4%
      green: 73.3%
      blue: 94.9%
    segments:
      - id: leds_internal
        from: 7
        to: 11
      - id: leds_internal
        from: 0
        to: 6
```
and then append an `effects: ` dictionary with my desired effects, or if I could do it by simply copying a portion of it, like so: 

```
  # User facing LED ring. Remapping of the internal LEDs.
  # Exposed to be used by the user.
  - platform: partition
    id: led_ring
    name: LED Ring
    effects:
      #...
```
In other words, I'm wondering if user-specified config in ESPHome Builder _appends_ or _overrides_ the loaded source YAML, and if it appends, how specific do I need to be?

SpencerDub, 2025-02-07:
err, maybe I should close as "not planned?"

Not trying to imply anything about the direction of the project, I just have a satisfactory answer. If a maintainer thinks "not planned" is a more appropriate close type, go for it.

closed 2025-02-07 by SpencerDub

StrandmonYellow, 2025-02-07:
Ah you already found the answer :) I agree it would be a great addon, but I think the dev team should decide the priority for this indeed.