esphome/home-assistant-voice-pe #314: Allow customization of TTS timeout

dividebysandwich, 2025-01-31:
As per discussion at https://community.home-assistant.io/t/http-timeout-for-voice-assistant-pe-even-though-the-response-is-recieved/834200/4 I would like to request some means to configure the timeout, since 5 seconds tends to be problematic when the speech generation takes longer.

jonathanarchergithub, 2025-02-03:
+1. having a timeout issue for responses longer Rhabarber a single sentence with local Piper on a NUC as well. Increasing the timeout in the code worked for me. Made me wonder hey it takes so Long though. According to the Voice debug, the whole pipeline executes in less that 5s. TTS almost 0s.

MorganBridget, 2025-02-03:
> +1. having a timeout issue for responses longer than a single sentence with local Piper on a NUC as well. Increasing the timeout in the code worked for me. Made me wonder why it takes so long though. According to the Voice debug, the whole pipeline executes in less that 5s. TTS almost 0s.

Can you provide a little bit detail of how to solve this. I've read in other threads about nabu custom compoenent and adding a config to ESPhome and I'm very confused. Any help you can provide is appreciated

MorganBridget, 2025-02-03:
> As per discussion at https://community.home-assistant.io/t/http-timeout-for-voice-assistant-pe-even-though-the-response-is-recieved/834200/4 I would like to request some means to configure the timeout, since 5 seconds tends to be problematic when the speech generation takes longer.

Can you provide a little bit detail of how to solve this. I've read in other threads about nabu custom compoenent and adding a config to ESPhome and I'm very confused. Any help you can provide is appreciated

jonathanarchergithub, 2025-02-03:
> Can you provide a little bit detail of how to solve this. I've read in other threads about nabu custom compoenent and adding a config to ESPhome and I'm very confused. Any help you can provide is [...]

I just followed the instructions gyrga posted. Effectively you have to create a custom nabu component where you change the code. Then you need to recompile the FW. Is a bit tedious…

MorganBridget, 2025-02-03:
> I just followed the instructions gyrga posted. Effectively you have to create a custom nabu component where you change the code. Then you need to recompile the FW. Is a bit tedious…

I don't mind if it's tedious. I just don't know what FW means and how to recompile it. Can you have mercy on me and maybe walk me through?

jonathanarchergithub, 2025-02-04:
Well, FW=Firmware (I.e. the Software operating on the devices micro controller).

You would need to setup ESPhome and add the Voice PE as a device there. Then you download the code of the nabu component to your local file system and proceed as gyrga describes.

In any case you would have to make yourself familiar with ESPhome and you lose automatic SW updates of your device.

MorganBridget, 2025-02-04:
> Well, FW=Firmware (I.e. the Software operating on the devices micro controller).
>
> You would need to setup ESPhome and add the Voice PE as a device there. Then you download the code of the nabu [...]

I saw their instructions and I don't understand where you're supposed to put the config code. I already have esphome with Voice PE there. That's how I know the timeout problem exists. That's what my situation is right now. I'm trying to fix it and I still am unclear. It just not helping to reference gyrga without further explanation because unfortunately it's not clear to me. I probably need a step by step guidance on this one if you can spare a moment of your time that would be very helpful to me as well as others I am certain.  Thank you so much.

jonathanarchergithub, 2025-02-04:
Ah, okay I see. You can simply add a new folder called e.g. “custom_components” below your ESPhome folder. There you copy the component “nabu” from this repository. In the code apply gyrgas change. In your device config you add to the beginning of the file:

```
custom_components:
  - source:
     type: local
     path: /config/esphome/custom_components
```

Then Save and Install.

RobMeades, 2025-02-20:
Just a note that, with the changes committed yesterday in b3f07f0cfadd71e8f524be5ce107c3e8e06f672e, the file being discussed above no longer exists, instead a generic audio component has been added over in the `esphome` repo:

https://github.com/esphome/esphome/blob/dev/esphome/components/audio/audio_reader.cpp

...and so the line that needs to be modified is [there](https://github.com/esphome/esphome/blob/dev/esphome/components/audio/audio_reader.cpp#L100) and not in this repo at all.

StrandmonYellow, 2025-03-03:
Ah so this is the issue why it is not playing longer responses. That makes sense. It is a bit annoying that I need to fix it myself tho..

jonathanarchergithub, 2025-03-04:
The latest change in the esphome audio component seems to have fixed the timeout issue for me.