esphome/home-assistant-voice-pe #315: PE device has no TTS responses (server SSL)

abliss, 2025-02-01:
My symptoms seem very similar to #238 but I noticed that was diagnosed as a lack of TLS1.3 support on the device and closed.

I also don't get TTS responses on my Voice Preview Edition. I believe I followed the instructions at https://community.home-assistant.io/t/how-to-configure-lets-encrypt-ssl-certificates-for-home-assistant-completely-100-free-updated-for-2022-2023/508329 to use letsencrypt through the DuckDNS addon .

I noticed that at https://www.home-assistant.io/integrations/http/#ssl_profile it claims that the default profile is "modern", where the linked Mozilla page says

Modern: Modern clients that support TLS 1.3, with no need for backwards compatibility
Intermediate: Recommended configuration for a general-purpose server

So maybe "Intermediate" should be the default? However, I tried setting it to "intermediate" and saw no difference. Whether on "intermediate" or the default "modern", I also note that from a command line I can do "curl --tls-max 1.2" and the negotiation works fine. So perhaps the root cause is not in fact an inability of Home Voice Preview Edition to negotiate a TLS version? Maybe the issue is with the server's cipher suite (or something about the LetsEncrypt cert), rather than the TLS version itself?

When I curl with -vvv --tls-max 1.2 I see SSL connection using TLSv1.2 / ECDHE-ECDSA-AES256-GCM-SHA384 With 1.3 I see SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384. 

I can use the device to set a timer, and after the timer goes off an alert chimes, so I know the microphone/speaker/network are working. But any response required text-to-speech doesn't seem to come through. 

Also in the HA android app, the onboarding wizard fails at the "customize voice assistant" stage, saying the device cannot connect to the server.

zackslash, 2025-02-01:
I'm having a similar issue with responses, my HA server is also HTTPS via a TLS 1.3 cert, I also noticed that it looks like support for TLS 1.3 [has been added to the firmware some time ago](https://github.com/esphome/home-assistant-voice-pe/pull/244).

Have you ever had voice responses? For me they are more intermiment working ~70% of the time.

abliss, 2025-02-01:
I did have some responses early on, but lately they seem to never work. I may have messed up a setting somewhere.

zackslash, 2025-02-03:
I am still experiencing this issue on all my devices, i've 'taken control' via ESPHome and can now see the following logs (Edited to remove my HA url):

```
[19:20:06][D][light:109]:   Effect: 'Replying'
[19:20:06][D][voice_assistant:641]: Event Type: 8
[19:20:06][D][voice_assistant:719]: Response URL: "https://<SNIP>.ui.nabu.casa/api/tts_proxy/MuaZqebpEUGD_aZPGt29nw.flac"
[19:20:06][D][voice_assistant:515]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[19:20:06][D][voice_assistant:522]: Desired state set to STREAMING_RESPONSE
[19:20:06][D][media_player:080]: 'Media Player' - Setting
[19:20:06][D][media_player:087]:   Media URL: https://<SNIP>.ui.nabu.casa/api/tts_proxy/MuaZqebpEUGD_aZPGt29nw.flac
[19:20:06][D][media_player:093]:  Announcement: yes
[19:20:06][D][voice_assistant:641]: Event Type: 2
[19:20:06][D][voice_assistant:733]: Assist Pipeline ended
[19:20:06][D][esp-idf:000][ann_read]: E (89588) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7280

[19:20:06][D][esp-idf:000][ann_read]: E (89591) esp-tls: Failed to open new connection

[19:20:06][D][esp-idf:000][ann_read]: E (89596) transport_base: Failed to open a new connection

[19:20:06][D][esp-idf:000][ann_read]: E (89602) HTTP_CLIENT: Connection failed, sock < 0

[19:20:06][E][nabu_media_player.pipeline:171]: Media reader encountered an error: ESP_ERR_HTTP_CONNECT
[19:20:06][E][nabu_media_player:305]: The announcement pipeline's file reader encountered an error.
[19:20:06][D][voice_assistant:515]: State changed from STREAMING_RESPONSE to IDLE
[19:20:06][D][voice_assistant:522]: Desired state set to IDLE
```

I am suprised that the device is attempting to use the remote HA URL rather than the URL for my local network, im unsure yet as to the cause of the issue. Continuing to investigate.

zackslash, 2025-02-03:
It seems to me that the firmware still takes issue with TLS 1.3 certificates, the 'nabu.casa' TLS is of version 1.3 and so the device fails to read the audo.

xpac1985, 2025-02-04:
Thanks for creating the issue and sharing your observations ♥
I can only add that I see the same issue - but only sometimes, and I couldn't see a pattern yet.
Often, when I haven't used my VoicePE for some time, the first command gets a reply, but all following commands don't work. But then, sometimes even the first command fails. Or multiple work. I've now changed my Internet URL to my local URL and will see if this works reliably or not.

jonofe, 2025-02-05:
> Going into **Config** -> **Network** -> and setting the **Home Assistant URL 'Internet'** URL to my local TLS URL seems to resolve this issue.

This does not work for me. It doesn't change the faulty behaviour.

> Often, when I haven't used my VoicePE for some time, the first command gets a reply, but all following commands don't work. But then, sometimes even the first command fails.

That's exactly what I observe as well.

Really annoying situation. Was there any reaction from Nabu Casa on that problem?

xpac1985, 2025-02-26:
> I've now changed my Internet URL to my local URL and will see if this works reliably or not.

Can confirm that after making that change, it worked reliably for 3 weeks now with 2 different Voice PE devices.
I'd still consider it a bug, but for me that is a valid workaround in the meantime.

dlmorgan999, 2025-02-28:
Changing the _Internet_ URL worked for me as well.  I'm a but surprised we haven't seen anything official from Nabu Casa on this yet.

davidjmeier, 2025-03-11:
I have the same issue and am a user of HomeAssistant Cloud - so I can't reliably change the Internet URL without breaking mobile connectivity.  This is really frustrating because it's not listed in any FAQ or elsewhere related to PE.  I bought a few of these devices and compared to other Voice hardware that is working - this is not reliable at all.  I might get 10-20% of audible responses from the PE device.  Far from usable.

Beyond that, the fact that this isn't listed anywhere in easy to find documentation led me down a rabbit hole for a few hours today since I do have other devices that are being used in an identical manner and work fine.

davidjmeier, 2025-03-12:
Hey, all - here's an update.

First and foremost - I opened a ticket with Nabu Casa and the tech support person I got is not only rude, but really had no skills in troubleshooting.  In fact they responded with this exact language:

```
Let's get this out of the way: I use several Voice PE with the Cloud TTS option without issue as do many others. The issue is not the Cloud URL, the Cloud services, or the Voice PE. You've asked why it isn't working for you in each reply. I don't have an answer for that. This is what we are exploring: the settings you have and the setup you have.
```

So, the support person (to remain nameless) was not only rude, but had no clue what they were talking about.  Again, very annoying given being a paying customer and how awesome the community is otherwise.  And if anyone from Nabu Casa is reading this feel free to check out ticket: 33201.

Anyway - on to the fix!

If someone else can test this fix - that would be amazing.

Here's my redacted config as a base for you:
```
substitutions:
  name: ha-voice-xxxx
  friendly_name: ha-voice-xxxx
packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  platformio_options:
    board_build.sdkconfig_defaults: sdkconfig.defaults
api:
  encryption:
    key: !secret your-key-here
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.x.x.x
    gateway: 10.x.x.x
    subnet: 255.255.255.0
    dns1: 10.x.x.x
```

Feel free to remove the manual ip configuration if you're doing it in another way.  The important part is the `board_build.sdkconfig_defaults: sdkconfig.defaults`.

You'll need access to the file system - the easiest way is via `Studio Code Server` or something else.  But in your `Config` file structure you should see `esphome`.  Create a file in the root of `esphome` called: `sdkconfig.defaults`.

That file should contain the following:
```
CONFIG_MBEDTLS_SSL_PROTO_TLS1_2=y
CONFIG_MBEDTLS_SSL_MIN_MINOR_VERSION=3
CONFIG_MBEDTLS_SSL_MAX_MINOR_VERSION=3
```

...as this is indeed a TLS1.3 issue as was a problem in the other Github issue.

Now go into `ESPHome Builder` and click on the 3 dots for the HA Voice instance and `Clean Build Files`, then `Validate` and then `Install`.

The first time I prompted the pipeline I got a could not connect error, but I think it was not fully running.  All subsequent requests now work as expected.

Here's a log as proof:
```
[16:54:31][D][voice_assistant:509]: State changed from STOPPING_MICROPHONE to AWAITING_RESPONSE
[16:54:31][D][voice_assistant:509]: State changed from AWAITING_RESPONSE to AWAITING_RESPONSE
[16:54:31][D][voice_assistant:646]: Event Type: 4
[16:54:31][D][voice_assistant:674]: Speech recognised as: "Feels like."
[16:54:31][D][voice_assistant:646]: Event Type: 5
[16:54:31][D][voice_assistant:679]: Intent started
[16:54:31][D][voice_assistant:646]: Event Type: 6
[16:54:31][D][voice_assistant:646]: Event Type: 7
[16:54:31][D][voice_assistant:704]: Response: "The current feels like temperature at home is 56."
[16:54:31][D][light:036]: 'voice_assistant_leds' Setting:
[16:54:31][D][light:051]:   Brightness: 30%
[16:54:31][D][light:109]:   Effect: 'Replying'
[16:54:31][D][voice_assistant:646]: Event Type: 8
[16:54:31][D][voice_assistant:726]: Response URL: "https://redacted.ui.nabu.casa/api/tts_proxy/****.flac"
[16:54:31][D][voice_assistant:509]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[16:54:31][D][voice_assistant:516]: Desired state set to STREAMING_RESPONSE
[16:54:31][D][media_player:073]: 'Media Player' - Setting
[16:54:31][D][media_player:080]:   Media URL: https://redacted.ui.nabu.casa/api/tts_proxy/****.flac
[16:54:31][D][media_player:086]:  Announcement: yes
[16:54:31][D][speaker_media_player:420]: State changed to ANNOUNCING
[16:54:31][D][voice_assistant:646]: Event Type: 2
[16:54:31][D][voice_assistant:742]: Assist Pipeline ended
[16:54:32][D][esp-idf:000][ann_read]: I (71771) esp-x509-crt-bundle: Certificate validated

[16:54:33][D][ring_buffer:034][ann_read]: Created ring buffer with size 1000000
[16:54:33][D][speaker_media_player.pipeline:114]: Reading FLAC file type
[16:54:33][D][speaker_media_player.pipeline:124]: Decoded audio has 1 channels, 48000 Hz sample rate, and 16 bits per sample
[16:54:33][D][ring_buffer:034]: Created ring buffer with size 9600
[16:54:33][D][speaker_mixer:306]: Starting speaker mixer
[16:54:33][D][speaker_mixer:314]: Started speaker mixer
[16:54:37][D][speaker_media_player:420]: State changed to IDLE
[16:54:37][D][voice_assistant:509]: State changed from STREAMING_RESPONSE to RESPONSE_FINISHED
[16:54:37][D][voice_assistant:516]: Desired state set to RESPONSE_FINISHED
[16:54:37][D][voice_assistant:509]: State changed from RESPONSE_FINISHED to IDLE
[16:54:37][D][voice_assistant:516]: Desired state set to IDLE
```

So, pretty clear that this is 100% still a TLS 1.3 issue.

cc: @abliss @Destarianon - in case you wanted to test as well.

cogneato, 2025-03-12:
For ESPHome devices like the VoicePE, HA should not be configured with ssl using the `http:` integration. Consider using a reverse proxy instead and removing the ssl configuration from configuration.yaml. The local url on the network page can then be set using the local address and port (ie, http://192.168.1.50:8123)

Basically the Internet URL and Local URL need to be set to whatever works to reach HA without seeing a warning in the browser.

Then the VPE should "just work" locally.

davidjmeier, 2025-03-12:
> For ESPHome devices like the VoicePE, HA should not be configured with ssl using the `http:` integration. Consider using a reverse proxy instead and removing the ssl configuration from configuration.yaml. The local url on the network page can then be set using the local address and port (ie, http://192.168.1.50:8123)
> 
> Basically the Internet URL and Local URL need to be set to whatever works to reach HA without seeing a warning in the browser, and the local URL should not be using https://
> 
> Then the VPE should "just work" locally.

This has nothing to do with my configuration.  You must be the person I was interacting with in support, but - if you actually looked at what I posted above there is no "http:" integration in my config.  So, please refrain from posting things that have no bearing on what we've identified is the root problem.  It has zero to do with what you've posted and you're still not following the actual issue.

Also, as I told you in the support thread - there are no browser warnings with my local HA instance.  In fact it also presents valid `Let's Encrypt` certs to me in my local network almost identically to how the HA Cloud does (which also uses LE).

It doesn't "just work" because there is an isue with the `mbedtls` module and TLS 1.3 which is why my workaround posted above works.

xpac1985, 2025-03-13:
I'm by no means a pro on certificates and crypto algorithms, but I noticed that both my local HASS (HA OS, using LetsEncrypt AddOn) and the Nabu Casa Cloud "Forwarding Service" offer TLS 1.3. Local works fine, cloud fails, as many of us noticed - so it's not generally an issue with TLS 1.3.

I ran testssl.sh against both FQDNs, and the most notable difference in the output was that the Nabu Casa service uses a RSA 2048 bit key (and therefore offers RSA ciphers), while my local HA uses an ECDSA key (and therefore offers ECDSA ciphers).

Therefore, I configured the LetsEncrypt addon to use an RSA key, re-issued the cert, tested my Voice PE again - and it still works 😕
Ran testssl.sh again, compared the output against the Nabu Casa result, and at that point there was pretty much no more difference, besides my local cert using a single wildcard SAN, while the Nabu Casa cert has two specific FQDNs (my CNAME and their domain).

I'll add all three outputs here, hoping that it either helps someone to spot something, or at least save them some time when investigating.

testssl.sh against Nabu Casa URL
```
 Start 2025-03-13 00:15:42        -->> 3.69.134.24:443 (hass-from-internet.mydomain.net) <<--

 rDNS (3.69.134.24):     ec2-3-69-134-24.eu-central-1.compute.amazonaws.com.
 Service detected:       HTTP

 Testing protocols via sockets except NPN+ALPN

 SSLv2      not offered (OK)
 SSLv3      not offered (OK)
 TLS 1      not offered
 TLS 1.1    not offered
 TLS 1.2    offered (OK)
 TLS 1.3    offered (OK): final
 NPN/SPDY   not offered
 ALPN/HTTP2 not offered

 Testing cipher categories

 NULL ciphers (no encryption)                  not offered (OK)
 Anonymous NULL Ciphers (no authentication)    not offered (OK)
 Export ciphers (w/o ADH+NULL)                 not offered (OK)
 LOW: 64 Bit + DES, RC[2,4] (w/o export)       not offered (OK)
 Triple DES Ciphers / IDEA                     not offered
 Obsolete CBC ciphers (AES, ARIA etc.)         offered
 Strong encryption (AEAD ciphers)              offered (OK)

 Testing robust (perfect) forward secrecy, (P)FS -- omitting Null Authentication/Encryption, 3DES, RC4

 PFS is offered (OK)          TLS_AES_256_GCM_SHA384 TLS_CHACHA20_POLY1305_SHA256 ECDHE-RSA-AES256-GCM-SHA384 ECDHE-RSA-AES256-SHA384 ECDHE-RSA-CHACHA20-POLY1305 TLS_AES_128_GCM_SHA256 ECDHE-RSA-AES128-GCM-SHA256 ECDHE-RSA-AES128-SHA256
 Elliptic curves offered:     prime256v1 secp384r1 secp521r1 X25519 X448
 Finite field group:          ffdhe2048 ffdhe3072 ffdhe4096 ffdhe6144 ffdhe8192

 Testing server preferences

 Has server cipher order?     yes (OK) -- TLS 1.3 and below
 Negotiated protocol          TLSv1.3
 Negotiated cipher            TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Cipher order
    TLSv1.2:   ECDHE-RSA-AES256-GCM-SHA384 ECDHE-RSA-CHACHA20-POLY1305 ECDHE-RSA-AES128-GCM-SHA256 ECDHE-RSA-AES256-SHA384 ECDHE-RSA-AES128-SHA256
    TLSv1.3:   TLS_AES_256_GCM_SHA384 TLS_CHACHA20_POLY1305_SHA256 TLS_AES_128_GCM_SHA256

 Testing server defaults (Server Hello)

 TLS extensions (standard)    "renegotiation info/#65281" "EC point formats/#11" "session ticket/#35" "supported versions/#43" "key share/#51" "max fragment length/#1"
                              "encrypt-then-mac/#22" "extended master secret/#23"
 Session Ticket RFC 5077 hint 7200 seconds, session tickets keys seems to be rotated < daily
 SSL Session ID support       yes
 Session Resumption           Tickets: yes, ID: yes
 TLS clock skew               Random values, no fingerprinting possible
 Signature Algorithm          SHA256 with RSA
 Server key size              RSA 2048 bits
 Server key usage             Digital Signature, Key Encipherment
 Server extended key usage    TLS Web Server Authentication, TLS Web Client Authentication
 Serial                       03B3583481874A171C8EBD539609FB7DE28C (OK: length 18)
 Fingerprints                 SHA1 87815E1572B4EBBEFCE0D8963C1BC6DD0A589CE3
                              SHA256 C6BD77944263D3FE42DD45ACA503B71841AC0C8F089AE24211589F2D0679B4E3
 Common Name (CN)             somewildstring.ui.nabu.casa  (request w/o SNI didn't succeed)
 subjectAltName (SAN)         hass-from-internet.mydomain.net somewildstring.ui.nabu.casa
 Issuer                       R11 (Let's Encrypt from US)
[... 404 lines cut ...]
 IE 6 XP                      No connection
 IE 8 Win 7                   No connection
 IE 8 XP                      No connection
 IE 11 Win 7                  TLSv1.2 ECDHE-RSA-AES256-SHA384, 256 bit ECDH (P-256)
 IE 11 Win 8.1                TLSv1.2 ECDHE-RSA-AES256-SHA384, 256 bit ECDH (P-256)
 IE 11 Win Phone 8.1          TLSv1.2 ECDHE-RSA-AES128-SHA256, 256 bit ECDH (P-256)
 IE 11 Win 10                 TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 256 bit ECDH (P-256)
 Edge 15 Win 10               TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 253 bit ECDH (X25519)
 Edge 101 Win 10 21H2         TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Safari 12.1 (iOS 12.2)       TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Safari 13.0 (macOS 10.14.6)  TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Safari 15.4 (macOS 12.3.1)   TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Java 7u25                    No connection
 Java 8u161                   TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 256 bit ECDH (P-256)
 Java 11.0.2 (OpenJDK)        TLSv1.3 TLS_AES_256_GCM_SHA384, 256 bit ECDH (P-256)
 Java 17.0.3 (OpenJDK)        TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
[... cut ...]
 OpenSSL 1.0.2e               TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 256 bit ECDH (P-256)
 OpenSSL 1.1.0l (Debian)      TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 253 bit ECDH (X25519)
 OpenSSL 1.1.1d (Debian)      TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 OpenSSL 3.0.3 (git)          TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
 Apple Mail (16.0)            TLSv1.2 ECDHE-RSA-AES256-GCM-SHA384, 256 bit ECDH (P-256)
 Thunderbird (91.9)           TLSv1.3 TLS_AES_256_GCM_SHA384, 253 bit ECDH (X25519)
```

davidjmeier, 2025-03-13:
So, here's how I see the Voice PE working in my network:
-  I am a paid subscriber to HA Cloud.  So my `Internet` URL is set to `Use Home Assistant Cloud` - this is automatically set and my instance is provisioned with a `https://redacted.ui.nabu.casa` URL.  Where `redacted` is replaced by a random string.
- My local HA instance also uses LE for cert and I have local DNS resolution that resolves appropriately everywhere.

Under `Voice Assistants` my setup is configured to prefer HA Cloud, currently.

The way this works is that when the pipeline runs the TTS is requested from HA Cloud.  The URL of the flac file (e.g. https://redacted.ui.nabu.casa/api/tts_proxy/****.flac) is sent to the Voice PE and it downloads the FLAC file directly.  The file never hits the HA instance.  You can see this in the logs I provided:

```
[16:54:31][D][voice_assistant:726]: Response URL: "https://redacted.ui.nabu.casa/api/tts_proxy/****.flac"
[16:54:31][D][voice_assistant:509]: State changed from AWAITING_RESPONSE to STREAMING_RESPONSE
[16:54:31][D][voice_assistant:516]: Desired state set to STREAMING_RESPONSE
[16:54:31][D][media_player:073]: 'Media Player' - Setting
[16:54:31][D][media_player:080]:   Media URL: https://redacted.ui.nabu.casa/api/tts_proxy/****.flac
[16:54:31][D][media_player:086]:  Announcement: yes
[16:54:31][D][speaker_media_player:420]: State changed to ANNOUNCING
[16:54:31][D][voice_assistant:646]: Event Type: 2
[16:54:31][D][voice_assistant:742]: Assist Pipeline ended
[16:54:32][D][esp-idf:000][ann_read]: I (71771) esp-x509-crt-bundle: Certificate validated
```

In this snapshot the Voice PE is being provided the URL to get the TTS response.  This is indicated by `Response URL`.  Then  the Voice PE sets the `Media URL` and goes to grab the file from that location directly.  The Voice PE never touches the local HA instance here - it is literally grabbing the TTS flac file from HA Cloud and playing it after download.

However, in the broken firmware the Voice PE _can't connect to HA Cloud directly_.  It has nothing to do with my local certs because I never see, in the pipeline, the Voice PE reaching out to my HA instance.  I can actually see all this traffic in my firewall - I just did a packet capture of my Voice PE and it initiates the connection to `52.23.103.22` directly  and I see the stream for the media download happen.

As I stated before - this is due to the `mbedtls` module.

By chance did you try my workaround in the Voice PE firmware @xpac1985 ?  Because when you disable TLS 1.3 in the Voice PE it works 100% of the time.  It is the way the Voice PE software is handling the certificate provided by HA Cloud.

I'm glad I found a fix / workaround - because now these devices all work.  But, this doesn't actually fix the problem.

Also, I attached a screenshot of the TLS connection directly from my Voice PE to HA Cloud.  You'll note that it selected 1.2 even though the server advertised 1.3 because I took that out of the Voice PE.  This shows the Voice PE contacting the HA Cloud directly to snag the flac file.  There is zero issues with my local instance in this because the Voice PE isn't even trying to grab the file from it.  It's going direct to the `Response URL` provided and loading it directly.

[image]

davidjmeier, 2025-03-13:
There's also a related thread here that looks to indicate folks who enabled TLS 1.3 support in `mbedtls` broke many things by default by failing to call `psa_crypto_init` when starting the handshake.  This is where I see the failure.  Curious if someone responsible for the firmware can validate this is being done correctly?

https://github.com/Mbed-TLS/mbedtls/issues/9223

jonofe, 2025-03-13:
> platformio_options:
>     board_build.sdkconfig_defaults: sdkconfig.defaults

I also trid this. In the first run it was not successfull, i still got an error in the logs.
As I am using an own CA with custom vertificates I tried to add:

`CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE=yes
CONFIG_MBEDTLS_CUSTOM_CERTIFICATE_BUNDLE_PATH=/ssl/ca-FELD.crt`

This di not change anything and I suppose this couldn't be the solution as the local URL is not used at all.
Then I had a walk with my dog and when I came back, and waht should I say: now it works.

I can't imagine, why it took longer until it worked, maybe some caching or something like that.

So I can now confirm, that the solution mentioned [here](https://github.com/esphome/home-assistant-voice-pe/issues/315#issuecomment-2719222169) worked for me.

Thanks to @davidjmeier.

jonofe, 2025-03-13:
Maybe I was to quick tto confirm that it works. After my last post, I tried again and got the following error:

 [10:03:04][D][voice_assistant:742]: Assist Pipeline ended
 [10:03:05][D][esp-idf:000][ann_read]: E (3188340) **esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7280**
 [10:03:05]
 [10:03:05][D][esp-idf:000][ann_read]: E (3188341) **esp-tls: Failed to open new connection**
 [10:03:05]
 [10:03:05][D][esp-idf:000][ann_read]: E (3188342) **transport_base: Failed to open a new connection**
 [10:03:05]
 [10:03:05][D][esp-idf:000][ann_read]: E (3188343) **HTTP_CLIENT: Connection failed, sock < 0**
 [10:03:05]
 [10:03:05][E][speaker_media_player.pipeline:112]: **Media reader encountered an error: ESP_ERR_HTTP_CONNECT**
 [10:03:05][D][speaker_media_player:420]: State changed to IDLE

davidjmeier, 2025-03-13:
> Maybe I was to quick tto confirm that it works. After my last post, I tried again and got the following error:
> 
> [10:03:04][D][voice_assistant:742]: Assist Pipeline ended [10:03:05][D][esp-idf:000][ann_read]: E (3188340) **esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7280** [10:03:05] [10:03:05][D][esp-idf:000][ann_read]: E (3188341) **esp-tls: Failed to open new connection** [10:03:05] [10:03:05][D][esp-idf:000][ann_read]: E (3188342) **transport_base: Failed to open a new connection** [10:03:05] [10:03:05][D][esp-idf:000][ann_read]: E (3188343) **HTTP_CLIENT: Connection failed, sock < 0** [10:03:05] [10:03:05][E][speaker_media_player.pipeline:112]: **Media reader encountered an error: ESP_ERR_HTTP_CONNECT** [10:03:05][D][speaker_media_player:420]: State changed to IDLE
> 
> Any idea on that?

That's the original error.  A couple of things here - unless you need a custom cert bundle take that out.  If you're using LE you don't need it.

Second is **_make sure you clean the build files first!_**  If you skipped that part there's a high liklihood of reusing a component from a prior build.  You need to start from scratch and rebuild everything.  When I first tried the custom build config it didn't work.  Then I cleaned and rebuilt and it has now been completely reliable.  I've executed the workflow about 20-30 times using both HA Cloud and local for TTS.  No issues and no more workarounds.

davidjmeier, 2025-03-13:
I just tested this again...  Here's what happens at the packet level on a good session and a bad session.  Note that the Nabu Casa server responds with a `FIN` after the `Client Hello`.  So, they are tearing down the session due to something not working correctly with the TLS setup.

Not working:
[image]

Working:

[image]

xpac1985, 2025-03-13:
> By chance did you try my workaround in the Voice PE firmware [@xpac1985](https://github.com/xpac1985) ? Because when you disable TLS 1.3 in the Voice PE it works 100% of the time. It is the way the Voice PE software is handling the certificate provided by HA Cloud.

Nope, I didn't feel like self-building this for now, my device is running with the unmodified firmware. I didn't want to break it yesterday late :D

> The way this works is that when the pipeline runs the TTS is requested from HA Cloud. The URL of the flac file (e.g. https://redacted.ui.nabu.casa/api/tts_proxy/****.flac) is sent to the Voice PE [...]

I think that's a misinterpretation. Per my understanding, it uses the Nabu Casa URL to use their tunnel back to your local instance, and get the file from there. TTS downloads the result file from the cloud service, and then offers it via that api/tts_proxy URL. But yeah, in that case the cert on your local instance isn't really involved.

> [10:03:05][D][esp-idf:000][ann_read]: E (3188340) **esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7280**

After searching through some other mbedtls related issues, that seems to be the code for "Network connection error", makes sense if the other side FIN-closes that connection early.

> I just tested this again... Here's what happens at the packet level on a good session and a bad session. Note that the Nabu Casa server responds with a `FIN` after the `Client Hello`. So, they are tearing down the session due to something not working correctly with the TLS setup.

Okay, now it gets super confusing.
I've no idea why this happens. Everything but that Client Hello packet contains zero data related to TLS, it's all TCP connection handshake/teardown. In that Client Hello, the Voice PE/mbedtls advertises what it wants/supports, and that's enough for the Nabu Casa server to immediately close the connection. At that point, the server has said zero about it's supported protocols/ciphers/etc.
However, I compared the two Client Hellos line by line, and they're effectively the same (as far as the Wireshark encoded details are concerned). Interestingly, Voice PE offers TLS 1.3 + TLS 1.2 in both Client Hellos. Same count of ciphers, everything seems to be the same. Still, in one scenario the server immediately bugs out, in the other it's fine.
BTW, I think Wireshark decodes one conversation as TLS 1.2, the other as TLS 1.3 based on stuff that happens later in the connection. If you only export the first four packets from the working .pcap, and look at that in Wireshark, it will likely consider that to be TLS 1.2 too.

So - we have now determined it's the server immediately closing the connection, but I can't see a reason in the few bytes of data it has received at that point. I'm a little lost.

davidjmeier, 2025-03-13:
> I think that's a misinterpretation. Per my understanding, it uses the Nabu Casa URL to use their tunnel back to your local instance, and get the file from there. TTS downloads the result file from [...]

This is incorrect.  If you take a PCAP you'll see, very directly, that the file is being downloaded by the Voice PE.  It never hits the local HA instance.  TTS doesn't download it in my case and this is evident by the PCAP.  I've logged all connections from the Voice PE and I am 100% certain at this point.

> So - we have now determined it's the server immediately closing the connection, but I can't see a reason in the few bytes of data it has received at that point. I'm a little lost.

I think I found the culprit...

In the failing TLS session, the client offers TLS 1.2 but includes TLS 1.3 extensions (key_share, psk_key_exchange_modes).
The server is probably rejecting the handshake due to these unexpected extensions when negotiating TLS 1.2.  Because this is the *only* time the connection fails.  I'm going to test something now to validate that - but it would describe why right after the client initiates the connection we get a FIN back from HA Cloud.  This is definitely in the firmware of the Voice PE if that's the case.

xpac1985, 2025-03-13:
> This is incorrect. If you take a PCAP you'll see, very directly, that the file is being downloaded by the Voice PE. It never hits the local HA instance. TTS doesn't download it in my case and this [...]

You wouldn't see this in the PCAP, because Voice PE connects to Nabu Casa, as you can see. However, that URL is a reverse proxy back to your local instance (which you can't see in your PCAP). So from a TCP point of view, it's two connections - Voice PE to Nabu Casa, and Nabu Casa to local HA. That wouldn't be visible in the PCAP.
You can verify that by running any voice command, then going to your HA instance, Voice, select Debug on your assistant, and on the bottom it has `Raw`.
At the very end of that, there'll be a `url: /api/tts_proxy/whatever.mp3` (Why do I get mp3 and you get .flac?)
If you append that piece to your local HA url (like https://myhass.home.net/api/tts_proxy/whatever.mp3) it will download the generated audio reply file from your local box.

davidjmeier, 2025-03-13:
> You wouldn't see this in the PCAP, because Voice PE connects to Nabu Casa, as you can see. However, that URL is a reverse proxy back to your local instance (which you can't see in your PCAP). So [...]

Again, not correct.  I can see the Voice PE download the file directly.  I have both the PCAPs from my Voice PE perspective as well as my local HA instance.  The HA instance never requests the file.  I am 1000% certain.

xpac1985, 2025-03-13:
> In the failing TLS session, the client offers TLS 1.2 but includes TLS 1.3 extensions (key_share, psk_key_exchange_modes).
> The server is probably rejecting the handshake due to these unexpected [...]

But it does this in both PCAPs? It also offers TLS 1.2 + 1.3 on both? The only difference I can see (visually) is that in the failing connection, the line `Random: ab...` has an arrow to extend for further information, while the other Client Hello does not have that.
Could you extend that to see what is hidden there?

jonofe, 2025-03-13:
I changed it back to exactly the proposed configuration, cleaned the build files, validated and installed.
Then I'm always trying the command "ok nabu, what time is it?"
First try not successfull, second was successfull, all subsequent tries were not successfull.
I always get the error mentioned in my last post.

Only config which is working for me, to set the external URL to my internal HTTPS-URL and then add 

> http_request:
>   verify_ssl: false

But I don't know for what the external URL is used in addition? Therefore I wanted a clean configuration, which works with the nabu-URL as my external URL.

xpac1985, 2025-03-13:
> Again, not correct. I can see the Voice PE download the file directly. I have both the PCAPs from my Voice PE perspective as well as my local HA instance. The HA instance never requests the file. I am 1000% certain.

Do you see that existing tunnel between your local HA instance and Nabu Casa? That's effectively a reverse proxy from the somerandomstring.ui.naba.casa back to your instance.

> 
> EDIT I also know this because the SNI for the HA Cloud is much different than the internal DNS I use. So it's very clear what host the Voice PE is talking to.

As I said - yes, your Voice PE connects to the HA Cloud service. Yes, that's why you see that SNI in the Client Hello. Still, HA Cloud has a standing connection back to your local instance, and that's where it proxies that request back to. I can see that on every voice request I make, my HA immediately makes an outbound connection to pull the TTS result file from the Nabu Casa service and it store it locally. I could confirm this with the trick I mentioned above.

davidjmeier, 2025-03-13:
> Do you see that existing tunnel between your local HA instance and Nabu Casa? That's effectively a reverse proxy from the somerandomstring.ui.naba.casa back to your instance.
>
>
> As I said - yes, your [...]

Yes, totally understand that is the proxy.  But what I'm saying is - it doesn't matter.  I can get that flac file directly from the HA Cloud reverse proxy or locally.  But, from the Voice PE perspective everything is coming from HA Cloud.  So what I have setup locally - really doesn't matter at all.  Since my Voice PE is on a different VLAN than my HA instance - I can very plainly see that there's no direct IP connection to my local instance - even though it is accessible.  So, really - the file is coming from HA Cloud, regardless of where it is.  Getting the file is not the issue.  The TLS handshake is.

davidjmeier, 2025-03-13:
Hey, @zackslash or @xpac1985 - can you test this? This seems to be reliable.  Tested it about 20-30 times.  Looking to see if you experience the same.  Thanks!  If you were doing the external method to set config you can delete that - I just copied the way the base yaml was implementing config flags and now when I `Validate` I'm seeing all the values in the log.

```
substitutions:
  name: ha-voice-00-xxxx
  friendly_name: ha-voice-00-xxxx

packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB: "y"

      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"

      CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"

      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "n"
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_2: "y"
      CONFIG_MBEDTLS_SSL_MIN_MINOR_VERSION: "3"
      CONFIG_MBEDTLS_SSL_MAX_MINOR_VERSION: "3"
      CONFIG_MBEDTLS_SSL_TLS1_3_KEY_SHARE: "n"
      CONFIG_MBEDTLS_SSL_TLS1_3_PSK_EXCHANGE: "n"

api:
  encryption:
    key: !secret api_encryption_ha-voice-00-xxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.x.x.x
    gateway: 10.x.x.x
    subnet: 255.255.255.0
    dns1: 10.x.x.x
```

zackslash, 2025-03-13:
@davidjmeier I removed the external workaround and confirmed I was getting the TLS failures, I built and flashed new firmware with your changes above. So far (around 20 tests) not a single failure. I will keep it running for longer and update

davidjmeier, 2025-03-13:
> [@davidjmeier](https://github.com/davidjmeier) I removed the external workaround and confirmed I was getting the TLS failures, I built and flashed new firmware with your changes above. So far (around 20 tests) not a single failure. I will keep it running for longer and update
> 
> Currently running
> 
> ```
>     sdkconfig_options:
>       CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
>       CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
>       CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
>       CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB: "y"
> 
>       CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
>       CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"
> 
>       CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"
> 
>       CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "n"
>       CONFIG_MBEDTLS_SSL_PROTO_TLS1_2: "y"
>       CONFIG_MBEDTLS_SSL_MIN_MINOR_VERSION: "3"
>       CONFIG_MBEDTLS_SSL_MAX_MINOR_VERSION: "3"
>       CONFIG_MBEDTLS_SSL_TLS1_3_KEY_SHARE: "n"
>       CONFIG_MBEDTLS_SSL_TLS1_3_PSK_EXCHANGE: "n"
> ```

Awesome!  I'm testing 1.3 only now, again - to see if we can squeeze two working configs out of this.  Glad to hear you're better off with this config too.  Will update once this builds...

davidjmeier, 2025-03-13:
Final update for now...

I tested a number of TLS 1.3 *only* configurations.  This seemed to be the best:

```
      # Enable both TLS 1.2 and TLS 1.3 (so WPA Supplicant works)
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_2: "y"
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "y"

      # Set TLS 1.2 as the minimum for WPA Supplicant (required)
      CONFIG_MBEDTLS_SSL_MIN_MINOR_VERSION: "3"

      # Set TLS 1.3 as the maximum for outbound HTTP connections
      CONFIG_MBEDTLS_SSL_MAX_MINOR_VERSION: "4"

      # Disable TLS 1.3 extensions that might be mistakenly sent with TLS 1.2
      CONFIG_MBEDTLS_SSL_TLS1_3_KEY_SHARE: "n"
      CONFIG_MBEDTLS_SSL_TLS1_3_PSK_EXCHANGE: "n"
      CONFIG_MBEDTLS_SSL_TLS1_3_EARLY_DATA: "n"

      # Force mbedTLS to use only TLS 1.3 for HTTP
      CONFIG_ESP_TLS_USE_TLS1_3_ONLY: "y"
      CONFIG_ESP_TLS_MIN_TLS_VERSION: "4"
      CONFIG_ESP_TLS_MAX_TLS_VERSION: "4"
```

However, **_it doesn't work_**.  I thought that combo would, based on what I've seen, filter out the problematic HTTP calls - but also leave in place TLS1.2 for Wifi - but, no dice.  But I only post the above to spare folks any time in trying.

---

For those coming to this thread now - you can get your Voice PE to work reliable with your Internet URL set to HA Cloud reliably with the config below.  This disables TLS 1.3 completely and has no downsides that I can see, other than not being able to use TLS 1.3 for the HTTP calls.  I don't like the downgrade - but it's a hell of a lot better than what Nabu Casa Support recommended to me: disable TLS altogether - yeah, no thanks.  Anyway, this is 100% an issue with the Voice PE firmware and most notably steps from the `mbedTLS` module.  I'm done messing with this for now - I'm not sure that anyone in Nabu Casa really seems to be concerned with this issue based on what I've seen - I just wanted it to work and not have to mess with changing my `Internet` setting in `Network`.

Here's what I'm running and it is, as of this writing, 100% reliable:

```
substitutions:
  name: ha-voice-00-xxxx
  friendly_name: ha-voice-00-xxxx

packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  flash_size: 16MB
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_ESP32S3_DEFAULT_CPU_FREQ_240: "y"
      CONFIG_ESP32S3_DATA_CACHE_64KB: "y"
      CONFIG_ESP32S3_DATA_CACHE_LINE_64B: "y"
      CONFIG_ESP32S3_INSTRUCTION_CACHE_32KB: "y"

      CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST: "y"
      CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY: "y"

      CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC: "y"

      # Overrides from default start here
      #
      # Disable TLS 1.3 completely
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_3: "n"

      # Enable TLS 1.2 (required for WPA Supplicant and ESP-TLS)
      CONFIG_MBEDTLS_SSL_PROTO_TLS1_2: "y"

      # Set both minimum and maximum TLS versions to 1.2 (forces only TLS 1.2)
      CONFIG_MBEDTLS_SSL_MIN_MINOR_VERSION: "3"
      CONFIG_MBEDTLS_SSL_MAX_MINOR_VERSION: "3"

      # Disable TLS 1.3-specific extensions that may still be present in ClientHello
      CONFIG_MBEDTLS_SSL_TLS1_3_KEY_SHARE: "n"              # Prevents sending key_share extension (TLS 1.3 key exchange)
      CONFIG_MBEDTLS_SSL_TLS1_3_PSK_EXCHANGE: "n"           # Disables PSK key exchange modes for TLS 1.3

      # Remove TLS 1.3 compatibility and mixed-mode features
      CONFIG_MBEDTLS_SSL_TLS1_3_COMPATIBILITY_MODE: "n"     # Ensures no fallback to TLS 1.3 behavior
      CONFIG_MBEDTLS_SSL_TLS1_3_MIXED_MODE: "n"             # Prevents using TLS 1.3 alongside older versions

      # Fully disable additional TLS 1.3 features
      CONFIG_MBEDTLS_SSL_TLS1_3_EARLY_DATA: "n"             # Disables 0-RTT early data (TLS 1.3 optimization)
      CONFIG_MBEDTLS_SSL_TLS1_3_MIDDLEBOX_COMPAT: "n"       # Prevents middlebox compatibility mode (avoids sending unnecessary extensions)

api:
  encryption:
    key: !secret api_encryption_ha-voice-00-xxxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.x.x.x
    gateway: 10.x.x.x
    subnet: 255.255.255.0
    dns1: 10.x.x.x
```

jonofe, 2025-03-13:
> Here's what I'm running and it is, as of this writing, 100% reliable:

That now seems to work for me. 🥇 
I tried it 10-20 times without any problem. Very good job!!! 💯 

> Right now the device **_DOES NOT_** use the local HA instance to retrieve the voice files. It ALWAYS traverses the external network. Why? There's absolutely no reason for that. Fix it, please. The Voice PE should try local first, there's no reason for it to do this:
> 
> (Voice PE) --> (HA Cloud) --> [proxy through to] --> (HA Local) --> (HA Cloud) --> (Voice PE)
> 
> This should just pick up the file locally once Cloud TTS generates it and it's cached. 🤦‍♂

@NabuCasa: And please add the option either to ignore the SSL certificate verification or to add the root certificate of an self hosted CA.

abliss, 2025-03-13:
Thanks for all the work. Is there any chance of this getting merged
upstream so I don't have to build my own firmware to use the device I
purchased?

davidjmeier, 2025-03-13:
> Thanks for all the work. Is there any chance of this getting merged
> upstream so I don't have to build my own firmware to use the device I
> purchased?

Do you have the `ESPHome Builder` installed by chance @abliss?  Should be pretty easy to just drop the config in and build.  I know a lot of people don't want to do that - but, in this case the issue seems to be on the backburner.

abliss, 2025-03-13:
Thanks, I do have it, but it doesn't seem to work very well, so I am
hesitant to "take control" for fear that I won't be able to "give control
back" and hop back on what will hopefully be a steady train of reliable
auto-updates from nabucasa ...

davidjmeier, 2025-03-13:
> Thanks, I do have it, but it doesn't seem to work very well, so I am
> hesitant to "take control" for fear that I won't be able to "give control
> back" and hop back on what will hopefully be a steady train of reliable
> auto-updates from nabucasa ...

You'll have to take it up with Nabu then.  I just sent them an email outlining the issue and linking back to the fix here in hopes that someone will take this up and provide a more formal fix.  🤞

synesthesiam, 2025-03-13:
> But, from the Voice PE perspective everything is coming from HA Cloud. So what I have setup locally - really doesn't matter at all. Since my Voice PE is on a different VLAN than my HA instance - I [...]

An ESPHome TTS proxy URL is created as relative ([code reference](https://github.com/home-assistant/core/blob/cdead8661d7c0a8fbdfd51f3b5039af5b1d30a21/homeassistant/components/esphome/ffmpeg_proxy.py#L118)), which means it ultimately ends up going through [`get_url`](https://github.com/home-assistant/core/blob/cdead8661d7c0a8fbdfd51f3b5039af5b1d30a21/homeassistant/helpers/network.py#L118) to get resolved. What happens here will depend on your exact configuration, but for many people this will be a local HTTP URL. Something is wrong if this is going to your external URL.

Our intention with designing the TTS proxy was for the ESPHome device (VPE or otherwise) to communicate directly with the Home Assistant server **on the local network**. The TTS audio is not being fetched by the ESPHome device from the TTS source directly here (like HA Cloud), but from the HA server so it can be transcoded with ffmpeg into a suitable format for playback.

If anyone's ESPHome device is hopping out of their LAN to talk to HA indirectly through the external URL for TTS, this is unexpected and should be corrected before taking control of the device and changing firmware flags. If this is the behavior you want (VPE <-> Internet <-> HA) then it sounds like a separate issue with TLS and ESP-IDF. This is an important issue to raise with Espressif (the authors of ESP-IDF), but we need to be cautious with adding new firmware flags so we don't break things for other users.

Destarianon, 2025-03-13:
> Hey @synesthesiam , thanks so much for the detail and context around this.
> 
> One commonality that seems to be present here is that the majority that have this issue are running the 'LetsEncrypt' addon.
> Could it be the case that the addon results in get_url always resolving to the external URL?
> 
> If that's the case it seems like maybe there could be two separate issues here and one is hilighting the other, potencially:
> 
> 1.  Lets encrypt addon causes external URL to always be used.
> 2. There are certificate issues with VPE when the external URL are used.
> 
> I can try dig further into the 1st issue, does that seem plausible? 

For reference: I am experiencing the issue and I am not using LetsEncrypt. I am using a custom addon that issues certificates from my local-only ACME provider using acme.sh as the client.

Effectively it does the same thing, just putting certificates in the /ssl/ folder, but that would be an issue with the HTTP integration in HomeAssistant, not the LetsEncrypt Addon specifically.

In my case I also use split-brain DNS, "homeassistant.mydomain.com" is being used in my local network and NabuCasa uses the same domain on the public internet, but the resolve to different ips depending on if your inside my network or not. Because of that both my Local URL and Internet URL are exactly the same.

synesthesiam, 2025-03-13:
> Could it be the case that the addon results in get_url always resolving to the external URL?

I can't see how the add-on itself could do that, but I'm wondering if the setup instructions for the add-on are encouraging something incorrect?

> Because of that both my Local URL and Internet URL are exactly the same.

Ah, that seems likely to be the source of the problem. Maybe the VPE reaches out to that URL in such a way that your router resolves it differently? Checking my own VPE logs, I can see it using the local URL (http://IP_ADDRESS:8123/...). What do you see?

Destarianon, 2025-03-13:
My firewall intercepts all regular DNS and blocks DNS over TLS so I am very confident that it is 100% of the time resolving to the internal IP address. Split-brain DNS is very common, so it's not an abnormal configuration, albiet not common in people's homes. 

I can confirm all other systems, webhooks, etc work 100% normally. Remote control from Nabu Casa works correctly, only VoicePE is having an issue.

I have been busy with work so I haven't had time to investigate the firmware customizations, but will do so this weekend. I want to rule out that my behavior is not caused by my internal CA being untrusted or some other behavior.

zackslash, 2025-03-13:
Reading the get_url function, it looks like it could be the case that if TLS is configured in HA the the external URL is preferred.

```
if prefer_external is None:
        prefer_external = hass.config.api is not None and hass.config.api.use_ssl

```

I did a search for this and it looks like it has been flagged https://github.com/home-assistant/core/issues/139060

Maybe I am misreading I hope to get some time to test this in the Core project locally, but I think that may explain it

synesthesiam, 2025-03-13:
@zackslash Interesting, thank you! I will start my investigation there.

davidjmeier, 2025-03-13:
As I've said numerous times and even shown this with packet capture screenshots - yes, my Voice PE goes directly to the Internet URL which is provided by HA Cloud.  I know there has been a number of folks that have said this isn't possible, but based on what @zackslash has shown above - it seems as though the way my Voice PE is operating, and the way it's been validated clearly shows that this can happen.

> Part of the reason we have an awesome community is by having patience and humility when talking to each other. Being rude to staff or other community members doesn't help anyone out.

Fully agreed and that goes both ways.  My poitn here is that giving people the advice to downgrade local networks to insecure is really frustrating coming from a company and project that is foundationally centric around end user privacy.  I think the upside here is we've proven that things are not working as advertised based on common configurations and hardware and when someone spends time proving these things out only to be met by "it doesn't work like that" and brushed off is not a great feeling.

Again, I truly appreciate the community and all the folks behind it.  Here's to hoping the Voice PE gets better through conversations like this!

Palleri, 2025-03-27:
There is also an issue on the native android app while using mTLS.
[Github issue](https://github.com/home-assistant/frontend/issues/21851)
That I reported and didnt get any feedback on. So thought it might be connecred with issue.

rjbrown99, 2025-03-31:
> For those coming to this thread now - you can get your Voice PE to work with your Internet URL set to HA Cloud reliably with the config below. This disables TLS 1.3 completely and has no downsides that I can see, other than not being able to use TLS 1.3 for the HTTP calls. I don't like the downgrade - but it's a hell of a lot better than what Nabu Casa Support recommended to me: disable TLS altogether - yeah, no thanks. Anyway, this is 100% an issue with the Voice PE firmware and most notably stems from the `mbedTLS` module. I'm done messing with this for now - I'm not sure that anyone in Nabu Casa really seems to be concerned with this issue based on what I've seen - I just wanted it to work and not have to mess with changing my `Internet` setting in `Network`.

Quick question - when you say "the config below" - are we expecting this config to be dropped into configuration.yaml on the HA device? Or is this a different config specific to rebuilding a custom firmware for the HA Voice device? 

As you might expect, I'm having the identical issue with TLS and a Let's Encrypt cert on my HA device. Would love to fix it, just a bit unclear on steps here. Thanks!

rjbrown99, 2025-03-31:
OK, I think I figured it out. I'll update this post as I proceed with applying the fix here.

- On your Home Assistant install, navigate to Settings -> Add Ons
- Click Add On Store in the bottom right
- Click on ESPHome Device Builder, install the add-on, start it
- On the Add On page, click to open web UI
- Assuming you already added your HA Voice to Home Assistant, at the top you should see "Discovered <x> device", click "Show" at the right
- On the HA Voice device, click "Take Control"
- Save off your encryption key, just in case. Now you should be presented with a YAML config file that looks a lot like [the one posted above](https://github.com/esphome/home-assistant-voice-pe/issues/315#issuecomment-2722101531)
- Change this YAML file to include the new section starting with esp32. Click save, then install. 
- It's going to compile now, and that will take a long while unless you have fast hardware.

davidjmeier, 2025-03-31:
Glad you got it figured out @rjbrown99!  Really there's two issues with VPE:

1. The VPE does not handle TLS 1.3 correctly as is outlined in all the posts above.  There is a workaround that I posted that you've successfully used.
2. The VPE software does not correctly reach out internally as it should.  This is validated by this issue: https://github.com/home-assistant/core/issues/139060  - however it seems as though there has been little to no progress on either of these issues.

Finally: Nabu Casa has stated that they do not support TLS for VPE.  It is nowhere in their documentation and they don't state this anywhere with respect to the product page.  They have told me to "downgrade" my internal setup to cleartext of which I told them that this goes against everything that VPE / HomeAssistant really stands for.  What I would suggest is that if you are a paying HA Cloud customer to open a ticket and bring up these issues.  I don't think (now that it's been a few weeks) that it seems like anyone is taking this as an actual issue.

I could be wrong, but the lack of movement in fixes for this seems to indicate that.

adamaze, 2025-03-31:
I've been following this issue for a while, hoping for a real fix... But maybe one isn't coming?

Would it be helpful if we tried to break this down as simply as possible? I could be wrong, but maybe this is all it takes to recreate:
1. Brand new HA install
2. Set up letsencrypt
3. Configure Settings>>System>>Network>>Home Assistant URL>>Internet: "nabu casa URL here"
4. Configure Settings>>System>>Network>>Home Assistant URL>>Local Network: "local https URL here"
5. Try to use HAVPE, and notice a distinct lack of TTS response (seemingly because it is sending traffic to the listed "Internet" URL, and is chokes almost every time

rjbrown99, 2025-04-04:
Appreciate the replies. I'm not quite working at this point - I was able to take control, roll my own yaml config, and have it successfully install to the HAVPE. I can see it in ESPHome Device Builder when I run a validation check.

But now it no longer connects to HA (at all), despite it showing up in the normal HA web UI. I've been through the factory reset, reinstall of firmware, recompilation, etc. Still working through it but wow "preview edition" is really code for "time consuming".

TLS support is such a standard thing in 2025, it's mind boggling that this doesn't work yet in the default firmware.

I'm currently looking into how I can "give control back" in Device Builder - IE the opposite of "take control" so I can truly be back at square 1 to try this again.

xpac1985, 2025-04-04:
To make it work in HA, I think you first need to delete. the device from the esphome integration so it can be rediscovered.
For "giving back control", that should happen by reinstalling the firmware via this process:
https://voice-pe.home-assistant.io/guides/firmware-reinstall/

davidjmeier, 2025-04-10:
> Appreciate the replies. I'm not quite working at this point - I was able to take control, roll my own yaml config, and have it successfully install to the HAVPE. I can see it in ESPHome Device [...]

Are you still having issues?  Mine has been working as expected, no issues with the YAML I used as an example above.  I've currently updated firmware twice on the HA VPE and am currently on ESPHome 2025.3.3.

If you look at the releases for the VPE firmware here: https://github.com/esphome/home-assistant-voice-pe/releases

...there have been no fixes.  And there's 2 major issues with the device and firmware at this point:

1. People are assuming, incorrectly, that the VPE will reach out internally first.  The device does not and this is the root cause of why so many people have issues during onboarding.
2. The TLS issue as it stands out of the box with respect to the handling of 1.3 with the `mbedtls` implementation.

It's really disappointing to watch Nabu Casa treat this as a nothing-burger, and again, if you are a paying customer open a ticket.

zackslash, 2025-04-10:
Confirming I still see the same issue in the stock firmware when using TLS. I'm continuing to use the modified firmware until it's resolved.
In addition to this issue it's worth watching [this related ticket](https://github.com/home-assistant/core/issues/139060) but it's not been addressed yet.

MorikaWeb, 2025-04-11:
I just got my Voice PE and am having this issue. Very frustrating and kind of mind boggling that a product like this is having these issues in 2025. Needless to say I will not be disabling TLS on my internal network. I will try to reach out to Nabu tomorrow, but this is not a good first impression.

davidjmeier, 2025-04-11:
> I just got my Voice PE and am having this issue. Very frustrating and kind of mind boggling that a product like this is having these issues in 2025. Needless to say I will not be disabling TLS on my internal network. I will try to reach out to Nabu tomorrow, but this is not a good first impression.

Hey @MorikaWeb, take control of the VPE in ESPHome and leverage the configuration I laid out here: https://github.com/esphome/home-assistant-voice-pe/issues/315#issuecomment-2722101531

This has been working reliably for me.

And, yes, please open a ticket and reference this thread as well as this one: https://github.com/home-assistant/core/issues/139060.

sky-luke, 2025-04-15:
Hello, I tried to run several times the install suggested here https://github.com/esphome/home-assistant-voice-pe/issues/315#issuecomment-2722101531
Unfortunately i get stuck every time after HAVPE reboots during execution

> [20:40:14][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
> [20:40:14][D][esp32.preferences:114]: Saving 1 preferences to flash...
> [20:40:14][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
> [20:49:18][D][light:036]: 'voice_assistant_leds' Setting:
> [20:49:18][D][light:051]:   Brightness: 66%
> [20:49:18][D][light:058]:   Red: 100%, Green: 0%, Blue: 0%

Do you have any suggestions of what i might be doing wrong ?

davidjmeier, 2025-04-16:
> Do you have any suggestions of what i might be doing wrong ?

Hey @sky-luke!

A couple of questions for you...

1. Have you had this previously onboarded into HA?
2. Can you share a redacted configuration you're trying to apply?  This looks like a network issue, potentially, so just trying to gauge where the issue is.

For example, this is how my VPE initially connects:
```
INFO Upload took 9.24 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from x.x.x.35 using esphome API
INFO Successfully connected to ha-voice-00-40a2 @ x.x.x.35 in 7.262s
INFO Successful handshake with ha-voice-00-40a2 @ x.x.x.35 in 0.243s
[18:17:15][I][app:100]: ESPHome version 2025.3.3 compiled on Apr 10 2025, 13:39:19
```

So in some respects I'm wondering if your OTA was compiled correctly but something in the network configuration or the auth is incorrect?

You should be able to onboard the device (even though if you're using HA Cloud it will appear to fail) into HA.  At that point the local connectivity should be fine, but then you'd want to apply the workaround to get the device to actually be able to be used in a properly configured TLS environment.

marcelveldt, 2025-04-29:
Hi all, I notice this issue report got a bit heated and also has a lot of unrelated comments.
First of all let me be clear of what the intended behavior/setup should be (regardless of what you think about that):

- A Home Assistant instance runs it's own (unencrypted) webserver on port 8123 to serve the Home Assistant webinterface, API and related services, such as the TTS audio. 

- Within the Home Assistant configuration you can configure an internal URL and an external URL.

- The whole idea is that the internal URL should point to the default, unencrypted webserver. While it does have authentication (so its not completely insecure) its not served over HTTP, so it is not encrypted. Our argumentation is that this should be fine as this webserver is invented to be only exposed/used within your own (trusted!) local network and never exposed to the internet.

- The second option you can configure is the external URL, in other words, an alternative URL to your home assistant instance which is using a secure connection. If you are a Home Assistant subscriber, we take care of that by securely tunneling the traffic to your local instance without the need for you even to have ports opened or whatnot. You can access your HA instance from all over the world with an HTTPS url we generate for you and you can even configure your own custom domain name if you want. If you are not a Nabu Casa subscriber, you need to manually setup this part by configuring a reverse proxy such as nginx to connect the outside world to your local instance with an HTTPS address. Or you choose to not use that at all and use a vpn tunnel to your home.

- Home Assistant historically also known the option to re-configure the builtin webserver with a certificate. In that case the builtin URL is upgraded to HTTPS serving the certificate of your choice, which can be an actually certificate on your personal domainname but it can also be something self signed. While this may sound tempting at first, this historical feature is (imo) bad for multiple reasons. First of all it breaks the entire principle of the local HTTP-based server on the internal network and the HTTPS one for external communication but second of all it can never be as secure and flexible as running a full fledged reverse proxy in front of the builtin server. Anyways, that station has passed now but I wanted to mention it here to get the complete picture.

- Microcontrollers like a ESP32 do not run a full Linux distribution. Per definition their HTTP stack is limited in terms of performance, not can they run/store encryption in a fast and reliable way. It gets better all the time but as you all have experienced for example connecting to a TLS1.3 only host can already be troublesome, next to the fact that serving just some audio feed (so no personal data) is a waste of resources to be done over HTTPS. It would be tons more reliable when the audio is served over a plain/simple HTTP path within the LOCAL network.

So summing up all this info you are running into issues because while developing the VPE there has been an assumption made that the internal webserver is the original, non-HTTPS webserver to connect to HA in your local network.
It was never intended to serve the audio data to the VPE (or any other ESP device) over the cloud or the local webserver with some self signed certificate.

For now there is little we can do. This is how it was designed/intended and it works well that way for many, many users.
Because most users do not touch the default webserver and run HA Cloud or an additional nginx proxy for the external URL only.

That said. I think there is room for an improvement here, which is then a feature request:
Maybe Home Assistant should by default host an additional webserver just for the sole purpose of hosting unimportant and unencrypted data, such as the TTS responses, some random doorbell sound and whatnot. There is no web UI, no API on that webserver, no sensitive information at all.

That way the default webserver can still be protected as you wish to configure it. As long as your internal clients (including the HA apps) work with that, its all completely fine. And for serving the TTS audio, the ESP device is served to the webserver URL on the unencrypted endpoint.

Note that this is a suggestion for a feature request, which can be implemented with enough user demand.
It should at least first be proposed as an architecture proposal.

Another possible direction of a solution is serving the audio as part of the upcoming new audio protocol we have in mind so it will be sent as raw data instead of served as HTTP/HTTPS endpoint. But yet again, this is not something that will eb available soon because that is all still in planning stage, nothing concrete yet.

If you want/need a solution today, setup your network the way we intended it to work. You may not like that approach but it is what works today. Then next, please propose an architecture discussion on the relevant channel for the future enhancement.

Thanks for understanding.

marcelveldt, 2025-04-29:
I agree 100% on that and I think steps should be taken to work towards that goal (have encryption by default even for the local connection) but we can't magically make that happen right now. That is what I meant with making a difference between what is supposed to work today and what feature requests there are to offer additional functionality in the future.

xpac1985, 2025-04-30:
Hey @marcelveldt,

thanks for your extensive reply. I'll spare you the obvious personal uproar I feel when reading about trusted home networks, but I understand that this has been a design decision (in the past). 😉

> So summing up all this info you are running into issues because while developing the VPE there has been an assumption made that the internal webserver is the original, non-HTTPS webserver to connect [...]

I have to disagree here, because I think your understanding of the problem is not complete.
The issue is not that Voice PE does not work with our local TLS setups - it does this very well. (at least I haven't read anybody in this thread report otherwise). The only TLS setup it does not work well with is that of Nabu Casa Cloud, for reasons that I didn't really understand so far (as both are TLS 1.3, there have to be some subtle differences I overlooked).

However, the actual reason for this whole problem is https://github.com/home-assistant/core/issues/139060 - an issue which causes the Voice PE to use the external URL, even though an internal URL is set and would work. I read https://github.com/home-assistant/core/pull/66039 to understand why this change was made - however this was before Nabu Casa Cloud became a thing, and using the external URL would increasingly mean that accessing HA via external URL results in a call to a cloud service, going back to your home, instead of staying inside your network.

As I understand it, the URL is handed from HA to the Voice PE, so the fix would've to be made in HA, not in the Voice PE code.
I tried looking in https://github.com/home-assistant/core/tree/441bca5bdaa2f9afcd970e7cb1aa0b75ade32c1f/homeassistant/components/esphome for any use of that `get_url` function, but failed, and to be fair that code is more complex than what I'm used too, so I hope someone more familiar with the code can figure out a fix (e.g. for Voice PE calls to have the parameter `prefer_external` set to `false`, so that it actually does what you said - use the internal URL to access HA.

Hope that sheds a little more light on it so we can get this fixed, because I doubt "Voice PE always pulls via Nabu Casa Cloud because I run LetsEncrypt at home" was planned to be the way it works. 😉

marcelveldt, 2025-04-30:
It has to do with eachother - the logic is simply not complete.
It will auto pick external URL if the internal one has SSL configured. Like I said, the design choices of the past are not helping here. The fact that you can configure the builtin webserver with your own SSL certificate is the reason it breaks.

If you configure the normal webserver without SSL, and set the internal URL to simply the local network url such as http://192.168.1.1:8123 it will all work fine.

To access your HA instance outside from home, setup a nginx proxy in front of HA that listens on the outside world to your dns name and SSL certificate and then forwards the traffic to HA's default webserver (insecure).
Or run HA Cloud ofc, that will take care of that step for you.

> because I doubt "Voice PE always pulls via Nabu Casa Cloud because I run LetsEncrypt at home" was planned to be the way it works.

Yeah, well... I'm afraid that is what happened when that is configured on HA directly. You will be fine if you configure lets encrypt on a dedicated nginx proxy, there are even a couple of add-ons available which do all the work for you.

[6 more replies]