esphome/home-assistant-voice-pe #316: [REQUEST] separate media_player for TTS and music playback?

dyanik9, 2025-02-02:
When connecting an external speaker to the voice PE, TTS is also played on it. Allowing simultaneous playback on both the internal and external speaker via two separate media_players would allow to use the internal speaker for TTS only and the external one for music playback.

This makes sense if, for example, a music system is connected, which turns on automatically when something is played and the user doesn't want it to be turned on for each voice command response. Since it can take some time for that to turn on and thus cut off part of the answer or even the whole answer.

StrandmonYellow, 2025-02-03:
This is not possible due to a hardware limitation. Take a look at https://github.com/esphome/home-assistant-voice-pe/issues/263.

ItsHarper, 2025-02-25:
This likely _is_ possible with an electrically-small hardware mod (I'm not sure how easy it would be physically). See https://github.com/esphome/home-assistant-voice-pe/issues/263#issuecomment-2681096314

Perchycs, 2025-03-27:
I am also keen to do exactly this. I have found [this HA forum post](https://community.home-assistant.io/t/only-use-external-speakers-for-music-but-for-voice-interaction-use-internal-speaker/868690/2?u=perchy) showing exactly how to disable the hardware MICDET. It is very easy.

Seems like since the ReSpeaker is also made by seeedstudio and has a similar setup for this, there is a good chance you can software control the audio output using i2c commands, but I'm not too keen to do this mod before knowing whether I will be able to figure out the software side on my own. 

Keen to hear more from others.

ItsHarper, 2025-03-28:
> Seems like since the ReSpeaker is also made by seeedstudio and has a similar setup for this, there is a good chance you can software control the audio output using i2c commands, but I'm not too keen to do this mod before knowing whether I will be able to figure out the software side on my own.

As has been further clarified in the forum post, you can solve this by hooking up the disconnected leg of the transistor to an available GPIO.

spudje, 2025-05-08:
I hope this is something that comes out of the box with a future revision of the hardware. I don't feel like modding it myself.

bitte-ein-bit, 2026-11-13:
So here we go.

Hardware mod, two options exists:
1. Use the same approach as [ReSpeaker Lite](https://github.com/respeaker/ReSpeaker_Lite?tab=readme-ov-file#disable-the-headphone-jack-detection-circuit) and short the pins on the audio jack (see attached picture). You'll lose detection if something is plugged in at all. Easier to solder and good for stationary setups.
2. lift a pin on the transistor as described in the video in the [forums](https://community.home-assistant.io/t/only-use-external-speakers-for-music-but-for-voice-interaction-use-internal-speaker/868690). I didn't try this one.

Software mod, again the ReSpeaker Docs lead in the right direction. Add this to your ESPHome Config:

```yaml
i2c_device:
  id: internal_i2c_device
  i2c_id: internal_i2c
  address: 0x18

select:
  - platform: template
    name: "Audio Output"
    id: audio_output
    icon: mdi:audio-video
    options:
      - "Speaker"
      - "AUX"
    initial_option: "Speaker"
    optimistic: true
    restore_value: false  # DONT restore old state, speaker mode on reboot
    on_value:
      then:
        - lambda: |-
            if (x == "Speaker") {
              ESP_LOGI("aic3204", "Switching to Speaker output...");
              id(internal_i2c_device).write_byte(0x00, 0x01);
              id(internal_i2c_device).write_byte(0x12, 0x00);
              id(internal_i2c_device).write_byte(0x13, 0x00);
              id(internal_i2c_device).write_byte(0x10, 0x40);
              id(internal_i2c_device).write_byte(0x11, 0x40);
              ESP_LOGI("aic3204", "Speaker output enabled.");
            } else if (x == "AUX") {
              ESP_LOGI("aic3204", "Switching to AUX output...");
              id(internal_i2c_device).write_byte(0x00, 0x01);
              id(internal_i2c_device).write_byte(0x10, 0x00);
              id(internal_i2c_device).write_byte(0x11, 0x00);
              id(internal_i2c_device).write_byte(0x12, 0x40);
              id(internal_i2c_device).write_byte(0x13, 0x40);
              ESP_LOGI("aic3204", "AUX output enabled.");
            }
```

This was sufficient to switch. One quirk remained for me, that after boot both outputs where active. I fixed it by adding:

```yaml
esphome:
  on_boot:
    - priority: -100
      then:
        - select.set:
            id: audio_output
            option: "Speaker"
```

[image]