esphome/home-assistant-voice-pe #317: [REQUEST] Button to trigger listening

MrsXenchen, 2025-02-04:
With Amazon's Echo it was possible to set up, that the echo was triggered to ask a question and set an action based on the answer. It was not easy by any means, and I only managed to do so with the examples provided in the explanations. But still, it was quite neat!
"All" that is missing is an action to make the PE listen for my answer. I know, that I could set that up myself, but I would not like to miss future updates for the PE.

A few examples of automations, I would realize with this (I am sure, I will find a few more, if I try ;)):

1.  When TV starts, ask which source should be played: Plex, Netflix, YT etc.
2. When oven starts program, ask who is cooking and send notification to the right person (we are 4 adults and don't want to stay in the kitchen nor does everybody wants to get a notification, if one puts something in the oven for himself).
3. When Garage is open too long, ask, if it should be closed.

Thanks for consideration and great work so far!
Manu

StrandmonYellow, 2025-02-05:
Take a look over here: https://github.com/esphome/home-assistant-voice-pe/pull/272

kahrendt, 2025-04-02:
This is supported in Home Assistant 2025.4 and VPE firmware 25.3.4 using the start_conversation action.

closed 2025-04-02 by kahrendt