esphome/home-assistant-voice-pe #318: Make "Timer Finished" an external switch

rwrozelle, 2025-02-04:
Would like to be able to use "Timer Finished" from HA, specifically from Automations that use Calendar Events or HA Timers.

andreasbrett, 2025-02-08:
I think there were plans to expose them as events. Next week is another voice chapter but I don't expect this to come.

rwrozelle, 2025-02-08:
That in the opposite direction of what I’m looking for.  I want to leverage the “timer finished” sound and how the HAVA reacts.

nascentt, 2025-05-28:
I just discovered thus issue and am really disapointed.  I had hope to replicate my automation from alexa, where I notify devices when a timer finishes.
I was sure it'd be possibly with Home Assistant and can't believe it's not.

Please add this capability.

nascentt, 2025-05-28:
> If you take control of your VA-pe, you can externalize the Timer Ringing Switch with the following added to your device's YAML:
> 
> ```
> switch:
>   - id: !extend timer_ringing
>     name: Timer Ringing
>     internal: false
> ```
> 
> See https://github.com/rwrozelle/VA-pe-Timer-Ringer-Used-with-Calendar-Alarm-Event for a writeup on how I added the ability to use Calendar Events to cause Timer Ringing to switch on

ah interesting.  I'm glad there is a way at least.  Although I'm not confident enough to take control of va-pe , especially as it apparently prevents new firmware updates.  
Is there any hope for the timer ringing event being exposed some day in the future without having to take control?

rwrozelle, 2025-05-28:
When you take control, what happens is a local YAML file is created and it references the VA-pe YAML as a package.
```
packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml
```
Then anytime ESPHome Integration is updated, each device signals that it wants to be updated in the ESPHome Builder UI and the System/Updates UI.  There is a little bit of chicken and egg, where you want to make sure that the home-assistant-voice.yaml is up to date prior to running you updates in ESPHome Builder UI.

nascentt, 2026-04-17:
I held strong for a year, but screaming at nabu to stop the timer without it complying finally wore out my patience so I "took control"  of it, and now it's not reachable. i've spent the last 2 days trying to get it working again with esphome, and then trying to recover it.  even trying to flash the firmware but the https://esphome.github.io/home-assistant-voice-pe/ site can't connect to it. i even installed some CH340 drivers from a dodgy chinese site

let this serve as a warning for anyone else hoping to try and make the timers of this more usable. don't try