esphome/home-assistant-voice-pe #319: Support for Action `assist_satellite.start_conversation`

HarvsG, 2025-02-06:
Home assistant 2025.02 introduced [a new action](https://www.home-assistant.io/integrations/assist_satellite/#action-assist_satellitestart_conversation) `assist_satellite.start_conversation`, despite the VA PE being a `assist_satellite` this action does not appear supported, as VA PEs do not appear in the UI target picker.

Forum issue: https://community.home-assistant.io/t/voice-assistant-pe-support-for-assist-satellite-start-conversation/842716

StrandmonYellow, 2025-02-07:
I suspect that this topic will be brought up during the [year or the voice chapter 9](https://www.youtube.com/watch?v=k6VvzDSI8RU) .

trevm8, 2025-02-07:
I have the same issue. I can manually add in the satellite entity and it appears to take it from the UI however when I run the automation, nothing happens

qJake, 2025-02-07:
I suspect that, even though the entity shows up as `assist_satellite.xxx`, it is not sending the `[START_CONVERSATION supported flag](https://developers.home-assistant.io/docs/core/entity/assist-satellite/#supported-features)`. Whether the firmware actually supports the feature, regardless of sending the flag or not, is a different issue. 🙂

I hope this gets added, I would love to prompt people for actions with Voice PE!

DanStennett, 2025-02-17:
If this can be implemented for voice PE, I can finally get rid of my Alexa devices. Cannot wait!

mazdarx2001, 2025-02-19:
I have the same issue with my Wyoming satellite with entity "assist_satellite.office_satellite". I get the following error: Error: Entity assist_satellite.office_satellite does not support action assist_satellite.start_conversation

erkamkavak, 2025-02-21:
As far as I can see, esp device codes are prepared to trigger the pipeline with wake word detection. Wake word detection sends a VoiceAssistantRequest message to the esphome integration and integration starts the voice assistant pipeline. After the pipeline is started, esphome integration sends VoiceAssistantResponse message back. This message is the one that triggers microphone streaming on esp device. 
For external start conversation, lots of things about the architecture need to be changed.

itispip, 2025-03-01:
yes much needed. Currently the "Press button“ is a event not a button, and voice_assistant.start, voice_assistant.stop, voice_assistant.start_continous are not exposed to HA as callable service, not know why.

kahrendt, 2025-04-02:
This is supported in Home Assistant 2025.4 and VPE firmware 25.3.4.

closed 2025-04-02 by kahrendt