esphome/home-assistant-voice-pe #320: [REQUEST] Support choosing response media_player in config

Ramblurr, 2025-02-08:
Note: this is not a duplicate of [issue 316](https://github.com/esphome/home-assistant-voice-pe/issues/316). Issue 316 is about using the AUX/3.5mm audio out on the PE for music and the builtin speaker for Assist's response.

This request is about configuring the PE such that the TTS response Assist generates can be configured to go to a different media_player entity entirely.

Apparently this is already possible for those building custom esphome voice satellites by specifing `voice_assistant.on_tts_end` in the esphome config.

- https://community.home-assistant.io/t/year-of-the-voice-chapter-5/655319/25
- https://community.home-assistant.io/t/voice-assistant-have-assistant-reply-on-a-different-media-player/625875/6

kosmaluk, 2025-02-12:
Popieram. Funkcja wyboru media playera do ogłaszania odpowiedzi jest bardzo pożądana

kosmaluk, 2025-02-21:
you can easily take over the code and make changes without losing the ability to update. A new Home Assistant Voice Firmware update sensor has been added. You can add your own media_player or create start/stop wakeword buttons and in automations make a continuation of the conversation, dialogflow, etc.

Ramblurr, 2025-02-21:
> you can easily take over the code and make changes without losing the ability to update

Can you describe how to do this? It isn't obvious to me. The PE doesn't show up in esphome builder.

kosmaluk, 2025-02-21:
You need to enable TAKE CONTROL and install the device in Esp Builder. If you don't see it, enable the three dots at the top and "show discovered devices". After installing, you can add any elements you are interested in to the section. For example, mediaplayer or other commands that you will use in automations

`on_tts_end:
     - homeassistant.service:
         service: media_player.play_media
         data:
           entity_id: media_player.google_home_mini # <- change this
           media_content_id: !lambda 'return x;'
           media_content_type: music
           announce: "true"`

phila-dot, 2025-02-22:
Have you successfully done this or just assuming because it works with the older VA’s?

I keep seeing people post this, but it doesn’t work with the VA PE nor the new build  S3 Box. There is nothing returned, so you get no tts audio.

kosmaluk, 2025-02-22:
If I write a solution, it means I did it and it works. Why wouldn't it work? The documentation for ESPHome is the same and applies to all these devices

https://esphome.io/components/voice_assistant.html 

You have to Adopt a device and adapt the code. The examples that are on the forums work perfectly.

I have collected various solutions and together with ChatGpt I have implemented the solutions that I wanted. I have done, among others:
1. Sending sound TTS to another media player
2. Question and answer sensors (I use them to debug STT operation and automation to continue the conversation without waking up again)
3. START and STOP wake word buttons, which I also use for these automations

Everything works great.

Everyone got scared by the entry in the documentation not to take control and are waiting for new functions.

In the movie Chapter 9 they said that they added a new Firmware update sensor which you would normally use to update the device after taking over the code

[image]

[image]

[image]

Ramblurr, 2025-02-22:
> You need to enable TAKE CONTROL

That's the bit I can't find. I have seen other people mention that. But I can't find that option anywhere. Where in HA is a "take control" button?

kosmaluk, 2025-02-22:
something is wrong with your eyesight :).
I wrote how to do it a few posts above :)
ESP Builder --- enable the three dots at the top and "show discovered devices".  HAVPE will show up and click ADOPT

Ramblurr, 2025-02-22:
> something is wrong with your eyesight :).
> I wrote how to do it a few posts above :)

I saw what you wrote, but that post don't contain the string "take control", so I assumed that entire post was for after you "took control". 

I appreciate your assistance here by the way.

> ESP Builder --- enable the three dots at the top and "show discovered devices". HAVPE will show up and click ADOPT

Ok, now we are getting somewhere, the actual button is "show discovered devices" and then "adopt", not "take control". I appreciate the exact instructions.

However, when I click "Show discovered devices", nothing happens (except now the button says "Hide discovered devices". The list of esp devices remains the same, and its just my own custom devices.

Presumably this is because my IOT devices and HA are on different vlans. Is there a way to import it manually without relying on MDNS?

itispip, 2025-02-26:
> Everyone got scared by the entry in the documentation not to take control and are waiting for new functions.

Exaclty, that's my fear. Even "taking control" is easy, and ESPHome added a new Firmware update sensor, but what I'm afraid is update new flash back to original firmware, then update, then ESP take control again, and then customize/add the section not available in factory firmware again --- that can be a repeated pain.

kosmaluk, 2025-02-27:
I don't know, but I updated the device to 2025.2 and it went without a problem and my configuration remained. We'll see how they implement some new functions or elements or add them

[image]

ev1502, 2025-04-15:
> If I write a solution, it means I did it and it works. Why wouldn't it work? The documentation for ESPHome is the same and applies to all these devices [...]

Thank you for posting screenshots, are you by chance able to upload to upload your full configuration file so we can see all the details of the progress you've made?

Ramblurr, 2025-07-26:
Here is my  esphome config for my Voice PE. Note I've changed from calling `media_player.play_media` for the TTS audio to doing instead `assist_satellite.announce`. 

This is because I have one Voice PE that I want to act as *only* a microphone (that's the one the config below is for). Then I have a linux computer running a [wyoming-satellite](https://github.com/rhasspy/wyoming-satellite) which is plugged into my whole home audio streaming system with a [hardware ducker](https://rolls.com/product/DU30b). 

This lets me speak into the Voice PE, but hear the TTS replies over my built-in wall speakers.

```yaml

...normal api/wifi/logging/etc that is in every esphome device is omitted...

packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml@dev

voice_assistant:
  id: va
  media_player: !remove
  on_tts_start:
    then:
      - logger.log: "on_tts_start-start"
      - homeassistant.action:
          action: assist_satellite.announce
          data:
            entity_id: assist_satellite.my_satellite
            message: !lambda 'return x;'
      - logger.log: "on_tts_start-end"
  # or, if you want to use media_player.play_media...
  # on_tts_end:
  #   - logger.log: "on_tts_end-start"
  #   - homeassistant.action:
  #       action: media_player.play_media
  #       data:
  #         entity_id: media_player.your_player
  #         media_content_id: !lambda 'return x;'
  #         media_content_type: music
  #         announce: "false"
```

dcybeldesign, 2026-08-20:
Same conclusion as the linked discussion (home-assistant/discussions#689): this needs firmware-level changes to separate capture and reply, not something fixable from outside the Voice PE.

For anyone landing here who mainly wants scripted announcements (not live conversational replies) routed to a Bluetooth speaker specifically, I built an add-on that exposes any Bluetooth speaker as a native media_player entity in HA: https://github.com/dcybeldesign/ha-mpd-bluetooth-bridge. Works with `tts.speak`/automations today, doesn't touch the core ask here.

limitless-code, 2026-09-10:
Hello, I needed a work around to output audio to a BT speaker (or any supported media player) as self-signed certificates are broken on later VPE firmware.

I have a working -ish solution where I use a helper selector to choose which "media_player" to output audio to and have a some script to takes that media_player value at boot and also allows to choose the media_player on the fly. 

Here are the espHome VPE yaml config customisations:

```
substitutions:
  name: home-assistant-voice-xxxx
  friendly_name: Small Bedroom assistant
packages:
  Nabu Casa.Home Assistant Voice PE: github://esphome/home-assistant-voice-pe/home-assistant-voice.yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
esp32:
  # board: esp32_s3_korvo_2
  board: esp32-s3-devkitc-1
  toolchain: esp-idf
  framework:
    type: esp-idf
http_request:
  verify_ssl: true
  ca_certificate_path: rootCA.crt
  
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  on_boot:
    priority: -100
    then:
      - lambda: |-
          ESP_LOGI("main", "System booted. Waiting for Home Assistant text sensor synchronization...");

# --- GLOBALS ---
globals:
  - id: route_to_external_speaker
    type: bool
    initial_value: 'false'
    restore_value: yes # Remembers the routing state across restarts
  - id: chosen_speaker
    type: std::string
    initial_value: '""'
    restore_value: yes # Permanently stores the entity string in memory

# --- TEXT SENSORS ---
text_sensor:
  - platform: homeassistant
    id: helper_selector_state
    entity_id: input_select.house_audio_source
    internal: false 
    # Automatically triggers at boot connection AND whenever the dropdown updates
    on_value:
      then:
        - lambda: |-
            id(chosen_speaker) = x;
            if (x.rfind("media_player.", 0) == 0) {
              id(route_to_external_speaker) = true;
              ESP_LOGD("main", "Routing updated: EXTERNAL SPEAKER ACTIVE (%s)", x.c_str());
            } else {
              id(route_to_external_speaker) = false;
              ESP_LOGD("main", "Routing updated: LOCAL SPEAKER ACTIVE");
            }

# --- API SERVICES ---
api:
  encryption:
    key: xxxxx
  actions:
    - action: update_audio_target
      variables:
        selected_target: string
      then:
        - lambda: |-
            id(chosen_speaker) = selected_target;
            if (selected_target.find("media_player.") == 0) {
              id(route_to_external_speaker) = true;
              ESP_LOGD("main", "ESPHome Routing Flag updated: EXTERNAL SPEAKER MODE ACTIVE (%s)", selected_target.c_str());
            } else {
              id(route_to_external_speaker) = false;
              ESP_LOGD("main", "ESPHome Routing Flag updated: LOCAL SPEAKER MODE ACTIVE");
            }

# --- VOICE ASSISTANT CONTROL ---
voice_assistant:
  id: va
  # (Your existing mic and hardware settings stay here)
  
  on_tts_start:
    - if:
        condition:
          lambda: 'return id(route_to_external_speaker);'
        then:
          - homeassistant.event:
              event: esphome.voice_pe_tts_redirect
              data:
                text: !lambda 'return x;'
                target_entity: !lambda 'return id(chosen_speaker);'
       
          - delay: 200ms
          - voice_assistant.stop
 ```
This is the helper selector that `text_sensor` uses in the above espHome VPE config:

```
entity_id: input_select.house_audio_source
state: media_player.nux_mighty_bt_speaker_85b1ecde_sendspin_bt_bridge_rc
last_changed: '2026-09-09T20:44:31.768Z'
last_updated: '2026-09-09T22:34:58.282Z'
attributes:
  options:
    - media_player.nux_mighty_bt_speaker_xxx_sendspin_bt_bridge_rc
    - media_player.small_bedroom_assistant_media_player
    - media_player.home_assistant_voice_xxxx_media_player
    - media_player.living_room_tv_2
  editable: true
  icon: mdi:audio-input-stereo-minijack
  friendly_name: Audio_Source
```

limitless-code, 2026-09-10:
I don't know if the above is optimum as a redirect was needed to get TTS to work due to SSL issues for me, how ever it works nicely so far, I can also send it to the other VPE devices in the home I believe but can't test that fully. Certainly the radio player works for that TTS I assume would output to both the origin and the chosen one in the 'helper selector'.

Perhaps something like this could be incorporated into the VPE firmware so it works "out-of-the-box", it would involve, VPE firmware, under HASSIO host > integration > espHome > "voice PE assistant device" there could be a audio_source section (with helper selector) and media controls would switch accordingly? Also I guess it should be trivial for another automation to change according to where someone is in the house or if they have left the house if they want to output to their phone (Home assistant companion app).

limitless-code, 2026-09-10:
Original post on Home assistant community site:

https://community.home-assistant.io/t/voice-pe-assistant-how-to-change-output-source-as-an-alternative-to-pairing-bluetooth-speaker-directly-to-vpe/1024483?u=csp170

limitless-code, 2026-09-12:
Regarding the "Voice assistant controls" automation this works but the confirmation response "done" resumes the music playing. In isolation e.g music_player.media_stop works, but not when using the assist pipeline. Not sure where this occurs? The "space hack" and other suggested hacks won't stop the player resuming after running those actions in this automation.

limitless-code, 2026-09-12:
So as to not add too many comments here, there are a few VPE crash issues caused by the custom config that are being addressed here: 

https://github.com/esphome/esphome/issues/19209

dcybeldesign, 2026-09-12:
@limitless-code I think the "done" is what restarts the music. Your redirect automation sends every Voice PE response to the speaker with tts.speak, including "Done" (and the space from the space hack). When the target is a Music Assistant player, MA treats that TTS as an announcement: it pauses what is playing and resumes it after. So the "Done" resumes the music you just stopped.

To check: disable the "Voice PE: Redirect Audio Stream" automation and say "stop". If the music stays stopped, that's it.

A possible fix, in the redirect automation, replace `conditions: []` with:

    conditions:
      - condition: template
        value_template: "{{ trigger.event.data.text | trim | lower not in ['', 'done'] }}"

The speaker will then just not say "Done". I haven't tested it on your setup, it's only my guess.

MA docs on announcements: https://www.music-assistant.io/integration/announcements/

limitless-code, 2026-09-12:
@dcybeldesign thank you for your suggested fix and well spotted. That works.

Now there's no need for the hack in the "radio player" automation.

```
set_conversation_response: "."
```

So far seems to work including the hardware physical control remapped here:

https://github.com/esphome/esphome/issues/19209#issuecomment-5647488859

limitless-code, 2026-09-12:
The only thing I can think of now is to sync volume control or pass it through to VPE as it is currently out of sync.

limitless-code, 2026-10-05:
@Ramblurr Here is a solution if you are interested.

https://community.home-assistant.io/t/voice-pe-assistant-how-to-change-output-source-as-an-alternative-to-pairing-bluetooth-speaker-directly-to-vpe/1024483/10?u=csp170