esphome/home-assistant-voice-pe #321: Home Assistant Voice PE Microphone Not Responding to Wake Word Activation

BlueMahib, 2025-02-09:
I followed the setup instructions for adding the Home Assistant Voice PE Microphone to my Home Assistant system. However, the device does not respond to speech for wake word activation. I have tried multiple wake words (Okay Nabu, Hey Jarvis, Hey Mycroft), but none seem to work.

Steps Taken:

- Successfully added the device to Home Assistant.
- Verified that the device appears in the Home Assistant UI.
- Attempted multiple wake words with no response.
- Repeated the setup process multiple times (factory reset) without success.

Expected Behavior:

- The device should detect the wake word as expected.

Actual Behavior:

- The device does not respond to any wake words, even after multiple attempts.

Would appreciate any troubleshooting guidance or insights from others who have experienced a similar issue.

fly2806, 2025-02-09:
Hello everyone, I have the same problem and an additional issue:

- I was able to connect the Voice PE to Home Assistant via the setup process. I can use HA to control the LED ring, for example.

- The microphones do not work at all. They do not respond to the wake word, nor does the voice recognition work. They appear to be muted, even though they are not (neither via the switch nor via the software).

- Audio via the loudspeaker does not work. The playback is so slow and distorted that you can't understand anything. This applies to text to speech, playback of music and even if you perform a reset (press and hold the button), the announcement cannot be understood.

I have already reinstalled the software. Unfortunately no improvement. Any support is welcome. Let me know if there is anything I can do to help with more information, testing, etc.

lstanton428, 2025-02-10:
I too am having this issue. In the logs (viewing from here https://esphome.github.io/home-assistant-voice-pe/) I see:
```
...
[13:17:30][E][voice_kit:026]: Communication with Voice Kit failed
[13:17:30][E][component:119]: Component voice_kit was marked as failed.
[13:17:30][E][component:164]: Component voice_kit set Error flag: unspecified
...
```
And then
```
...
[13:17:41][E][component:082]:   Component voice_kit is marked FAILED
...
```

lstanton428, 2025-03-26:
Yes, I was also able to have my hardware replaced.

BlueMahib, 2025-03-29:
Closing this issue as the hardware needs to be replaced.

closed 2025-03-29 by BlueMahib